F5OEO/tezuka_fw #11: Add LibreSDR ADI Pluto+ (hgeek)

sq6emm, 2024-11-25:
Please add support for HamGEEK LibreSDR ADI Pluto+ platform.

https://www.hgeek.com/collections/software-defined-radio/products/hamgeek-adi-pluto-70mhz-6ghz-sdr-software-defined-radio-ad936x-for-libiio-iioscope-sdrsharp-matlab

G0MJW, 2024-11-25:
Plus one for that - but does Evariste have access to one to test?

sq6emm, 2024-11-25:
@F5OEO I can provide you network access to RPI5 with this device connected and also perform tests.

F5OEO, 2024-11-25:
I don't have information of the manufacturer of LibreSDR. Unless I have one at home, it could not be integrated. Maybe community could help me purchase one ?

VasylSamoilov, 2024-12-21:
It's available on aliexpress, I got mine here: https://www.aliexpress.com/item/1005007497350967.html
Not very well build in terms of soldering, but nothing critical. Works fine. Even overclocks with this firmware: https://github.com/hz12opensource/libresdr

I'm not sure how to donate, but if @F5OEO be so kind to set up paypal or some other common platform I can contribute.

F5OEO, 2024-12-23:
Thanks for your support. @sq6emm has made me a donation in order to buy libresdr. However the seller just canceled because I choose the wrong transportation (weird). I just order again, and waiting patiently the shipment. You can however support me by paypal donation https://www.paypal.com/paypalme/f5oeo . Thanks !

G0MJW, 2024-12-27:
Done last week but you may not know who it was.

F5OEO, 2024-12-27:
HamGeek vendor asked me now to clear custom. I never had such difficulties to receive a product from Alixpress. Can't wait to test libresdr but really like to find a better vendor.

G0MJW, 2024-12-27:
This is where I got mine https://www.aliexpress.com/item/1005005165734650.html but it was late 2023.

F5OEO, 2025-01-02:
OK..Nearly there but DHL wait for my custom clearance : So : Libresdr (from hgeek) is 185 + 40 euros transportation + 42 Euros custums : around 265 Euros. Need to compare with other vendors or other equivalent products.

FPSychotic, 2025-01-03:
Same happened to me with TZT shop with the b210 Mini (BlackSDR) and LibreSDR, looks the same than yours but not sure if totally, read something wrong with the PCB , it is said in the product review, no idea is mine too. It looks some people have quite more info that available in common places. Hardware issues. DJI drone hacking, not sure where they get such info.

F5OEO, 2025-01-03:
At least ! Just received Libresdr ! Stay tuned for tezuka compatible firmware...

VasylSamoilov, 2025-01-03:
Great news, indeed. Is debug port operational on your unit? Because mine has "usb device malfunction" situation from the start - effectively I don't have debug/dfu port...

sq6emm, 2025-01-03:
> At least ! Just received Libresdr ! Stay tuned for tezuka compatible firmware...

Happy to hear, sorry that you had so many issues with taxes etc. I have forgotten I had that issue too.. but not that expensive;(

FPSychotic, 2025-01-03:
> Great news, indeed. Is debug port operational on your unit? Because mine has "usb device malfunction" situation from the start - effectively I don't have debug/dfu port...

How to check that? Also read something about the PCB has some layer issues, no clue

sq6emm, 2025-01-03:
This is what happens when I connect my unit to DEBUG port:

```

[99393.819217] usb 5-1: new high-speed USB device number 7 using xhci_hcd
[99393.949710] usb 5-1: New USB device found, idVendor=0403, idProduct=6011, bcdDevice= 8.00
[99393.949722] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[99393.949725] usb 5-1: Product: Quad RS232-HS
[99393.949728] usb 5-1: Manufacturer: FTDI
[99393.985751] usbcore: registered new interface driver ftdi_sio
[99393.985786] usbserial: USB Serial support registered for FTDI USB Serial Device
[99393.985909] ftdi_sio 5-1:1.0: FTDI USB Serial Device converter detected
[99393.985976] usb 5-1: Detected FT4232H
[99393.986579] usb 5-1: FTDI USB Serial Device converter now attached to ttyUSB1
[99393.986615] ftdi_sio 5-1:1.1: FTDI USB Serial Device converter detected
[99393.986639] usb 5-1: Detected FT4232H
[99393.987190] usb 5-1: FTDI USB Serial Device converter now attached to ttyUSB2
[99393.987238] ftdi_sio 5-1:1.2: FTDI USB Serial Device converter detected
[99393.987277] usb 5-1: Detected FT4232H
[99393.987877] usb 5-1: FTDI USB Serial Device converter now attached to ttyUSB3
[99393.987903] ftdi_sio 5-1:1.3: FTDI USB Serial Device converter detected
[99393.987931] usb 5-1: Detected FT4232H
[99393.988508] usb 5-1: FTDI USB Serial Device converter now attached to ttyUSB4

```

VasylSamoilov, 2025-01-03:
usb-c connector named debug should show up as several com ports, afaik.
mine does not respond to usb inquiries.
board don't vias (small thru-holes on the pcb) near lines that connect
transformers to external connectors - board acting as capacitor, at some
frequencies you have terrible attenuation. can be checked with the free
SATSAGEN tool, but I bet you also have it - it is just bad board design.
need to solder coax cable from transformer to connector.

VasylSamoilov, 2025-01-03:
usb-c connector named debug should show up as several com ports (like shown). mine does not respond to usb inquiries. probably bad soldering or bad chip.

board don't vias (small thru-holes on the pcb) near lines that connect transformers to external connectors - board acting as capacitor, at some frequencies you have terrible attenuation. can be checked with the free SATSAGEN tool, but I bet you also have it - it is just bad board design. need to solder coax cable from transformer to connector.

sq6emm, 2025-01-03:
This happens when I connect my unit to OTG port:

```

[99547.287909] usb 5-1: new high-speed USB device number 8 using xhci_hcd
[99547.419643] usb 5-1: New USB device found, idVendor=0456, idProduct=b673, bcdDevice= 5.15
[99547.419658] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[99547.419661] usb 5-1: Product: PlutoSDR (ADALM-PLUTO)
[99547.419665] usb 5-1: Manufacturer: Analog Devices Inc.
[99547.560743] rndis_host 5-1:1.0 eth0: register 'rndis_host' at usb-0000:c5:00.3-1, RNDIS device, 00:e0:22:ad:c8:3b
[99547.560803] usbcore: registered new interface driver rndis_host
[99547.561308] cdc_acm 5-1:1.3: ttyACM0: USB ACM device
[99547.561345] usbcore: registered new interface driver cdc_acm
[99547.561347] cdc_acm: USB Abstract Control Model driver for USB modems and ISDN adapters
[99547.561347] usb-storage 5-1:1.2: USB Mass Storage device detected
[99547.561714] scsi host0: usb-storage 5-1:1.2
[99547.562012] usbcore: registered new interface driver usb-storage
[99547.566111] usbcore: registered new interface driver uas
[99547.575926] rndis_host 5-1:1.0 enx00e022adc83b: renamed from eth0
[99548.616186] scsi 0:0:0:0: Direct-Access     Linux    File-Stor Gadget 0515 PQ: 0 ANSI: 2
[99548.616731] sd 0:0:0:0: Attached scsi generic sg0 type 0
[99548.617670] sd 0:0:0:0: [sda] 61441 512-byte logical blocks: (31.5 MB/30.0 MiB)
[99548.617869] sd 0:0:0:0: [sda] Write Protect is off
[99548.617875] sd 0:0:0:0: [sda] Mode Sense: 0f 00 00 00
[99548.618131] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[99548.620368]  sda: sda1
[99548.620601] sd 0:0:0:0: [sda] Attached SCSI removable disk

```

VasylSamoilov, 2025-01-03:
@sq6emm OTG port behavior is controlled by kernel when it finally boots, and can be both usb gadget or usb otg, depending on what is connected. Doesn't do much by default.

F5OEO, 2025-01-04:
In progress...
[image]

VasylSamoilov, 2025-01-05:
Great news! Will it support overclock like https://github.com/hz12opensource/libresdr ? Or timestamping to work with LTE srsRAN like https://www.quantulum.co.uk/blog/private-lte-with-plutoplus-sdr/ ?

F5OEO, 2025-01-06:
@VasylSamoilov Overclock OK.
Not planned for timestamping for now. 
First beta seems working around 50Mhz band over network.
[image]

FPSychotic, 2025-01-06:
Did you note this on yours? Is it yours Hamgeek or OpensoureLabs. I want to check mine if have that issue but I don't know what the reviewer.mean. Maybe it's important for your firmware 
[image]

F5OEO, 2025-01-06:
No issue with debug port : just need to use the right one as only one is working : ttyUSB2.
I need to check how it handles on Windows.

[3264968.007636] usb 1-13: new high-speed USB device number 64 using xhci_hcd
[3264968.156015] usb 1-13: New USB device found, idVendor=0403, idProduct=6011, bcdDevice= 8.00
[3264968.156018] usb 1-13: New USB device strings: Mfr=1, Product=2, SerialNumber=0
[3264968.156020] usb 1-13: Product: Quad RS232-HS
[3264968.156020] usb 1-13: Manufacturer: FTDI
[3264968.158449] ftdi_sio 1-13:1.0: FTDI USB Serial Device converter detected
[3264968.158462] usb 1-13: Detected FT4232H
[3264968.158540] usb 1-13: FTDI USB Serial Device converter now attached to ttyUSB0
[3264968.158664] ftdi_sio 1-13:1.1: FTDI USB Serial Device converter detected
[3264968.158673] usb 1-13: Detected FT4232H
[3264968.158749] usb 1-13: FTDI USB Serial Device converter now attached to ttyUSB1
[3264968.158856] ftdi_sio 1-13:1.2: FTDI USB Serial Device converter detected
[3264968.158867] usb 1-13: Detected FT4232H
[3264968.158940] usb 1-13: FTDI USB Serial Device converter now attached to ttyUSB2
[3264968.159047] ftdi_sio 1-13:1.3: FTDI USB Serial Device converter detected
[3264968.159056] usb 1-13: Detected FT4232H
[3264968.159132] usb 1-13: FTDI USB Serial Device converter now attached to ttyUSB3

FPSychotic, 2025-01-06:
Look if it is enumerated is enough to know if it is free of that bias issue that I don't know what it means. I know the overclocked firmware didn't work,.also didn't know how to connect via ethernet.
With pluto firmware and usb is how I have it working, maybe is my fault.

VasylSamoilov, 2025-01-06:
> Did you note this on yours? Is it yours Hamgeek or OpensoureLabs.
 Yep. Mine. Hamgeek. I also ordered opensourcelabs, will see the difference. As for attenuation, I'm talking about this: 
[image]

VasylSamoilov, 2025-01-06:
> @VasylSamoilov Overclock OK. Not planned for timestamping for now. First beta seems working around 50Mhz band over network.
Great news @F5OEO ! I don't mean to implement timestamping and srslte support from scratch, I mean to port them over from fpga/firmware/soapysdr changes that pgreenland made in his repo - he had success with lte up and running.
Soapy fork here: https://github.com/pgreenland/SoapyPlutoSDR/tree/sdr_gadget_timestamping , 
firmware build here: https://github.com/pgreenland/plutosdr-fw/tree/v0.37_timestamp (with hdl , buildroot, uboot linked as git submodules).
All changes documented and links to all repos here: https://www.quantulum.co.uk/blog/private-lte-with-analog-adalm-pluto/#fpga_updates

FPSychotic, 2025-01-06:
> Did you note this on yours? Is it yours Hamgeek or OpensoureLabs.
> Yep. Mine. Hamgeek. I also ordered opensourcelabs, will see the difference. As for attenuation, I'm talking about this: [...]

The  Satsagen's curve doesn't mean many except if you made a new calibration table with a calibrated device as a Adalm Pluto or un Calibrated Noise Source.

VasylSamoilov, 2025-01-06:
> The Satsagen's curve doesn't mean many except if you made a new calibration table with a calibrated device as a Adalm Pluto or un Calibrated Noise Source.

This is understood. I used USRP as source and sink, and while I don't have calibration tables and absolute values doesn't mean much, there is a clear and significant power drop at about 2 GHz on receive line, and at about 1.3 GHz on transmit line, for my specific board. I'll try to do some physical modding and will compare values.

closed 2025-01-07 by sq6emm

VasylSamoilov, 2025-01-07:
I tried booting from sd card by putting content of libresdr 0.0.6 sdimg folder into root of fat32 sd card - it didn't boot. As I have no working debug port it's hard to tell what went wrong. No activity on otg usb port, and no network activity.

F5OEO, 2025-01-07:
Wait for the next build...Hope it will be ok this time !

VasylSamoilov, 2025-01-08:
New build didn't work. Is it overclocked to the max? Because I can boot overclocked firmware up to specific frequency on my device. Maybe it's just bad specimen or just bad luck :(

F5OEO, 2025-01-08:
Oh sorry, yes...this is a version with max overclocked (CPU 1100Mhz and ddr 750Mhz). I will build a "normal" one.

sq6emm, 2025-01-08:
It works on mine...

VasylSamoilov, 2025-01-08:
> Oh sorry, yes...this is a version with max overclocked (CPU 1100Mhz and ddr 750Mhz). I will build a "normal" one.

mine work with cpu900_ddr650 variant quite stable (stress tested it). I think this is as fast as my DDR chips can go with timings.

F5OEO, 2025-01-08:
Just launch a new build process with NO overclock...ready in about 2 hours !

G0MJW, 2025-01-08:
I don’t have a system to do that on.

 

Mike

VasylSamoilov, 2025-01-08:
Yep, this version boots fine. network is fine, usb seems to be ok, but serial port don't have console on it :) probably a bug.

F5OEO, 2025-01-08:
@VasylSamoilov , I assume you have compiled it on your machine ? I didn't check the physical serial port .. Do you speak about serial over otg usb ? If it is the case, I disable console in order to have a standard serial for now (830ec0d209d2a6c3b6f7b6a90dbc11062fb8c93c)

VasylSamoilov, 2025-01-08:
Yes, usb otg. It is still being created, as ttyGS0. I see you point in having standard serial port, console can be an option, I presume. It's just different behavior from all other firmwares.

On a side note, usb keyboard gadget is loading properly:
Starting UDC Gadgets: mkdir: can't create directory '/sys/kernel/config/usb_gadget/composite_gadget/functions/hid.keyboard': No such file or directory
sh: write error: Invalid argument
/etc/init.d/S23udc: line 139: can't create /sys/kernel/config/usb_gadget/composite_gadget/functions/hid.keyboard/protocol: nonexistent directory
/etc/init.d/S23udc: line 140: can't create /sys/kernel/config/usb_gadget/composite_gadget/functions/hid.keyboard/subclass: nonexistent directory
/etc/init.d/S23udc: line 141: can't create /sys/kernel/config/usb_gadget/composite_gadget/functions/hid.keyboard/report_length: nonexistent directory
/etc/init.d/S23udc: line 145: can't create /sys/kernel/config/usb_gadget/composite_gadget/functions/hid.keyboard/report_desc: nonexistent directory

Positive news: usb audio gadget works perfectly.
From USB host standpoint, usb audio adapters (usb audio class devices) work! also mass storage works, as well as usb hubs.

VasylSamoilov, 2025-01-08:
it would be nice to add bluetooth drivers and bluetooth utils into firmware - a single low power bluetooth adapter can be an audio card, serial port and set of input devices.

FPSychotic, 2025-01-09:
> > Look if it is enumerated is enough to know if it is free of that bias issue ? that written in the review if unconnected bias I don't know what it means. I know the overclocked firmware didn't work for me , only his non OC version,.also didn't know how to connect via ethernet. With pluto firmware and usb is how I have it working, maybe is my fault.
> 
> yes, that's how mine responds: [ 2749.868592] usb 3-3: new full-speed USB device number 3 using xhci_hcd [ 2749.996611] usb 3-3: device descriptor read/64, error -71 [ 2750.232575] usb 3-3: device descriptor read/64, error -71 [ 2750.468543] usb 3-3: new full-speed USB device number 4 using xhci_hcd [ 2750.596558] usb 3-3: device descriptor read/64, error -71 [ 2750.832536] usb 3-3: device descriptor read/64, error -71 [ 2750.940607] usb usb3-port3: attempt power cycle [ 2751.352492] usb 3-3: new full-speed USB device number 5 using xhci_hcd [ 2751.352675] usb 3-3: Device not responding to setup address. [ 2751.560701] usb 3-3: Device not responding to setup address. [ 2751.768482] usb 3-3: device not accepting address 5, error -71 [ 2751.896479] usb 3-3: new full-speed USB device number 6 using xhci_hcd [ 2751.896690] usb 3-3: Device not responding to setup address. [ 2752.104652] usb 3-3: Device not responding to setup address. [ 2752.312471] usb 3-3: device not accepting address 6, error -71 [ 2752.312604] usb usb3-port3: unable to enumerate USB device

My LibreSDR is not new or recent is from 6-8 momths ago . It looks there was a Series1, some used to hack DJI video system with a firmware (if someone has info about this would love know more) . Later other batch was released, the mine, it looks mine cannot flashed with DJI firmware,except if I bough a 1st batch, but I dont think so. This is the report, mine is not hangeek nor openlabsource. It is one from TZT shop , the shop no is selling Hamgeek, but maybe the exactly the same than mine if they didn't sell all the units. Maybe they just changed the name in the listing but the hardware is the same I bought. Impossible to know.

[image]

francistheodorecatte, 2025-01-12:
physical ethernet works on mine (albeit with a long delay before getting a DHCP address), but actually connecting to the LibreSDR over iio causes some bizarre jumbo-frame-esque packet mangling, where packets are somehow way longer than they should be.

these are what should be the same packet coming from the LibreSDR while probing it w/ SoapySDRUtil, side-by-side in two wireshark windows. the left is working, using the usb ethernet gadget, while the right uses the physical ethernet interface. in both cases, each end has its MTU set to 1500. notably, the failure is with a cable going from the LibreSDR to my laptop directly, without any router or switch inbetween, with switch nor router seeming to affect this behavior:
[image]

every packet in both captures prior to this one are <100 bytes in length with an MTU of 1500, and have identical contents.

the only bit of customization from the base image that I've done is enabling both CPU cores by unsetting maxcpus in uboot, and enabling 2r2t.

VasylSamoilov, 2025-01-12:
@francistheodorecatte you source and destination addresses also look like packet is being routed. are you sure that you are on the same physical subnet and ip routing doesn't do something strange and configured properly? these amd zynq soc physically don't have jumbo frame capability, so whatever is assembling these packets for you - is not libresdr.

francistheodorecatte, 2025-01-12:
@VasylSamoilov the screenshot is from when I first stumbled upon this problem; I thought the same thing, that it was probably my router or switch causing this behavior. but between using a dumb switch, and the last time I tried it with a direct copper connection between my laptop and the LibreSDR, the exact same thing happens w/ the jumbo-esque frames. I ran out of time to get clean packet captures; I'll grab those later and post them up.