F5OEO/tezuka_fw #14: Enable external reference clock

sq6emm, 2025-01-06:
If possible, please enable external reference to be possible to be used.

FPSychotic, 2025-01-06:
I have GPSDO, SI5351 device  and Hackrf output to help to test that if needed

F5OEO, 2025-01-07:
This request is for libresdr

VasylSamoilov, 2025-01-09:
I also have GPSDO, can help with testing. It's just unclear how to set refclock to external and how to read status is it locked on external clock or not...

F5OEO, 2025-01-09:
This is done by modifying the devicetree in u-boot-env : I plan to rewrite the UbootEnv.txt to be universal

sq6emm, 2025-01-11:
I am ready for testing;P mmcx to sma jumper arrived;P

lubuntu-l100, 2025-01-18:
I have tested the LibreSDR-Tezuka firmware of January 16 2025 with SDR Console v3.4 beta. SDR Console v3.4 is now running fine in RX as well as TX over the QO100.

I am waiting for the enablement of the 10 Mhz external clock, due to the fact that the internal VCTTXO is not on frequency and is drifting a littlebit during transmission on the QO100. Locking on 10 Mhz is a need during qso's. I am following the github discussion around the Renovate of the original HDL in LibreSDR and the incorporation in the LibreSDR Tezuka firmware.

I am able to test this configuration with 10 Mhz and the Libresdr due to the fact that I have running already several Pluto's Rev B for DATV and measurement activities with GPSDO's and OCXO's since 2019. 

LibreSDR with 10 Mhz supported could be a good and cheap alternative for the AD Pluto's.

lubuntu-l100, 2025-01-18:
I forgot my callsign in the previous post. My callsign is PA2JSA, see my QRZ.com pages.

sly547, 2025-01-22:
> I am following the github discussion around the Renovate of the original HDL in LibreSDR and the incorporation in the LibreSDR Tezuka firmware.
Sorry to hijack but I can't seem to find that discussion (too many places to check, I'm lost). Could you give me a pointer please?

Also I wonder if the currently unused DAC that controls the VCTCXO can help with the drift. Without a reference the LibreSDR cannot measure the drift and thus not regulate the frequency, right?

lubuntu-l100, 2025-01-22:
Hello Michael,

 

Some roughly explanation how the 10 Mhz needs to work. The 10 Mhz signals is connected to a driver IC and then connected to the FPGA. The 7z020 FPGA compares the 10 Mhz signal with the 40 Mhz from the VCTCXO. The differences in frequency is going to the DAC that corrects the VCTCXO. The output of the 40 Mhz is divided in 2 signals, 1 is going to the AD9363 and the other is going the FPGA.

 

So the frequency correction is completely done outside the AD9363. So defining an external clock is not working due to the fact that the external clock is related to the AD9363 and not the way the LibreSDR is doing frequency correction.

 

Below some links where this discussion is going.

 

https://github.com/F5OEO/tezuka_fw/issues?q=state%3Aopen%20label%3A%22libresdr%22

 

https://github.com/sdy623/hdl

 

https://github.com/sdy623/hdl/commit/5b9bcef97dec9202b1c5d43871a1ca2c543c0460

 

https://github.com/maia-sdr/plutosdr-fw

 

Hope this will help you.

 

73 Jaap

 

PA2JSA

 

 

Call Sign: PA2JSA

Operator: Jaap Schekkerman

Location: JO22QE

Country: The Netherlands

Email:  ***@***.*** ***@***.***>  

__________________________IMPORTANT NOTICE_______________________

The content of this email and any attachments are confidential and intended for the named recipient(s) only. If you are not the intended recipient of this e-mail, any dissemination, distribution or copying of this e-mail, and any attachments thereto, is strictly prohibited. If you have received this e-mail in error, please immediately notify the sender by e-mail or by telephone and permanently delete all copies of this e-mail and any attachments. The name of the sender at the bottom of this message or contained anywhere else in this e-mail is only for identification and is not intended to and shall not act as a signature indicating agreement or the formation of a contract unless that is specifically stated in the body of the e-mail message. 

WARNING: Although PA2JSA has taken reasonable precautions to ensure no viruses are present in this email, PA2JSA cannot accept responsibility for any loss or damage arising from the use of this email or attachments.

sly547, 2025-01-22:
Hi!

> Some roughly explanation how the 10 Mhz needs to work. The 10 Mhz signals is connected to a driver IC and then connected to the FPGA. The 7z020 FPGA compares the 10 Mhz signal with the 40 Mhz from [...]

Ahh right, the AD9363 gets the 40 MHz directly from the oscillator, not through the FPGA, so we can't just multiply the external 10 MHz in the FPGA and route that to the AD9363. Thanks for explaining!

Are you aware of another PlutoSDR variant that is already doing this correction with the FPGA? I checked Pluto rev.D and Pluto+, they use the external frequency directly on the AD9363.

> Below some links where this discussion is going.

Thanks! I need to check maia-sdr, this I did not really have on my radar yet. I thought there might be a single place where this discussion is happening, but then it is still scattered. I started one myself here: https://github.com/F5OEO/tezuka_fw/discussions/42

lubuntu-l100, 2025-01-22:
Hi Michael,

The Libresdr is a clone of the Ettus B210 and not a clone of the Adalm Pluto.

The Ettus software supports the external 10 Mhz, the Pluto cloned software does not have supported for that.

F5OEO got the routines from the Ettus B210 software and need to combine these with the pluto software in new to develop firmware for the Libresdr Tazuka project.

I use at this time the fw from Tazuka v3e31 from last week that is running fine on my libresdr. F5OEO has announced to incorporate the 10 Mhz external clock in upcomming versions.

I use the Libresdr together with ad plutos for communication over the QO100 satellite. Tx at 2.4 Ghz, Rx at 10 Ghz with a downconverter. Libresdr is 5.8 khz off frequency at 2.4 Ghz and drift around 500 hz.

So F5OEO is working on a solution.

73 Jaap
Pa2jsa
⁣BlueMail voor Android downloaden ​

sly547, 2025-01-23:
Hello Jaap,

Thank you for clarifying some things. These devices are all new to me. I started looking into the [Ettus repository](https://github.com/EttusResearch/uhd) a bit. It's great to find all of this as open source.

Communicating over QO100 looks like a good motivation to get the external clock working. Really nice that this QO100 is a thing now. I don't have a clear target application yet, but maybe something related to (mobile) telecom networks (see OSMOCOM / srsRAN). Also something with low drift requirements. But as I said, I'm new to this.

I will try to get the DAC working on my board, so I can control it from software and do simple tests.

Notupus, 2025-02-11:
With the new commit apparently there is support for pps and 10mhz on tezuka now, but I dont know how to enable it since the led stays off when using my "gpsdo" which is the timepulse from a ublox gps receiver, which is a pps or 10mhz square wave . Not sure if that can work or be adapted to work.

F5OEO, 2025-02-11:
This is a very early try thanks to discussion with https://github.com/hz12opensource/libresdr/issues/10#issuecomment-2472162424
It should be automatic as soon as you have a 10Mhz, >2.5vpp.
I don't have opportunity to test right now.

sq6emm, 2025-02-11:
Testing it right now...
But how can I detect if it is fixed to my reference?:)
Any reaction I should expect from LED or some /sys file content?

lubuntu-l100, 2025-02-12:
When your fixed reference is connected at start up, de blue led of 10 mhz  or pps should blink permanently when  de firmware support the 10 Mgz reference in the firmware. When at startup de 10 mhz led blink shortly and the fpga led start blinking permanently, there is no support for the external reference.

The 10 Mhz external connection goes to the Z7020 fpga where the 10 mhz is compared with the 40 mhz vtcxo in a pll circuit. A dac is controlling the 40 mhz vtcxo. The output of the 40 mhz is splitup and buffered. 1 signal is going back to the clock input of the ad9363 and the other part is going to the fpga.

So the external 10 mhz is not connected to the external clock of the ad9363.

This way of synchronisation is based on the firmware of the Ettus B210.

⁣BlueMail voor Android downloaden ​

lubuntu-l100, 2025-02-12:
[image]

Zyncsdr Layout and led function explanation. Chinese Source.

PA2JSA

sq6emm, 2025-02-12:
It blinks/lights during the startup, but later not.

lubuntu-l100, 2025-02-12:
Hi David, that means that the firmware used do not support the external 10 Mhz function.

I have tested the Libresdr with fw for the B210 and than the 10 Mhz led is blinking all the time. However that fw was not supporting the Z7020 fpga.

⁣BlueMail voor Android downloaden ​

sq6emm, 2025-02-12:
Hey, well yes most probably.
I am using master branch of tezuka_fw - just reporting the state;) But I think it might be also connected with no LEDs are blinking during the firmware update... so maybe its just different pins are LEDs...

lubuntu-l100, 2025-02-12:
Hi David I use the F5OEO Tezuka v0.0.95 firmware that is running fine ftom the Sd card  except the support of the 10 Mhz external clock. However the same fw  0.0.95 loaded and booted from flash has a problem during Tx. I use the Libresdr on the QO100 satellite in Nb. During tx over the satellite I have a broad spectrum of spikes in the signal. Booting from the Sd card, the tx signal over the QO100 is clean. So there is something different between booting from sd or booting from flash.

⁣BlueMail voor Android downloaden ​

Notupus, 2025-02-12:
. I have a 10mhz at maybe 3 volts and 1pps at 3.3V. I do no see a difference in the tx frequency at 1.6ghz going to a rtl-sdr with or without the reference, it seems to not work, however turning the led on an off at whatever the vctcxo is being controlled with would make sense, and I would prefer PPS since I can measure that with a simple multimeter.  I expect to see a small jump in frequency when plugging in the ref clock.

lubuntu-l100, 2025-02-12:
What do you mean with turning the led on and off?

I have tested the vctcxo at 2.4 Ghz over the satellite. The result is that the 2.4 Ghz is 5.6 khz of frequency compared to a gpsdo locked receiver. Even so the vctcxo is drifting a 500 hz during warming up.

⁣BlueMail voor Android downloaden ​

Notupus, 2025-02-12:
Okay, so you are claiming with tezuka_fw 0.1.0 and 10mhz input you can see the tuning? If so I will try to confirm this in due course maybe it doesnt like thats is a 10mpps signal.

lubuntu-l100, 2025-02-12:
Tezuka v0.1.0 does not support the 10 Mhz external reference at this time.

Booted from sd card v0.1.0 is working fine with Sdr Console v3.4 beta, transmitting and receiving in full duplex over the QO100 satellite. Tx at 2.4 Ghz, Rx at 10 Ghz.

⁣BlueMail voor Android downloaden ​

lubuntu-l100, 2025-02-13:
The Link below on Github is a description of how to implement the 10 Mhz and 1PPS External clock reference in the Libresdr. Header is Renovate the original HDL in LibreSDR from SDY623.

I hope F5OEO can use these scripts as input for implementing the 10 Mhz and 1 PPS in the TAZUKA Firmware.

Link: https://github.com/sdy623/hdl/commit/5b9bcef97dec9202b1c5d43871a1ca2c543c0460#diff-6b183c60e27cdb85d400546826482fe3e196d85e3abfe0e56d000736af943d7c

PA2JSA

lubuntu-l100, 2025-02-17:
**Hardware Modification: 40 Mhz External Reference Clock for AD9363 in ZyncSDR / LibrSDR**

The Zyncsdr / Libresdr has a 10 Mhz external reference to lock the 40 Mhz internal VCTCXO via the FPGA 7Z020, however to support for this 10 Mhz synchronization is not available in current Tezuka Firmware for the Zyncsdr. With my experiences to modify the internal VTcxo of Adalm Pluto’s and replace that with an external 40 Mhz reference from a Leobodnar GPSDO. I decided to test if I can do the same with the Zyncsdr with the Tazuka FW 0.1.0.
Looking at the hardware diagram, you can see that the 40 Mhz VCTCXO is controlled by a PLL circuit in the FPGA 7Z020 where the 10 Mhz external reference is compared with the 40 Mhz of the VCTCXO. Differences in frequency are converted by a DAC5311 into a voltage that is steering the 40 MHZ VCTCXO.
The output of the 40 Mhz oscillator is buffered by a SN74AHC1G14DBVR and then split in two 40 Mhz signals and again buffered. 1 Signal is going back to the FPGA 7Z020 and the other part of the 40 Mhz signal is going to the AD9363 as 40Mhz internal clock signal.
At this point I decided to modify the 40 Mhz signal that is going to the AD9363 and replace that with a 40 Mhz signal from a 40 Mhz Leobodnar GPSDO signal.

In attached document:
[ZyncSDR 40 Mhz External Reference.pdf](https://github.com/user-attachments/files/18815126/ZyncSDR.40.Mhz.External.Reference.pdf)

you can find how to modify the ZyncSDR to use an external 40 Mhz Reference signal for the AD9363. This hardware modification is equal to simular modifications in Adalm Pluto SDR transceivers. Test with the Tezuka 0.1.0 Firmware at the QO100 Satellite shows no difference between an AD Pluto or the Zyncsdr in terms of Stability and on Frequency.

Evenso at QRZ.com you can find the modification at PA2JSA page.

PA2JSA

F5OEO, 2025-02-17:
Thanks for this hardware mod @lubuntu-l100 . Could be useful until the 10Mhz input could work directly (work in progress - not trivial).

lubuntu-l100, 2025-02-17:
Hi Evariste,

Thanks for your reply.

This hardware mod. for the 40 Mhz of the AD9363 is working fine with the Tezuka fw 0.1.0.

At the time the 10 Mhz external clock is supported by the firmware I will do a comparison test between ext. 40 Mhz ref clock and the 10 Mhz ext. ref clock.

I will follow your firmware developments and will contribute to it where possible.

73 Jaap

PA2JSA

⁣BlueMail voor Android downloaden ​

sq6emm, 2025-02-19:
So, I have made a test on original LibreSDR from opensourcesdrlabs with their original firmware on the flash.
I have connected -10dBm 10MHz signal source and LED under 10M immediately beacome BLUE ;)
So ... -10dBm signal should be enough! My source of 0dBm also works, and tested with generator untill +10dBm also still works. Below -10dBm no lock.

[image]

F5OEO, 2025-02-19:
Before plugin reference source, are led off or blinking ?

sq6emm, 2025-02-19:
Before reference source PPS blue LED is blinking (twice a second). 10M LED is off.

sq6emm, 2025-02-19:
And the same led is used by the flashing process (update.sh led0:green).

F5OEO, 2025-02-20:
Story continue : it seems that VCTXO is not a real one on my Hamgeek (as mentionned https://github.com/F5OEO/tezuka_fw/issues/45#issuecomment-2635769660). I will build a firmware with a test to see if the frequency change with vctxo is working.

F5OEO, 2025-02-26:
@sq6emm it could be very interesting that I could test the firmware from opensourcelabs. We need to extract it from qspi and save it to sd for example or in ram and then transfer them over ssh.

sq6emm, 2025-02-26:
@F5OEO happy to help, but how?:) I still have not flashed the opensourcelabs so its possible;)

F5OEO, 2025-02-26:
I need to make a script to extract the firmware. As far as I know, it is the only one which is working with rf clck

sq6emm, 2025-02-26:
Please remember that the original firmware is Linux (lubuntu) based....

sq6emm, 2025-02-26:
dd of=boot.bin if=/dev/mtdblock0 bs=64k
dd of=u-boot-env.bin if=/dev/mtdblock1 bs=64k
dd of=firmware.frm if=/dev/mtdblock3 bs=64k

should do the job?

F5OEO, 2025-02-26:
Exactly !

sq6emm, 2025-02-26:
[firmware.frm.gz](https://github.com/user-attachments/files/18996476/firmware.frm.gz)

[u-boot-env.bin.gz](https://github.com/user-attachments/files/18996480/u-boot-env.bin.gz)

[boot.bin.gz](https://github.com/user-attachments/files/18996481/boot.bin.gz)

sq6emm, 2025-02-26:
> Please remember that the original firmware is Linux (lubuntu) based....

I was wrong with it. This is firmware on the sd card.
But the one on flash is:

```
______ _       _        _________________
| ___ \ |     | |      /  ___|  _  \ ___ \
| |_/ / |_   _| |_ ___ \ `--.| | | | |_/ /
|  __/| | | | | __/ _ \ `--. \ | | |    /
| |   | | |_| | || (_) /\__/ / |/ /| |\ \
\_|   |_|\__,_|\__\___/\____/|___/ \_| \_|

v0.37-dirty
https://wiki.analog.com/university/tools/pluto
```

F5OEO, 2025-02-26:
Wow....Fantastic ! Will give it a try tomorrow !

sq6emm, 2025-02-26:
```
# cat /opt/VERSIONS /etc/os-release 
device-fw v0.37-dirty
hdl dev_prj_2018_r1-1763-ge5660-dirty
buildroot 2021.02.7-adi-6643-g0d5db4-dirty
linux adi-xilinx-2020.1-98725-g3eae70-dirty
u-boot-xlnx v0.20-PlutoSDR-25-g90401c-dirty
NAME=Buildroot
VERSION=-g0d5db4a280-dirty
ID=buildroot
VERSION_ID=2022.02.3
PRETTY_NAME="Buildroot 2022.02.3"
```

sq6emm, 2025-02-26:
This might help!

[https://gitee.com/zyq-77/hdl/tree/dev_prj_2018_r1](url)

F5OEO, 2025-02-26:
This is an hdl repo, but where is the buildroot project !

F5OEO, 2025-02-27:
Experimental version using 10Mhz (de18b11869c5e5eef4783ac7899cd0a63f9bfc67)
Put a 10Mhz source with at least 7dm, you should see led 10mhz ON.
If you have chance to have a vctxo (zynqsdr), it should pilot it. Not on my hamgeek libresdr.
Workaround : 
Use a script /root/adj_10mhz.sh which should calculate and set xo_correction. (Need to check if the sign is ok).
Feedback welcomed !

F5OEO, 2025-02-27:
And seems that chrony adjustement is not far from value calculated by 10mhz ref in.
# ./adj_10mhz.sh 
Xo Correction 40000042
Xo Correction 40000044
Xo Correction 40000042
Xo Correction 40000044
Xo Correction 40000042
^C
# ./adj_chrony.sh 
High accurate ppm = 1.249 +/- 0.385 slow
Correct 40000050
High accurate ppm = 1.249 +/- 0.385 slow
Correct 40000050

sq6emm, 2025-02-27:
```
  Downloaded 23 crates (12.4 MB) in 4.08s (largest was `aws-lc-sys` at 8.3 MB)
error: rustc 1.79.0 is not supported by the following packages:
  home@0.5.11 requires rustc 1.81
  litemap@0.7.5 requires rustc 1.81
  native-tls@0.2.14 requires rustc 1.80.0
  native-tls@0.2.14 requires rustc 1.80.0
  native-tls@0.2.14 requires rustc 1.80.0
  zerofrom@0.1.6 requires rustc 1.81
Either upgrade rustc or select compatible dependency versions with
`cargo update <name>@<current-ver> --precise <compatible-ver>`
where `<compatible-ver>` is the latest version supporting rustc 1.79.0

>>> maia-httpd sweep Building
xz: /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/build/maia-httpd-sweep//maia-httpd/target/armv7-unknown-linux-gnueabihf/release/maia-httpd: No such file or directory
>>> maia-httpd sweep Installing to target
/usr/bin/install -D /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/build/maia-httpd-sweep//maia-httpd/target/armv7-unknown-linux-gnueabihf/release/maia-httpd.xz /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/target/usr/bin/
/usr/bin/install: cannot stat '/home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/build/maia-httpd-sweep//maia-httpd/target/armv7-unknown-linux-gnueabihf/release/maia-httpd.xz': No such file or directory
make[1]: *** [package/pkg-generic.mk:380: /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/build/maia-httpd-sweep/.stamp_target_installed] Error 1
make: *** [Makefile:83: _all] Error 2
```

F5OEO, 2025-02-27:
You need to migrate to latest buildroot : wget https://buildroot.org/downloads/buildroot-2024.11.1.tar.gz

sq6emm, 2025-02-27:
> You need to migrate to latest buildroot : wget https://buildroot.org/downloads/buildroot-2024.11.1.tar.gz

Yeah my mistake;) Thanks, on it.

sq6emm, 2025-02-27:
I dont have time to test more today, but LED is not ON as on original firmware;)

F5OEO, 2025-02-28:
Could be a good idea to test with 1 pps also..I will try to use a low cost GPS pour that. Don't know if 3.3V pulse is not too much

F5OEO, 2025-02-28:
Yes...

Notupus, 2025-02-28:
Well no led and adj 10mhz is not working

F5OEO, 2025-02-28:
Can you uncomment this line (directly with nano under /root/adj_10mhz.sh) https://github.com/F5OEO/tezuka_fw/blob/696b9be11b2047b6960104ab21edc12a6d6c5ee1/board/pluto/overlay_maia/root/adj_10mhz.sh#L5

And give me what it outputs...

sq6emm, 2025-02-28:
> [firmware.frm.gz](https://github.com/user-attachments/files/18996476/firmware.frm.gz)
> 
> [u-boot-env.bin.gz](https://github.com/user-attachments/files/18996480/u-boot-env.bin.gz)
> 
> [boot.bin.gz](https://github.com/user-attachments/files/18996481/boot.bin.gz)

@F5OEO based on those files... you think this version of software might be needed running yet for any tests yet or I can flash it to tezuka?:)

F5OEO, 2025-02-28:
I began to analyse it...but not yet compelety extracted. It is precious..keep it on the flash, and test tezuka on sd !

sq6emm, 2025-02-28:
> I began to analyse it...but not yet compelety extracted. It is precious..keep it on the flash, and test tezuka on sd !

Yes Sir!

Notupus, 2025-02-28:
> Can you uncomment this line (directly with nano under /root/adj_10mhz.sh)
> 
> [tezuka_fw/board/pluto/overlay_maia/root/adj_10mhz.sh](https://github.com/F5OEO/tezuka_fw/blob/696b9be11b2047b6960104ab21edc12a6d6c5ee1/board/pluto/overlay_maia/root/adj_10mhz.sh#L5)
> 
> Line 5 in [696b9be](/F5OEO/tezuka_fw/commit/696b9be11b2047b6960104ab21edc12a6d6c5ee1)
>  #echo "Reg $read_register" 
> 
> And give me what it outputs...

WIth PPS the register reads as such
# devmem 0x41210000
0x00000008
10MHZ square wave from ublox( probably weak but I have the opensourcelabs version)
# devmem 0x41210000
0x0000000A
and nothing
# devmem 0x41210000
0x00000008
Maybe you can use a radiosonde to do pps for testing.

[13 more replies]