F5OEO/tezuka_fw #16: Fail flash when use update_frm.sh

crackerjacques, 2025-01-08:
Hello.
I am enjoying Tezuka Firmware forLibreSDR. Everything is currently working without a hitch when booted from SDCard.

However, if I send pluto.frm via SCP or other means and write it to the internal disk from update_frm.sh, it did not work.
Is it correct to say that this is not yet supported?

I also does not know how to write tezuka firmware to a blank SD card.

Best regards.

sq6emm, 2025-01-08:
I am unable to update boot.frm either from mounted storage nor cli...

```
# update_frm.sh ./pluto.frm 
423+1 records in
423+1 records out
Done
# 
# update_frm.sh ./boot.frm 
Failed
```

F5OEO, 2025-01-08:
Until I add update_boot.sh , it is not possible to write boot.frm on flash except by using usb (put boot.frm in virtual file mounted, then eject).

crackerjacques, 2025-01-09:
Thank you very much.
I am so supportive I want to buy you a coffee. I can only offer you virtual coffee now, but I support you.☕