F5OEO/tezuka_fw #23: Make maia startup optional

VasylSamoilov, 2025-01-09:
If maiasdr is exposed to any local network, things like homeassistant can open port 8000 to scan network device and, a result, get maiasdr running radio, which results in iio interface behaving in a strange way.
One of the options is to make maiasdr startup an option, another is to make maia interface protected by password, or at least require a button press to start signal acquisition.

F5OEO, 2025-01-09:
Maia is working in parallel and thus does not disturb iio. So accessing it is not an issue, isnt it ?

VasylSamoilov, 2025-01-09:
well, not exactly. Several time I had an issue that I'm receiving just noise on a pc, with no signal at all. Probably maia shifting frequency or uploads something to fpga that conflicts with whatever pc is requesting via iio.

F5OEO, 2025-01-09:
Maia could change eventually frequency if it is the first time your browser is connected to ip of your sdr. After that a cookie is set not to disturb and set any variable. Need to double check this.

VasylSamoilov, 2025-01-13:
I am also not sure about root cause, but even in home/local network virtually anything can make a request via HTTP to port 8000, just as a part of auto-discovery. Prevent maia from firing up on first request to front page is a minimum precaution for predictable behavior. Something like "start radio" button. More advanced mechanism would be login page or optional service startup.

helmarw, 2025-03-25:
i just flashed the fw 0.1.1 and trying some things.
on thing i notices maia doenst seem to be running
no port 8000 is exposed
do i need to start it manually ?

helmarw, 2025-03-25:
mhm strange, after an other reboot a lot more ports came online also 8000
its working now, never mind then :)