F5OEO/tezuka_fw #24: Add timestamps

BloodSqueezer, 2025-01-11:
Hi!
Is it possible to add timestamps like Phil Greenland did?
https://github.com/pgreenland/plutosdr-fw/tree/v0.38_plutoplus_timestamp
https://www.quantulum.co.uk/blog/private-lte-with-plutoplus-sdr/

VasylSamoilov, 2025-01-12:
I think it's not just about timestamp, it's about software that can be used with tezuka firmware. Definetly, running private lte is a big step for SDR platform. pgreenland also had a look at osmo-trx ( https://github.com/pgreenland/plutosdr-fw/issues/3 ) but still didn't have time to deep dive and make it work (osmo-trx is sdr component for 2G base station).
some other software that would be nice to have working is osmocom-analog (0g/1g communication stack), openwifi (which libresdr was designed to be running, I suppose), hacktv to transmit analog tv (right now it has bandwidth issues). 

Probably I should start a documentation/howto section what is working and what is not, and why.

Notupus, 2025-04-14:
So about 2 months later I have booted an image that sends timestamps(in an odd format, which I will have to debug) from my libresdr to soapy using the [quantulum](https://www.quantulum.co.uk/blog/private-lte-with-plutoplus-sdr/) repo. This is not a really easy fight. Currently usb is completely broken and ethernet works without timestamping and it shows an lte like signal but no ue has found it yet. With timestamping I am getting some odd dropped out of order datagram warnings and no tx is done( no rx either) (see https://gist.github.com/Notupus/ab197968631d2e4822f22bb767093fb4 for the error)
I have but not a clue as to what to try next to get this working on a known good platform to try to port it into tezuka. If anyone has figured timestamping on libre maybe I could be given a hint.

F5OEO, 2025-07-22:
Adding timestamps to run LTE could be effectively possible, but as you understand well, I need to work on other activities to pay my life. I would be happy if I can collect money to work on. For now I doubt about community money, as even hundreds of people use tezuka, nearly nobody has made some donation.

jojoosinga, 2025-07-22:
Hi thanks for the quick response. If I would donate 150-200€ would that make a difference?

jojoosinga, 2025-09-22:
Hey @F5OEO Have you thought about the offer? Thanks

F5OEO, 2025-09-22:
Thanks for your offer. If this feature is of interest and could be sponsored by several persons, I could work on it. Sponsor goal around 600$.