F5OEO/tezuka_fw #27: IPv6 Adress Issue

crackerjacques, 2025-01-12:
It may be better to create such a configuration file in some cases(e.g. SDR++, ), as the PlutoSDR plug-in for some SDR software does not support the handling of IPv6 addresses or may cause crashes.

```
# tezuka_fw/board/pluto/overlay_maia/etc/sysctl.d/40-ipv6.conf

net.ipv6.conf.all.disable_ipv6 = 1
net.ipv6.conf.default.disable_ipv6 = 1
net.ipv6.conf.lo.disable_ipv6 = 1
```
And then turn it on and off with a shell script.

```
#!/bin/sh

enable_ipv6()
{
    echo 0 > /proc/sys/net/ipv6/conf/all/disable_ipv6
}

disable_ipv6()
{
    echo 1 > /proc/sys/net/ipv6/conf/all/disable_ipv6
}

if [ "$1" = "1" ] ; then
    disable_ipv6
elif [ "$1" = "0" ] ; then
    enable_ipv6
fi
```

F5OEO, 2025-01-12:
OK to disable ipv6 for some cases. Can you elaborate which software is crashing with ipv6 ?

crackerjacques, 2025-01-12:
PlutoSDR Plugin for SDR++.

VasylSamoilov, 2025-01-12:
disabling ipv6 for whole platform due to plugin for a single client may be not the best way to do it.

F5OEO, 2025-01-12:
Please use the sdrpp fork wich is not buggy and extended feauturs (CS8 and rx1/rx2 selection) : https://github.com/F5OEO/SDRPlusPlus

crackerjacques, 2025-01-13:
> disabling ipv6 for whole platform due to plugin for a single client may be not the best way to do it.
[plutosdr_source.tar.gz](https://github.com/user-attachments/files/18394494/plutosdr_source.tar.gz)

I was thinking about that myself and ended up Tinkering with the PlutoSDR Plugin.

ashjmckenzie, 2025-01-18:
I'm on a mac I also can confirm SDR++ (Fork version or non Fork) Crashes when running Tezuka (LibreSDR) if I open the app before connecting the SDR, then connect ethernet, I can tap refresh and sometimes the SDR appears.

F5OEO, 2025-03-30:
Ipv6 disable by default. 2c245d9a4d89ed780d645bca2036b8552c41e14f

closed 2025-03-30 by F5OEO