F5OEO/tezuka_fw #29: dhcpc client works only at boot

sq6emm, 2025-01-13:
If Ethernet cable is not connected at boot and connected later on.. IP address will not be assigned to an interface.
DHCP client works only at boot. Should work always imo?

F5OEO, 2025-01-13:
DHCP client is running but it is caused by avahi-autoipd.action which set a 169.254.255.255 ip address when ,ip is first found (link not here). Need to remove it when link is up again. Hope find a clean solution.

VasylSamoilov, 2025-01-13:
udhcpcd (part of busybox) can't run indefinitely in background. It's one-shot binary. See buildroot start scripts here: https://github.com/buildroot/buildroot/blob/master/package/busybox/udhcpc.script

Solution is to remove udhcpd and to switch to more feature-rich dhcpcd package (it can run in background as daemon). It will also provide more dhcp features.

VasylSamoilov, 2025-01-13:
@F5OEO please remove a bug label, it's not a bug, it's a feature (of busybox/udhcpc)

VasylSamoilov, 2025-01-13:
If full linux is required I would recommend to install kuiper linux (raspbian from analog devices) and use tezuka boot, kernel and device tree to boot normal Linux environment.

F5OEO, 2025-01-31:
Should be fixed with 3a7dfdd1b322643e565f7b0a0e61319ab05d1678
Please confirm.

sq6emm, 2025-01-31:
I confirm, works fine now. Thx.

F5OEO, 2025-01-31:
Wonderfull ! Thanks for confirming. Time to make a new build I guess.

closed 2025-01-31 by F5OEO