F5OEO/tezuka_fw #30: USB bluetooth (audio) support

VasylSamoilov, 2025-01-13:
I have (limited) success with usb bluetooth support via otg port.

I have successfully started it on kuiper linux with tezuka kernel, but buildroot is still missing some packages to get it to work.
added to defconfig:
```CONFIG_BT=y
CONFIG_BT_BREDR=y
CONFIG_BT_HS=y
CONFIG_BT_LE=y
CONFIG_BT_BT_MSFTEXT=y

CONFIG_BT_HCIVHCI=y
CONFIG_BT_HCIUART=y
CONFIG_BT_HCIBCM203X=y
CONFIG_BT_HCIBPA10X=y
CONFIG_BT_HCIBFUSB=y
CONFIG_BT_HCIDTL1=y
CONFIG_BT_HCIBT3C=y
CONFIG_BT_HCIBLUECARD=y

CONFIG_BT_HCIBTUSB=y
CONFIG_BT_HCIBTSDIO=y

CONFIG_BT_INTEL=y
CONFIG_BT_ATH3K=y
CONFIG_BT_MRVL=y
CONFIG_BT_MRVL_SDIO=y
CONFIG_BT_MTKSDIO=y
CONFIG_BT_MTKUART=y
CONFIG_BT_QCOMSMD=y
CONFIG_BT_BCM=y
CONFIG_BT_RTL=y
CONFIG_BT_QCA=y
CONFIG_BT_MTK=y
```

next, packages for bluez are required. 
theoretically, this should do it:
```select BR2_PACKAGE_BLUEZ_TOOLS
select BR2_PACKAGE_BLUEZ5_UTILS
select BR2_USE_MMU
select BR2_USE_WCHAR
select BR2_TOOLCHAIN_HAS_THREADS
select BR2_PACKAGE_BLUEZ_ALSA
select BR2_PACKAGE_BLUEZ5_UTILS_HEADERS
select BR2_PACKAGE_BLUEZ5_UTILS_CLIENT
select BR2_PACKAGE_BLUEZ5_UTILS_TOOLS
```
but I end up with "bt-.." tools that kinda work when bluetoothd is started, but bluetoothctl is missing. 

to get bluetooth audio working (under kuiper linux) I compiled kernel image with bluetooth support, installed bluez, configured it and installed https://github.com/arkq/bluez-alsa backend. 

Then bluetooth device should be paired. `aplay -D bluealsa file` or `mpg123 -a bluealsa file` should work.

F5OEO, 2025-01-13:
bluetoothctl should be installed with BR2_PACKAGE_BLUEZ5_UTILS_CLIENT.

VasylSamoilov, 2025-01-14:
Yes, it should be installed, but it doesn't. Seems some kind of build problem here.