F5OEO/tezuka_fw #31: [LibreSDR] Ethernet TCP segmentation offload (TSO) breaks iio

francistheodorecatte, 2025-01-14:
Continuing on from here:
https://github.com/F5OEO/tezuka_fw/issues/11#issuecomment-2585933391

It looks like the LibreSDR's Zynq ethernet interface has TSO enabled by default, which is good for performance, but mangles TCP traffic such as iio if the receiving end does not support TSO, or if any equipment between the LibreSDR and the receiving end does not support TSO.

This can be disabled via `ethtool -K eth1 tso off` or `sysctl -w net.inet.tcp.tso=0`, but ethtool is not included in the image, and `/proc/sys/net/inet/tcp/tso` is not populated in sysfs. So beside adding either of those two methods of toggling TSO, if possible, TSO should also be disabled by default.

F5OEO, 2025-01-14:
Nice to have your analysis...I could maybe find if there is any option on the driver ethernet build or adding ethtool to see if it could solve this issue

F5OEO, 2025-01-14:
Adding eth tool and invoke your command , result 
[image]

francistheodorecatte, 2025-01-14:
Looks like the kernel needs to be compiled with `CONFIG_ETHTOOL_NETLINK` for ethtool to work.

I poked around at the Zynq documentation, and looks like TSO is enabled in the Linux TCP/IP stack, not explicitly at the driver or hardware level, which I assume means it's meant to be toggled via sysctl. But I'm not sure what needs to be done to populate the required sysctl paths required, so `net.inet.tcp.tso=0` could be added to sysctl.conf for persistence across reboots.

francistheodorecatte, 2025-01-14:
okay, according to the Linux kernel documentation, TCP segmentation offload is only normally disabled when TCP checksum offloading is disabled, which isn't desirable. ethtool is probably the best method, then.

VasylSamoilov, 2025-01-14:
If you disable rx checksum, you will loose network connectivity due to ```[  303.552061] macb e000b000.ethernet eth0: incorrect FCS```.
You can disable tx checksum offloading, along with TSO (it will switch off everything tx), with not much of a performance penalty. 
```root@analog:~# ethtool -K eth0 tx off
root@analog:~# ethtool -k eth0
Features for eth0:
rx-checksumming: on
tx-checksumming: off
        tx-checksum-ipv4: off [fixed]
        tx-checksum-ip-generic: off
        tx-checksum-ipv6: off [fixed]
        tx-checksum-fcoe-crc: off [fixed]
        tx-checksum-sctp: off [fixed]
```
Ethernet implementation of Zynq platform (and it's driver) is not like something you may encounter on a PC. While I'm not expecting it to be the best ethernet adapter ever, it should send data properly. TSO have nothing to do with source and destination IP, that's for sure, like we see in #11 .

Packet captures that you have attached in #11 demonstrate that TSO is working on the client side (while receiving, nothing to do with data sent). What we see is that device with microsoft mac address is aggregating received packets into bigger one before sending them up the stack. _There is no way 11k sized packets are being generated by Zynq of traversed the physical ethernet)_
If you are having problems, I would recommend you to:
1) Check you power supply. I found that raspberry pi 4/5 PSU with 5V/3A or 5V/4A rating work the best. I had some strange problems when powering LireSDR from phone charger.
2) Check you thermals - is thermal pads or radiators seated properly?
3) Try different client (or just boot clean ubuntu from usb stick to test network connectivity, highly recommended)
4) Don't use VM or virtualized solution like WSL to test - they bend network in a strange way sometimes. Better try option 3.

F5OEO, 2025-01-17:
_Is this why mine keeps crashing on USB?_
No this should not be related. Surely an issue on your third party software.

_Also, can you perhaps give it a name for the router next time you build the firmware._
Can you be more precise ?

_Finally, is it possible to set a fixed ethernet IP in the config?_
Example
ssh to board 
fw_setenv ipaddr_eth 10.0.0.1
fw_setenv netmask_eth 255.255.255.0
fw_setenv gateway_eth 10.0.0.254

You can also do it in the config.txt when usb plugged.

G0MJW, 2025-01-17:
> ___________                  __
>
> \__    ___/___ __________ __|  | _______
>
>   |    |_/ __ \\___   /  |  \  |/ /\__  \
>
>   |    |\  ___/ /    /|  |  /    <  / __ \_
>
>   |____| \___  >_____ \____/|__|_ \(____  /
>
>              \/      \/          \/     \/
>
> Universal Zynq70x/AD936x firmware builder
>
> Version tezuka-3e31
>
> By F5OEO (2024)
>
> # fw_setenv ipaddr_eth 192.168.1.191

It doesn’t get as far as loading the software, IIO crashes after a while, however on ethernet it is quite stable. I just found the later firmware 0.9 so trying that now. Any DATV TX software working with this yet?

 

On the name, my DHCP server has “No Device Name” in the DHCP log, so I assume there is a name somewhere. I did this.

 

login as: root

# fw_setenv gateway_eth 192.168.1.1

francistheodorecatte, 2025-02-09:
🤦 this is an issue with SoapyPlutoSDR and/or the version of libiio on my machine.

you cannot use `"driver=plutosdr,hostname={ipaddr}"`, as that will only enumerate local iio contexts. to enumerate remote iio contexts, you must use `"driver=plutosdr,uri=ip:{ipaddr}"` instead. with the latter invocation, rtl_433 is working perfectly with tezuka entirely over the network.

anyway, sorry about the misdirection!

closed 2025-02-09 by francistheodorecatte