F5OEO/tezuka_fw #32: Build a Kuiper sd image

F5OEO, 2025-01-15:
Providing a iso sd image which boot on first sd partition and use a rootfs kuiper could be interesting. 
Need to study how we can automate this.

VasylSamoilov, 2025-01-15:
Kuiper linux is not build from scratch, it's raspberry pi raspbian + some modifications.
Original raspberry pi image generator is here: https://github.com/RPi-Distro/pi-gen
Analog devices fork for kuiper linux is here: https://github.com/analogdevicesinc/adi-kuiper-gen

VasylSamoilov, 2025-01-15:
Benefit of using raspbian is that packages can be installed from raspbian repo and it's mirrors. 
Negatives is that complex, highly-optimized packages like FFmpeg crash with "unsupported instruction" message, as zynq don't have some instructions and GPU like raspberry pi does.

F5OEO, 2025-01-15:
First steps : 
[image]

VasylSamoilov, 2025-01-15:
here is usb gadget daemon .scheme file that does something like usb gadget script:
/usr/local/etc/gt/adi/iio_acm_generic.scheme (modified, attached)
[iio_acm_generic.scheme.txt](https://github.com/user-attachments/files/18430729/iio_acm_generic.scheme.txt)

F5OEO, 2025-01-16:
After testing Kuiper, I don't see real advantages vs a buildroot build. Main avantage could be debian package, but as tezuka handles ipk package this is not a real plus. Maybe option to have rootfs on sd could be usefull.

VasylSamoilov, 2025-01-16:
Software Defined Radio usefulness is defined by software. Full linux is not necessary, but a _very_ nice option to have - depends on application. 

LibreSDR is now a _cheapest full-duplex SDR_ available, but it's usefulness is limited due to software support.
So far the most interesting thing is that pgreenland made lte base station running. There is also OpenWiFi firmware (with it's limitations). 

Software of interest that I have on the radar:
HackTV - supports Soapy, doesn't work properly (samplerate is too low, maybe something else).
Osmocom (2G base station) - doesn't support SoapySDR, needs osmo-trx variant for plutosdr ( https://github.com/pgreenland/plutosdr-fw/issues/3 ) .  
Osmocom-analog (0G/1G cell networks, paging) - doesn't work: soapy sdr rx overflow ( https://gitea.osmocom.org/cellular-infrastructure/osmocom-analog/issues/28 )

Just to make myself clear - full linux on sd card will make it possible to run (for example) osmocom-analog and asterisk together on a same device, making for a nice AMPS of NMT base station with asterisk connecting to (again, for example) to phone via bluetooth. I have made such setup with raspberry pi, limesdr and android smartphone - but libresdr is much better as it is everything combined in one device.

There are also other options to run software on libresdr itself and send only resulting data over network (or transmitting). Maybe DMR hotspot, maybe paging gateway, maybe frequency scanner or smart radio multi-frequency repeater. You never know :)

VasylSamoilov, 2025-01-16:
Bottom line - good to have that option, as buildroot is more suitable for embedding, and raspbian is more user-friendly for apps and customization. But definitely not better.

crackerjacques, 2025-01-17:
For a non-specialist hobbyist like myself, I find it very gratifying that the Debian-based system makes it easier to try out ideas that come to mind.
But it also increases the likelihood of starting to build SD images again.

F5OEO, 2025-02-13:
Closed until new trials.

closed 2025-02-13 by F5OEO