F5OEO/tezuka_fw #34: Problems with transmitting signals of relatively high bandwidth with LibreSDR

VasylSamoilov, 2025-01-15:
I have no problems receiving signals with LibreSDR, but I have problems transmitting anything more complex than narrow fm.

I tried to use latest Sdrangel, build in docker for debian 12. I used DATV DVB-S encoder and decoder with default setting and QPKS, because even if I'm not sending data, I can receive and monitor QPKS constellation to ensure I'm getting clear and clean signal with proper phase shifts. Default bandwidth setting is about 1MHz, so 2-4 MSPS should be ok and withing LibreSDR capabilities, I presume. 
As a reference device I used USRP B210 connected to LibreSDR via attenuator.
USRP source and sink - all good
USRP source and LibreSDR sink - all good
LibreSDR source and USRP sink - on transmit noise level jumps to very high level, and signal cannot be decoded (I can't even identify it on waterfall). 

Sending something like narrow fm with bandwith of 12 KHz works OK.

Can somebody confirm that sending relatively wideband signals works for them?
My ultimate goal is to get hacktv working _properly_ with LibreSDR.
 
I can't quite understand is the the problem with my unit, problem with firmware, or problem with SoapySDR driver, or maybe I'm just missing some settings that should be set.
Thanks in advance.

VasylSamoilov, 2025-01-16:
Seems to be the same issue:
https://github.com/hz12opensource/libresdr/issues/18

F5OEO, 2025-01-16:
I am experimenting some level tx issue also with libresdr with sdrangel ! I am investigating.

F5OEO, 2025-01-16:
This is the spectrum out of libresdr : a wide band spectrum of noise (which is the IQ symbolrate) and we can see a litlle datv spectrum on it. Output has a real issue. Time to open the libresdr.

[image]

VasylSamoilov, 2025-01-16:
Maybe a SDRAngel plugin issue or iio issue (client or server or firmware related), or hardware, or combination of both. Hard to tell.

Probably somewhat related: experiments with modding original pluto for datv:
https://www.f5uii.net/en/transmit-datv-over-qo100-with-sdr-adalm-pluto-f5oeo-plutodvb/4/

F5OEO, 2025-01-16:
Same firmware with antsdr e200 and sdrangel work perfectly. Need to double check fpga, but I doubt. Should be relative to RF line.

I note also that tx2 is not working at all !

Do not use mod from plutosdr rev b, this is just for replacing tcxo, which is fine on libresdr.

[image]

F5OEO, 2025-01-16:
TX 1 ouput spectrum should be fixed with 3e3178f5657a19b3768ab9c8950d058f17a5ee19

See difference : 

[image]

F5OEO, 2025-01-16:
TX2 is working also but has around 4db less power.

VasylSamoilov, 2025-01-16:
It definitely works now! Question: what parameters to SoapySDR should be provided to use TX2? or RX2? 

[image]

F5OEO, 2025-01-16:
SoapySDR for Tezuka is still under development. I didn't add rx2/tx2 selection yet.
But in the meantime you can ssh to your libresdr and
For RX2 : ./switch_rfinput.sh rx2
For TX2 : ./switch_rfoutput.sh tx2
After changing selection, you should move the gain to be taken in account.

closed 2025-01-16 by F5OEO