F5OEO/tezuka_fw #35: Problems with NFM/AM/WFM transmission

VasylSamoilov, 2025-01-16:
After wideband transmission fix there is a problem with narrowband signal transmission (at least for my setup)
AM and NFM modulation produces audible beep when carrier is on, probably due to tx artifacts? Modulation also is not very good and once a second or two there is visible and audible spikes (clicks).

widefm modulation not working at all, at least in sdrangel (I can't even describe what is happening, see screenshot)
I tested with two platforms and two versions of sdrangel, usrp b200 as reciever, with attenuator and coax and with simple antennas. 

This may be a radio path/hardware problem, but it doesn't explain why widefm modulation is all over the place programmatically.

[image]
[image]

VasylSamoilov, 2025-01-16:
OK, wide fm plugin in sdrangel seems to be not best quality, NFM and AM transmission problems _partially_ are due to DC spike in central frequency, and any attempt to create frequency offset moves DC spike. Even with that being said, there is a lot of signal irregularities and widefm is completely broken. 

And it would be nice to have possibility to transmit a little bit to the side of DC spike (LO offset capability in driver?), otherwise it's pretty useless.

VasylSamoilov, 2025-01-17:
OK, this is strange. Now it's working just fine on transmit. Is there a chance that iiod can miss or skip some commands via network and get stuck in some mode that will transmit distorted signal? Or stop reacting to some control commands?

F5OEO, 2025-01-17:
Third party software use a scan iio to get the context. Unfortunatelly, right now, if you have usb and ethernet , the uri is the same for ip over usb and ethernet ip .
Example 
0: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=4UT7MDH5LMOJIF75 [usb:1.80.5]
1: 10.0.0.55 (PlutoSDR libresdr), serial=4UT7MDH5LMOJIF75 [ip:libresdr.local]
2: 192.168.2.1 (PlutoSDR libresdr), serial=4UT7MDH5LMOJIF75 [ip:libresdr.local]

Connecting to libresdr.local could be thus usb or ethernet. If it is usb , you are constraint in bandwidth and should have to big issues with spectrum for high bandwidth/

VasylSamoilov, 2025-01-17:
Thank you. So far - so good, I'll close that one and reopen if issue will be properly reproducible.

closed 2025-01-17 by VasylSamoilov