F5OEO/tezuka_fw #36: How to Switch Antennas

ashjmckenzie, 2025-01-17:
Hi there, If I am using SDR++ how might I swap RX ports ?

Many Thanks

VasylSamoilov, 2025-01-17:
still work in progress, but you can test like this:
https://github.com/F5OEO/tezuka_fw/issues/34#issuecomment-2596136568

ashjmckenzie, 2025-01-17:
Thanks so much, so if I am running SDR++ and I SSH and send that command will it switch the RX live? Will it stay on RX2 till reboot or will it persist until changed again ?

F5OEO, 2025-01-17:
Easy way is to use the fork version https://github.com/F5OEO/SDRPlusPlus/releases/tag/nightly

[image]

VasylSamoilov, 2025-01-17:
Is there anything that needs to be set? because what I have:
iio_info.exe -s
Unable to create Local IIO context : Function not implemented (40)
Available contexts:
        0: 192.168.160.192 (LibreSDR Rev.5 (Z7020-AD9361)), serial=4GGSOS6HJ4ICTRIR [ip:libresdr.local]
        1: 192.168.2.1 (LibreSDR Rev.5 (Z7020-AD9361)), serial=4GGSOS6HJ4ICTRIR [ip:libresdr.local]
        2: fe80::205:f7ff:fef7:799c%6 (LibreSDR Rev.5 (Z7020-AD9361)), serial=4GGSOS6HJ4ICTRIR [ip:libresdr.local]
        3: 0456:b673 (Analog Devices Inc. PlutoSDR (ADALM-PLUTO)), serial=4GGSOS6HJ4ICTRIR [usb:1.47.5]

and LibreSDR is not being detected at all (using fork version).

F5OEO, 2025-01-17:
Ooops need to whitelist libresdr...Will add it.
For a workaround until this, you can 
fw_setenv force_model "PlutoSDR libresdr"

F5OEO, 2025-01-17:
There is a bug when mutltiple interfaces exist (usb + ethernet) . Try to unplugged usb otg , and use only ethernet for test.

F5OEO, 2025-02-14:
Closed as issue is not firmware.

closed 2025-02-14 by F5OEO