F5OEO/tezuka_fw #38: Investigate LibreSDR linaro-ubuntu14 SD image from OpenSourceSDRLab (link)

VasylSamoilov, 2025-01-19:
Just received LibresDR (yes, with that capitalization :) from OpenSourceSDRLab shop from China. Better PCB design, better case in terms of RF shielding, better soldering (at least it's present where it should be). 
Comes with SD card with pre-loaded linaro ubuntu 14 with some modifications.
Attaching dropbox link so anyone willing can investigate what is different.

https://www.dropbox.com/scl/fi/mggckr1vx7vhsdroj5mec/libresdr_sd_card.img.gz?rlkey=igmw7nkgn2k89ah4e5espjbs0&st=mnt8bz8j&dl=0

F5OEO, 2025-01-19:
Wow, very interesting. But the file seems corrupt as I could not  extract it. Archive is 2GB but contains a 1.2G img file : weird.

crackerjacques, 2025-01-19:
I've got an ≈32GB image file and it's currently being written with Etcher.
[image]

F5OEO, 2025-01-19:
Dropbox says that the file is too big. Thus I can only download 2GB (maybe
because I have a basic account?)

VasylSamoilov, 2025-01-19:
Let me upload it somewhere else

crackerjacques, 2025-01-19:
> Dropbox says that the file is too big. Thus I can only download 2GB (maybe
> because I have a basic account?)
> 
> Le dim. 19 janv. 2025 à 15:14, crackerjacques ***@***.***> a
> écrit :
> […](#)

I'm a basic account too.
The cause is unknown.

And now I'm trying it out as an SDR device.
Satdump[forked] works perfectly, but SDR++ doesn't even seem to grab USB devices. This is even without the forked version.

VasylSamoilov, 2025-01-19:
> Wow, very interesting. But the file seems corrupt as I could not extract it. Archive is 2GB but contains a 1.2G img file : weird.

My bad, upload limit was reached on my side.
Please try this magnet link, should be more reliable:
magnet:?xt=urn:btih:4b46b6cc99880094249a81e25c2a75fbc38bcace&dn=libresdr.img&tr=http%3A%2F%2Fopen.tracker.cl%3A1337%2Fannounce&tr=udp%3A%2F%2Fns-1.x-fins.com%3A6969%2Fannounce&tr=udp%3A%2F%2Ftracker.torrent.eu.org%3A451%2Fannounce&tr=http%3A%2F%2Ftracker.xiaoduola.xyz%3A6969%2Fannounce&tr=http%3A%2F%2Fwww.genesis-sp.org%3A2710%2Fannounce&tr=http%3A%2F%2Ftracker.sbsub.com%3A2710%2Fannounce&tr=udp%3A%2F%2Fisk.richardsw.club%3A6969%2Fannounce&tr=http%3A%2F%2Ftracker810.xyz%3A11450%2Fannounce&tr=udp%3A%2F%2Ftracker.opentrackr.org%3A1337%2Fannounce&tr=http%3A%2F%2Ftracker.bz%3A80%2Fannounce&tr=http%3A%2F%2Ftracker.ipv6tracker.org%3A80%2Fannounce&tr=udp%3A%2F%2Fopen.stealth.si%3A80%2Fannounce&tr=http%3A%2F%2Fwww.torrentsnipe.info%3A2701%2Fannounce&tr=http%3A%2F%2Ftracker.dmcomic.org%3A2710%2Fannounce&tr=udp%3A%2F%2Fopen.demonii.com%3A1337%2Fannounce&tr=udp%3A%2F%2Fexplodie.org%3A6969%2Fannounce&tr=udp%3A%2F%2Fexodus.desync.com%3A6969%2Fannounce&tr=udp%3A%2F%2Fdiscord.heihachi.pw%3A6969%2Fannounce&tr=udp%3A%2F%2Ftracker.dump.cl%3A6969%2Fannounce&tr=http%3A%2F%2Ftracker.moxing.party%3A6969%2Fannounce

VasylSamoilov, 2025-01-19:
I also uploaded trimmed version to google drive (zeroed out free and unused space + gzip)
https://drive.google.com/file/d/1KMb6e77HrsQb43iuDqKu7MRxqjyqnAOE/view?usp=drive_link

For full sd card image better to use torrent magent link (just in case someone needs the full sector-to-sector copy)

F5OEO, 2025-01-20:
Even your google drive file seems to have issue until it is on my side ?

VasylSamoilov, 2025-01-20:
What kind of issues? Mounts fine on my side.

VasylSamoilov, 2025-01-20:
I can see where it's coming from. A lot of gui tools don't understand 32gb file size and shows 1.2gb file inside 1.9gb archive. Just gunzip it under linux cli, it should be fine.

VasylSamoilov, 2025-01-20:
the last attempt: packed with winrar:
https://drive.google.com/file/d/1sl9FT8Gkf9FeCurlxJGUZlQuiNt1OXcS/view?usp=sharing

F5OEO, 2025-01-20:
My fault...disk space was not enough on my disk !!! 
Without flashing on SD, it appears to be a kuiper distro.
Strange : there is no dtb , so surely need to use a zynq-common one, so surely some missing features.
So relative question : Is ethernet supported straight foward ? (without using tezuka dtb) ?

VasylSamoilov, 2025-01-20:
Yes, it works just fine, and gets address via dhcp by default

F5OEO, 2025-01-20:
Just with this image, without overwiting anything ? (devicetree.dtb?)

VasylSamoilov, 2025-01-20:
yes, out of the box

VasylSamoilov, 2025-01-20:
There are two partitions in this image - fat32 for boot and ext for linux itself. Maybe device tree is on the linux partition?

VasylSamoilov, 2025-01-20:
boot log from debug port:

```
DHCP client bound to address 192.168.160.221 (4339 ms)
Hit any key to stop autoboot:  0
376 bytes read in 9 ms (40 KiB/s)
Loaded environment from uEnv.txt
Importing environment from SD ...
Running uenvcmd ...
Copying Linux from SD to RAM...
** No boot file defined **
Device: mmc@e0100000
Manufacturer ID: fe
OEM: 3432
Name: SDBus Speed: 50000000
Mode : SD High Speed (50MHz)
Rd Block Len: 512
SD version 3.0
High Capacity: Yes
Capacity: 29.1 GiB
Bus Width: 4-bit
Erase Group Size: 512 Bytes
6338888 bytes read in 371 ms (16.3 MiB/s)
## Loading kernel from FIT Image at 10000000 ...
   Using 'conf@system-top.dtb' configuration
   Verifying Hash Integrity ... OK
   Trying 'kernel@1' kernel subimage
     Description:  Linux kernel
     Type:         Kernel Image
     Compression:  uncompressed
     Data Start:   0x100000f4
     Data Size:    6316056 Bytes = 6 MiB
     Architecture: ARM
     OS:           Linux
     Load Address: 0x00008000
     Entry Point:  0x00008000
     Hash algo:    sha1
     Hash value:   9c3f1c24e0fefe6aa1f20c28f83eed335b73675c
   Verifying Hash Integrity ... sha1+ OK
## Loading fdt from FIT Image at 10000000 ...
   Using 'conf@system-top.dtb' configuration
   Verifying Hash Integrity ... OK
   Trying 'fdt@system-top.dtb' fdt subimage
     Description:  Flattened Device Tree blob
     Type:         Flat Device Tree
     Compression:  uncompressed
     Data Start:   0x1060620c
     Data Size:    20939 Bytes = 20.4 KiB
     Architecture: ARM
     Hash algo:    sha1
     Hash value:   bea418517866233c78c75b33d1a4deedaedbb614
   Verifying Hash Integrity ... sha1+ OK
   Booting using the fdt blob at 0x1060620c
   Loading Kernel Image ... OK
   Loading Device Tree to 07ff7000, end 07fff1ca ... OK

Starting kernel ...

Booting Linux on physical CPU 0x0
Linux version 4.19.0 (oe-user@oe-host) (gcc version 8.2.0 (GCC)) #1 SMP PREEMPT                                                                                                                                    Wed Jun 12 13:06:56 UTC 2024
CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt: Machine model: Analog Devices ADRV9364-Z7020 (Z7020/AD9364)
earlycon: cdns0 at MMIO 0xe0000000 (options '115200n8')
bootconsole [cdns0] enabled
Memory policy: Data cache writealloc
cma: Reserved 128 MiB at 0x38000000
random: get_random_bytes called from start_kernel+0xa0/0x404 with crng_init=0
percpu: Embedded 16 pages/cpu @(ptrval) s33548 r8192 d23796 u65536
Built 1 zonelists, mobility grouping on.  Total pages: 260608
[... 185 lines cut ...]
VFS: Mounted root (ext4 filesystem) on device 179:2.
devtmpfs: mounted
Freeing unused kernel memory: 1024K
Run /sbin/init as init process
Mount failed for selinuxfs on /sys/fs/selinux:  No such file or directory
random: init: uninitialized urandom read (12 bytes read)
init: hwclock main process (1159) terminated with status 1
random: mountall: uninitialized urandom read (12 bytes read)
 * Setting up X socket directories...                                    [ OK ]
 * STARTDISTCC is set to false in /etc/default/distcc
 * /usr/bin/distccd not starting
 * Starting IIO Daemon iiod                                                     configfs-gadget ci_hdrc.0: failed to start ffs: -19
sh: echo: I/O error
                                                                         [ OK ]

Last login: Thu Jan  1 00:00:10 UTC 1970 on tty1
Welcome to Linaro 14.04 (GNU/Linux 4.19.0 armv7l)

 * Documentation:  https://wiki.analog.com/ https://ez.analog.com/

New release '16.04.6 LTS' available.
Run 'do-release-upgrade' to upgrade to it.

root@analog:~#
```

VasylSamoilov, 2025-01-20:
It seems to acquire DHCP address even before it boots. Maybe a network boot option?

zx-ecco, 2025-01-28:
what listing did you order from? (opensourcesdr labs has three listings with three different prices but I can't see any actual difference in the specs)

closed 2025-02-13 by F5OEO