F5OEO/tezuka_fw #41: buildroot: broken Git references to maia and FutureSDR

sly547, 2025-01-21:
I'm having issues with buildroot. Git does not like some of the source definitions. I had similar problems with hz12opensource/libresdr and found this comment:  https://github.com/analogdevicesinc/buildroot/issues/65#issuecomment-1190116333

Here in this case it's a bit different.. I can fix it manually by changing the version from `main` to `origin/main`.. this is how it fails:

```
>>> maia-httpd main Downloading                                                                                                                                                               
GIT_DIR=/home/michael/src/ZynqSDR/tezuka_fw/buildroot-2024.08/dl/maia-httpd/git/.git git init .                                                                                               
Initialized empty Git repository in /home/michael/src/ZynqSDR/tezuka_fw/buildroot-2024.08/dl/maia-httpd/git/.git/
GIT_DIR=/home/michael/src/ZynqSDR/tezuka_fw/buildroot-2024.08/dl/maia-httpd/git/.git git remote add origin 'https://github.com/F5OEO/maia-sdr.git'                                            
GIT_DIR=/home/michael/src/ZynqSDR/tezuka_fw/buildroot-2024.08/dl/maia-httpd/git/.git git remote set-url origin 'https://github.com/F5OEO/maia-sdr.git'                                        
Fetching all references                                                                                                                                                                       
GIT_DIR=/home/michael/src/ZynqSDR/tezuka_fw/buildroot-2024.08/dl/maia-httpd/git/.git git fetch origin                                                                                         
remote: Enumerating objects: 1641, done.                                                                                                                                                      
remote: Counting objects: 100% (558/558), done.                                                                                                                                               
remote: Compressing objects: 100% (334/334), done.                                                                                                                                            
remote: Total 1641 (delta 326), reused 265 (delta 224), pack-reused 1083 (from 2)                                                                                                             
Receiving objects: 100% (1641/1641), 1.94 MiB | 10.18 MiB/s, done.                                                                                                                            
Resolving deltas: 100% (1015/1015), done.                                                                                                                                                     
From https://github.com/F5OEO/maia-sdr                                                                                                                                                        
 * [new branch]      maia-iio   -> origin/maia-iio                                                                                                                                            
 * [new branch]      main       -> origin/main                                                                                                                                                
 * [new branch]      sweep      -> origin/sweep                                                                                                                                               
GIT_DIR=/home/michael/src/ZynqSDR/tezuka_fw/buildroot-2024.08/dl/maia-httpd/git/.git git fetch origin -t -f                                                                                   
Could not fetch special ref 'main'; assuming it is not special.                                                                                                                               
Commit 'main' does not exist in this repository.                                                                                                                                              wget -nd -t 3 -O '/home/michael/src/ZynqSDR/tezuka_fw/buildroot-2024.08/output/build/.maia-httpd-main-git4.tar.gz.ZwxFF9/output' 'https://sources.buildroot.net/maia-httpd/maia-httpd-main-git
4.tar.gz'                                                                                                                                                                                     
--2025-01-20 00:15:58--  https://sources.buildroot.net/maia-httpd/maia-httpd-main-git4.tar.gz                                                                                                 Resolving sources.buildroot.net... 2606:4700:20::681a:125, 2606:4700:20::681a:25, 2606:4700:20::ac43:4838, ...                                                                                
Connecting to sources.buildroot.net|2606:4700:20::681a:125|:443... connected.                                                                                                                 
HTTP request sent, awaiting response... 404 Not Found                                                                                                                                         
2025-01-20 00:15:58 ERROR 404: Not Found.                                                                                                                                                     
                                                                                                                                                                                              
wget -nd -t 3 -O '/home/michael/src/ZynqSDR/tezuka_fw/buildroot-2024.08/output/build/.maia-httpd-main-git4.tar.gz.nCRT80/output' 'https://sources.buildroot.net/maia-httpd-main-git4.tar.gz'  
--2025-01-20 00:15:58--  https://sources.buildroot.net/maia-httpd-main-git4.tar.gz                                                                                                            
Resolving sources.buildroot.net... 2606:4700:20::681a:125, 2606:4700:20::681a:25, 2606:4700:20::ac43:4838, ...                                                                                
Connecting to sources.buildroot.net|2606:4700:20::681a:125|:443... connected.                                                                                                                 
HTTP request sent, awaiting response... 404 Not Found                                                                                                                                         2025-01-20 00:15:59 ERROR 404: Not Found.                                                                                                                                                     
                                                                                                                                                                                              make: *** [package/pkg-generic.mk:179: /home/michael/src/ZynqSDR/tezuka_fw/buildroot-2024.08/output/build/maia-httpd-main/.stamp_downloaded] Error 1
```

Here is my great "fix":

```
diff --git a/package/futuresdr/futuresdr.mk b/package/futuresdr/futuresdr.mk
index 1f55d81..df668e4 100644
--- a/package/futuresdr/futuresdr.mk
+++ b/package/futuresdr/futuresdr.mk
@@ -3,7 +3,7 @@
 # FutureSDR
 #
 ################################################################################
-FUTURESDR_VERSION = main
+FUTURESDR_VERSION = origin/main
 FUTURESDR_SITE = https://github.com/FutureSDR/FutureSDR.git
 FUTURESDR_SITE_METHOD = git
 CROSS_COMPILE = arm-none-linux-gnueabihf-
diff --git a/package/maia-httpd/maia-httpd.mk b/package/maia-httpd/maia-httpd.mk
index 3f46bed..7059c26 100644
--- a/package/maia-httpd/maia-httpd.mk
+++ b/package/maia-httpd/maia-httpd.mk
@@ -4,7 +4,7 @@
 #
 ################################################################################
 
-MAIA_HTTPD_VERSION = main
+MAIA_HTTPD_VERSION = origin/main
 MAIA_HTTPD_SITE = https://github.com/F5OEO/maia-sdr.git
 MAIA_HTTPD_SITE_METHOD = git
 MAIA_HTTPD_DEPENDENCIES = host-openssl
diff --git a/package/maia-wasm/maia-wasm.mk b/package/maia-wasm/maia-wasm.mk
index 7da17ce..71de0f7 100644
--- a/package/maia-wasm/maia-wasm.mk
+++ b/package/maia-wasm/maia-wasm.mk
@@ -5,7 +5,7 @@
 ################################################################################
 
 
-MAIA_WASM_VERSION = main
+MAIA_WASM_VERSION = origin/main
 MAIA_WASM_SITE = https://github.com/F5OEO/maia-sdr.git
 MAIA_WASM_SITE_METHOD = git
 MAIA_WASM_DEPENDENCIES = rust-wasm wasm-pack
```

F5OEO, 2025-01-22:
What defconfig do you use ? Normally futuresdr is not build

sly547, 2025-01-22:
> Weird as there is no issue with the github CI build neither on my machine. I have to investigate why there should be a issue like this.

Maybe my version of Git is more recent and something changed about how it handles things? Just speculation. Ubuntu [jammy (22.04LTS)](https://packages.ubuntu.com/jammy/) for example still packages Git 2.34..
```
$ git --version
git version 2.45.2
```

> What defconfig do you use ? Normally futuresdr is not build

Ah, I was using `pluto_maiasdr_defconfig` so I guess FutureSDR was not an issue - I just changed all occurences of VERSION = main that I could find.

F5OEO, 2025-02-14:
Maybe 24.04LTS based...not yet migrated. I close for now util I can reproduce the issue with recent OS.

sq6emm, 2025-02-14:
FYI I run all builds on 24.04 LTS Ubuntu, no issues at all.