F5OEO/tezuka_fw #44: activating the PTT output from the Pluto Plus

ruediger-moeller, 2025-01-25:
Hi all,
I have successfully unbricked my Pluto Plus with the help of the Tezuka software and a Micro SD card. Also I am using the Tezuka firmware now for QO100 QSOs. Controling the PTT function from the Pluto Plus with the SDR console shows, that the optocoupler on the PCB will not be activated from the Tezuka software. That's too bad. The ingress port from the PTT optocoupler (U90010) is already connected over the gate from QA5 with the PSMIO0 port (F6) from U31E. So it should be possible with the help of the Tezuka firmware to use this port to rialize the PTT functions.
Best regards
Ruediger
DJ1MR

F5OEO, 2025-01-28:
Need to double check, but this should work. Remember however that there is a hardware issue with pluto+ ptt as current is not enough to drive optocoupler. Workaround is to use GPIO directly.

ruediger-moeller, 2025-01-28:
Hello,

I have had a closer look on the circuit diagram from the Pluto+ and have figured out, that you may be right. The reason for the current limitation is laying on RA11, a drain resistor from QA5 and the internal optocoupler emitter-diode. RA1 has a value from 4,7 KOhm and therefore the possible current is too low. So changing the value from RA11 against 470 ohm should solve the problem. It is very simple to unsolder the original resistor RA11 and solder a new one with a value of 470 ohm. I would really like to do this changings on the Pluto+ PCB if you are able to provide a Tezuka firmware which could control the pin PS_MIO0_500 (F6) from IC U31-E for PTT services.

I am looking forward to your decision.

Best regards

Ruediger

DJ1MR

F5OEO, 2025-01-28:
From the author of plutoplus hardware (several months ago) : 
_RA11 is too high 4.7K which should be less (1K) to properly drive the optocoupler_

GPIO is already implemented in tezuka : https://github.com/F5OEO/tezuka_fw/blob/478b14f134dad752c60ff0609dfe595e9375db96/board/pluto/overlay_maia/root/watchconsoletx.sh#L47

ruediger-moeller, 2025-01-31:
Hello,

 

I have changed the value of RA11 as visible in the attached circuit diagram.
The PTT egress port flapps during the deactivation phase of the PTT signal.
Also, during the boot phase of the Pluto Plus the PTT egress port is active
and flapps before it goes in the inactive state. This behaviour is very
unfavorable and could damage the 2,4 GHz PAs. Do you see a chance  to
overcome these shortcomings via software?

 

Best regards

Ruediger 

DJ1MR

(Issue #44)

 

https://github.com/F5OEO/tezuka_fw/blob/478b14f134dad752c60ff0609dfe595e9375
db96/board/pluto/overlay_maia/root/watchconsoletx.sh#L47


6TA5AVCNFSM6AAAAABV3GHBB2VHI2DSMVQWIX3LMV43OSLTON2WKQ3PNVWWK3TUHMZDMMJZGQZDK
NRTG4> .
<https://github.com/notifications/beacon/AOEDYCVHP2ZDJ5T6NSGRADL2M6TA5A5CNFS
M6AAAAABV3GHBB2WGG33NNVSW45C7OR4XAZNMJFZXG5LFINXW23LFNZ2KUY3PNVWWK3TUL5UWJTU

F5OEO, 2025-01-31:
Can you provide a recording (oscilloscope image) of the ptt output ?

ruediger-moeller, 2025-01-31:
Will try it to realize next week. I will borrow an digital Oszi and will make a recording file. Best regardsRuedigerDJ1MRVon meinem HONOR-Telefon gesendet-------- Ursprüngliche Nachricht --------Von: F5OEO ***@***.***>Datum: Fr., 31. Jan. 2025, 13:26An: F5OEO/tezuka_fw ***@***.***>Cc: ruediger-moeller ***@***.***>, Author ***@***.***>Betreff: Re: [F5OEO/tezuka_fw] activating the PTT output from the Pluto Plus (Issue #44)
Can you provide a recording (oscilloscope image) of the ptt output ?

ruediger-moeller, 2025-01-31:
First I will send you a video. This video file was rejected from the mail server. I will try it again as a zip file. Von meinem HONOR-Telefon gesendet-------- Ursprüngliche Nachricht --------Von: F5OEO ***@***.***>Datum: Fr., 31. Jan. 2025, 13:26An: F5OEO/tezuka_fw ***@***.***>Cc: ruediger-moeller ***@***.***>, Author ***@***.***>Betreff: Re: [F5OEO/tezuka_fw] activating the PTT output from the Pluto Plus (Issue #44)
Can you provide a recording (oscilloscope image) of the ptt output ?

ruediger-moeller, 2025-01-31:
two Mails from me with one video each about 11 MB are rejected from the Github Mailserver because of size limitations. Do you have a better mail account? Von meinem HONOR-Telefon gesendet-------- Ursprüngliche Nachricht --------Von: F5OEO ***@***.***>Datum: Fr., 31. Jan. 2025, 13:26An: F5OEO/tezuka_fw ***@***.***>Cc: ruediger-moeller ***@***.***>, Author ***@***.***>Betreff: Re: [F5OEO/tezuka_fw] activating the PTT output from the Pluto Plus (Issue #44)
Can you provide a recording (oscilloscope image) of the ptt output ?

F5OEO, 2025-01-31:
***@***.***

F5OEO, 2025-01-31:
You can send me message : evaristec@gmail.com

F5OEO, 2025-02-13:
This is a hardware issue. Not related to firmware. Reopen with your fixed harware solution please.

closed 2025-02-13 by F5OEO