F5OEO/tezuka_fw #45: Libresdr manufacturer source

F5OEO, 2025-01-28:
> what listing did you order from? (opensourcesdr labs has three listings with three different prices but I can't see any actual difference in the specs)
>  

 _Originally posted by @zx-ecco in [#38](https://github.com/F5OEO/tezuka_fw/issues/38#issuecomment-2619128515)_

zx-ecco, 2025-01-28:
thanks, It was meant for @VasylSamoilov since he now has a known good unit after the first one with no vias, but I'll take any feedback I can get

VasylSamoilov, 2025-01-28:
@zx-ecco @F5OEO  This one is better quality one: https://www.aliexpress.com/item/1005007369276242.html?spm=a2g0o.order_list.order_list_main.216.5ee41802uMWGMI

fkb83, 2025-01-28:
How to tell the difference?
I bought mine in summer last year on banggood and it has the LibreSDR marking on the top.

[image]

[image]

VasylSamoilov, 2025-01-29:
This one is fine. See them perforations on the board everywhere? These are "vias", or holes to connect layers on the PCB. On the lower quality one they are missing, turning part of the board into capacitor (on specific frequencies).

zx-ecco, 2025-01-29:
is there any difference between that listing and https://www.aliexpress.com/item/1005005165734650.html it seems to be the exact same device with a different price point (bigsave channel?)

zx-ecco, 2025-01-29:
@VasylSamoilov also, what frequencies did you experience issues with? @G0MJW did you have any of those issues with yours? (I see here that you've ordered from the same cheaper link I mention above) https://github.com/F5OEO/tezuka_fw/issues/11#issuecomment-2563849878 )

lubuntu-l100, 2025-02-03:
Hello Evariste, I have the cheap LibreSDR in test since November 2024. It is the version without the vias, so I have solderd a short piece of RG402 semi ridgid coax between the TX transformer and the TX SMA connector to prevent that the output is fluctuating for example at 2.4 Ghz.

After the modification with the coax the output was stable from 2 M up to 13 cm.

See attached photo of the modification.

[image]

I did also several tests with an External 40 Mhz ref signal, as a replacement of the internal 40 MhzVTCXCO but without success, due to the fact that the PLL circuit of the 40 Mhz is controlled by the FPGA and the output of the 40 Mhz signal is split-up in 2 signals, 1 for the FPGA and 1 for the AD9363. The pll circuit in the FPGA needs to look to the 10 Mhz external clock, compares the differences between the 10 Mhz and the 40 Mhz VTCXCO and via a DAC it is steering the VTCxCO.
This way of controlling the 40 Mhz oscillator came from the Ettus B210.

I tested your 0.0.95 version of the Tazuka firmware yesterday and that is running fine from the SD card.

I am using the LibreSDR for the QO100 NB with SDRConsole v3.4 beta. It is working fine, except the frequency stability of the 40 MHZ oscillator. So a solution to use the LibreSDR with an external 10 Mhz reference would be the ideal solution.

73 Jaap
PA2JSA

VitGGGGGGGGGGGGGGGGGGGG, 2025-02-04:
VIAS actually exist, it's just a very good mask and they're not visible. Problem in wrong type of zener diode that's installed  in this model and has a capacity of 10 pF. Remove zener near the SMA connector  and everything will work fine.
[image]

see my video https://www.youtube.com/watch?v=ow0Q6Wli4es

FPSychotic, 2025-02-04:
How I know if my libresdr has the diode issue?

lubuntu-l100, 2025-02-04:
I removed the Tx diode earlier but the rf instability at 2.4 Ghz was still there.

The output power with the diode was around 2 db lower than after removing the diode.

By cutting the original tx stripline between transformer and connector and replace the connection with a small piece of rg405 or rg402 semi ridgid coax cable was solving the instability problem.

Attention required to solder the outer shield of the coax to Gnd on both sides.

⁣BlueMail voor Android downloaden ​

FPSychotic, 2025-02-04:
I changed that baluns in a blacksdr n210mini by the mini-circuit ones. Just
in case you want a better mod. I read also a company removed the some coil
of the baluns to reduce cross-talking or something like that.
Didn't test if they improved anything but I guess, anyways they were
damaged so no other option.
There was a copy of minicircuits and others that are minicircuit but they
don't go to 6ghz

VitGGGGGGGGGGGGGGGGGGGG, 2025-02-05:
I also want to draw attention to the fact that in a cheaper version of LibreSDR known as AMGeekSDR, instead of VCTXO a regular TCXO is installed, which does not have external voltage control, and this device will not work from an external reference signal. To do this, you need to change the reference generator to VCTXO. But I advise you not to do this. In my experience, operation from an external reference signal does not stand up to any criticism. The PLL circuit is implemented on an 8-bit DAC that works too roughly and we have constant micro-jumps in frequency due to the fact that the PLL cannot make a frequency latch due to the large step of the tuning voltage. It would be better to use at least a 10-bit DAC there, and better yet, a 12-bit one. i`m use the freq calibration in config file. this work fine without external 10 MHz reference.

VitGGGGGGGGGGGGGGGGGGGG, 2025-02-05:
> How I know if my libresdr has the diode issue?
> 
 if your zener diode marked as C5 then remove it. If Z3 then everything is fine, it has a capacity of 0.2 pF and it corresponds to what is indicated in the Libre original datasheet.

 By the way, don't forget to remove the C5 zener diodes from the RX part because they also significantly worsen reception at high frequencies. The original AdalmPluto and Pluto+ don't have any zeners at all.

lubuntu-l100, 2025-02-05:
The last sentence is not completely correct.

The AD Pluto rev D has also diodes at the tx and rx connector sites. Earlier pluto versions don't have these diodes.

The implementation in the LibreSDR is copied from the Pluto rev D.

⁣BlueMail voor Android downloaden ​

VitGGGGGGGGGGGGGGGGGGGG, 2025-02-05:
> The last sentence is not completely correct.
> 

 ok. the Pluto versions that I had did not have diodes and I made a conclusion based on my experience. if they started installing diodes later, then apparently there were some reasons for this. but I personally do not see the need for them. if the original baluns from minicircut are installed, then they have all outputs connected to the ground via DC and the input will not be able to accumulate static charge. if you apply a powerful high-frequency signal to the input, then the zener most likely will not help and will burn out along with the rf chip =)
the necessary zener diodes are available in radio stores, look for them by the name esd8472mut5g. they should be marked Z3 and you will need to gain a lot of experience to solder them to the board =)

VitGGGGGGGGGGGGGGGGGGGG, 2025-02-05:
two screenshots of LibreSDR working with external synchronization connected and without it. 

[image]

[image]

in the first case, the received signal is periodically shifted. I tried to redesign the PLL circuit to reduce the adjustment step using attenuation between the DAC and the TCXO, but nothing helped, it only got worse. this is a bad software module or a schematic design error

lubuntu-l100, 2025-02-05:
Thanks for the pictures.

I have tried to use an external 40 mhz signal at the 40 Mhz vtxcxo output, by disabling the vtxcxo. The problem was that the pll circuit stops to run. Another option which I want to test up coming weeks, is to bring the external 40 mhz to the external clock input of the Ad9363 and define the external clock input and frequency with putty for the Ad9363.

If that will run, the stabilisation problem is solved. It is the stability of the 40 mhz signal at the ad9363 that gives drifting at 2.4 ghz when transmitting in ssb over the QO100 satellite in a bandwidth of 2.8 kc.

My 40 mhz vtxcxo is about 5.5 khz of frequency at 2.4 Ghz and drifts about 500 hz over a time period of 4 hours. But for ssb this is a real problem.

⁣BlueMail voor Android downloaden ​

Notupus, 2025-02-05:
So there is a reference clock implementation in tezuka apparently?

VitGGGGGGGGGGGGGGGGGGGG, 2025-02-05:
I haven't checked the latest version yet, but it wasn't in the previous one.

Notupus, 2025-02-05:
So how did you get even that broken version working? I reckon I can try to do it like the b210 does it. This and timestamping would make libre basically a b210-mini with a bw limit.

VitGGGGGGGGGGGGGGGGGGGG, 2025-02-05:
I can't remember where, but I found a firmware for SPI flash to work without a SD card. It has a name libreplutosdr-fw-v0.37-1-gf5ad-dirty.zip. I loaded pluto.frm and boot.frm into SPI and my LibreSDR works with UTP, USB and supports external sync 10 MHz when loaded without a SD card. I haven't seen  support ext clk in any other firmware, and I don't know who made it. =(

lubuntu-l100, 2025-02-05:
I have tested this libreplutosdr-fw-v0.37-1-gf5ad-dirty.zip version however what I noticed that the 40 Mhz vctcxo is drifting from left to right around the required frequency without a proper lock. I have exported the 10 mhz lock led to see if there is a lock. But the lock led is not on.

It look likes that the fpga is looking for the frequency correction without the correct reference. Even without the external 10 mhz the 40 mhz vctcxo was drifting all the time around 40 mhz. I akso checked the pps external, also without succes.

I also tested an unknown Chinese firmware for the Libresdr. As a pluto clone firmware. In that version 10 mhz external clock led was directly on at startup, however the fpga was not running. So that fw was not working for this version of fpga.

⁣BlueMail voor Android downloaden ​

Notupus, 2025-02-05:
Okay when I get mine I will test a few stuff and first try to do something for the vctcxo with the pps because thats easier to get, to make pps work on the rising edge  so it works ootb with radiosondes like rs41 as gpsdo some sort of interrupt on that could do the job though I still have no idea how to program fpgas and my libresdr will be the first fpga I can play with. This and timestamping is what I want to work on libresdr by end of February  or soon ™

lubuntu-l100, 2025-02-05:
Be aware that the LibreSDR is more or less an Ettus B210 clone instead of an AD Pluto clone. Especially around the 10 Mhz external clock reference combined with the FPGA.

⁣BlueMail voor Android downloaden ​

Notupus, 2025-02-05:
yeah I figured , otherwise it would appear too simple. I have schematics , original adi whatever adi is( like a 30 gb uncompressed directory). Would love the sync to be fpga only but doubt how possible that would be. I still have way too many questions to figure out. I think I can do the sync stuff not sure about timestamps, would be interesting to hear f50e0's input on this, and maybe a sneakpeak of what they're working on.

lubuntu-l100, 2025-02-05:
Go to F5OEO/Tezuka on github. There you can find the current developments as well the future.

Also on Github the Maia Sdr project.

⁣BlueMail voor Android downloaden ​

FPSychotic, 2025-02-06:
> No it is not. Well, yes it is but also no it is not. There are two, maybe
> more LibreSDR models, one is a B210 clone, the other a Pluto clone. I
> expect neither is the same as the model they copy. The RF front end may be
> the same in both cases, or maybe not.
>
>
>
> Mike

Yes,.the last model labeled by LibreSDR is a B210 clone, likely a
B210mini/BlackSDR , but likely cutted down in the external reference input,
only one Ipex connector vs 2 Ipex in BlackSDR.
They differ from Ettus one in they use a Artix7 instead a Spartan. So need
a special file to work with Ettus driver. It has not CPU built in as the
Zynq,.so.they are very different. The BlackSDR use USB3.0, AdiXX91 instead
xx93 unlocked, and it really gets the 61mhz bandwidth.
Any other LibreSDR are pluto based,.very different

FPSychotic, 2025-02-06:
I had good luck  I have the good one with holes everywhere and z3 diode .
Also I include picture of the XO in case is different.
What I can see the baluns are not the mini-circuit rates to 6ghz that I installed in B210mini, those in LibreSDR are 3-4ghz rated I think, installing the 6ghz baluns I think would improve.
Also from LibreVNA devs, I dead they removed a wind from the balun, and works much better and reduce cross-talking, will try find the conversation, maybe is possible improve further the SDR. Mine performed as good as original pluto even better in some aspects.

[image]
[image]
[image]
[image]

Notupus, 2025-02-06:
First of all where did you buy this and secondly what sort of balun mod are you talking about

FPSychotic, 2025-02-06:
> First of all where did you buy this and secondly what sort of balun mod are you talking about If its as easy as removing a winding this should a an easy hw mod, also I am very glad you are enjoying your libresdr
I bought this in AliExpress in TZT shop like 1 year ago or more. PCB says V5.
It was quite expensive and need paid customs twice.
Ended in £250 or so.
But careful now that shop I thing is selling the hamgeek version, I don't think you will receive this one. Also this was previous to hamgeek or OpensoureLabs versions. 
Those are the baluns I swapped in my.b210mini that came with chinese branded clone of the Mini-circuits. They comes marked by GU.
The ones that comes in LibreSDR, are the same than PlutoSDR, they are rated 3-4ghz, this ones I installed 6ghz.

FPSychotic, 2025-02-06:
[image]
[image]
[image]
[image]

lubuntu-l100, 2025-02-06:
Your pictures are showing the Libresdr b210 clone.

⁣BlueMail voor Android downloaden ​

FPSychotic, 2025-02-06:
> Your pictures are showing the Libresdr b210 clone.
> 
> ⁣BlueMail voor Android downloaden ​
> 
> Op 6 feb 2025 21:01, om 21:01, FPSychotic ***@***.***> schreef:
> […](#)

This last pictures are BlackSDR/B210Mini.
It is different to LibreslSDR B210, specially the PLL part, mine has two ipex, libresdr b210 has one. 
BlackSDR has ADxx91, I think libreSDR has ADxx93 as pluto, at least some of them

VitGGGGGGGGGGGGGGGGGGGG, 2025-02-07:
I have a separate 9361 chip, if I replace 9363 with it in Libre SDR, will it work or do I need to make a special firmware?

F5OEO, 2025-02-07:
Libresdr firmware assume already that it is ad9361...so it works without other modification on firmware.

VitGGGGGGGGGGGGGGGGGGGG, 2025-02-07:
according to the datasheet, several pins of these chips do not match. It probably won't be possible to do this by simply replacing them.  =(
software substitution of 9361 instead of 9363 is most likely not the same thing. i.e. in fact we are simply expanding the range of operating frequencies, but what about software support the bandwidth range and the LO tuning step?

G0MJW, 2025-02-07:
That’s interesting. Will it do 10m directly for DATV now?

 

Sample Rate is set to 61.4MS/56Mhz bandwidth, LO is extended to 47Mhz-6Ghz even with ad9363

F5OEO, 2025-02-07:
DATV on 10m  : Need LO at 47Mhz, and Symbolrate at lease 47Mhz-28Mhz -> around 20MS. Then use NCO and interpolator : that's it.

G0MJW, 2025-09-28:
> Hello everyone, I am the founder of the Libresdr brand. Libresdr currently has three signals: one is zynqsdr (plutosdr), one is the compatible B210mini/B220mini, and the other is the recent PCIe m2sdr. As you know, Chinese people like to copy, but I never expected there to be copying of copying. My products comply with open-source licenses and also include a little bit of my own design.
>
> 2025-08-25.164837.png (view on web) <https://github.com/user-attachments/assets/948de624-71e9-490b-9d00-a540c10eaa24>
>
> 257849884.jpg (view on web) <https://github.com/user-attachments/assets/4f6c552a-7e69-46a8-bcfb-c465eb2fae6a>
>
> 1283975430.jpg (view on web) <https://github.com/user-attachments/assets/ab7b89b1-3b4a-4fa1-ae9c-78badec83083>

How can we tell which are genuine and which are copies? I really like the CNC cases and functionality of the LibreSDR, I have two, but they are different. So one is a copy.

G0MJW, 2025-09-28:
> The one with the LibreSDR logo on the casing is original, and the other one, which has the 40M crystal mentioned above as a VCTCXO, is also original.
>
> 不懂技术也得工作啊

There are two Logos..

 

 

LIbresDR and LibreSDR

F5OEO, 2025-09-28:
Welcome to @jiskra as designer of LibreSDR. I admit that this is the board which is the most difficult to adapt for Tezuka firmware : many copies such a Zynqsdr...all boards are not equal (mine for example has a TXCO instead a VCTXO which does not allow me to have a ref clock source).
Original source firmware is not clear, specially linux dts and fpga bitstream.

F5OEO, 2025-09-28:
Do you think it is possible to get a free sample of the original LibreSDR for be properly supported by Tezuka Firmware ?

jiskra, 2025-09-28:
> Please send me a mail : ***@***.***
> 
> Le dim. 28 sept. 2025 à 12:40, 吕晨阳 ***@***.***> a écrit :
> […](#)

I can't see the email address, my email is i@jiskra.net

VitGGGGGGGGGGGGGGGGGGGG, 2025-09-28:
> Thank you！
> All that do not support VCTCXO are incorrect.

Hi! Is it possible to set up external frequency synchronization in Libresdr so that the frequency doesn't constantly jump around the operating frequency?

76nic, 2025-09-28:
Dear @jiskra & @F5OEO,

It is nice to see "hardware meeting software"! I do have a LibreSDR/ZynqSDR and I like it very much (nice job jiskra!). Up to now the firmware was a big weak point, but now with tezuka that has changed!!! Without good and well maintained firmware the best SDR hardware design is worthless. Therefore, I'm really happy to see you F5OEO working on it. I'm very thankful for what you are doing and wish you strong support from the SDR community. I hope your firmware/software will further mature, fixing the current weaknesses/bugs and covering other hardware features like external clock reference or the Cortex processor.

@jiskra: you have mentioned your new "PCIe m2sdr", could you please share a link to get more information on it?

Many thanks for all your commitment to SDR hardware and - very important - software!!!
Best regards,
DC0NG

jiskra, 2026-01-18:
Can't guarantee for sure, but watch out for second-hand chips.

不懂技术也得工作啊
***@***.***

  
 
 So the one I have is one of yours ? Excellent.

G0MJW, 2026-01-19:
> As a retired chip designer and have a 401k with ADI, I can tell you that it is very unlikely anyone is building boards with second hand chips. A second hand chip would require desoldering from some existing product and creating tape and reel. That would be a lot of work.
>
> If you meant second source chips, then maybe true for the digital products. I can tell you no analog chip I designed had a successful second source. (One attempt in decades of design.) Only one attempt and they gave up. It is really hard to copy analog mixed mode chips let alone SiGe.
>
> Third, ADI has been in China since 2020 to head off second sources.
>
> You can criticize poor assembly but I am doubtful of second hand or second source analog chips.

I suppose this is the difference between US English and English english.  Second hand doesn’t mean used, it just means you bought it and sold it on. Maybe you used half the stock for a project and sold on the rest. I am not sure what 401k has to do with this. Can’t be resistance, it must be a US qualification.

 

Mike

jiskra, 2026-01-19:
We are in China, so we understand China better. When we say second-hand, we mean chips that are dismantled from other discarded electronic products and re-balled, which is why some products can be so cheap.

不懂技术也得工作啊
***@***.***