F5OEO/tezuka_fw #47: Libresdr; Applying only powersupply there is no  ethernet connection

pa3gsb, 2025-01-30:
Hi,

I use the latest firmware running from sd card.

Applying only powersupply and connect the ethernet to PC.

It is not possible to ping to the libresdr. Using a port scanner the libresdr is found  but no active services are found; so no ping or other connection can be made.

Connect the PC via usb and apply the power via usb there is a connection.

Are there ideas where to look?

sq6emm, 2025-01-30:
First ethernet cable, then power on. DHCP client does not work in the background if fails.

sq6emm, 2025-01-30:
tezukafw is a DHCP client sob your comfort should be DHCP server.... Try connection to a network  router/switch

pa3gsb, 2025-01-30:
i have done that as well. Also without luck.

F5OEO, 2025-01-31:
If there is an issue with dhcp, you can try to set static ip address to eth : 
- Boot with USB
- Edit config.txt, modify fields with a proper ip address

[USB_ETHERNET]
ipaddr_eth = 
netmask_eth = 255.255.255.0
gateway_eth = 

-Eject

Reboot without USB

F5OEO, 2025-02-01:
Can you try with the latest 0.0.95 version ?

pa3gsb, 2025-02-01:
I have tried your latest 95 version.  In the router i see that the libre is getting an ip address.  

But iam not able to ping or work with the device; when using the port scan i do no service / ports coming available.

i have tried a dynamisch and static address both are not responding to pings.

Only having the usb connected; it is working.

F5OEO, 2025-02-01:
Can you plug the serial usb cable (power) and monitor messages (even ssh in it) : this is serial link at 115200.

pa3gsb, 2025-02-02:
setup:
- modified an usb cable; removed the 5V line. connect this modified cable to pc for putty terminal.
- connect the ethernet cable. 
- connect the usb - otg to *only* power supply.

Result:

- no ping to libresdr.

- libre sdr got ip address. but no services / ports available on the interface / ipaddress.

see the log:

Welcome to Tezuka
libresdr login:
Welcome to Tezuka
libresdr login:
Welcome to Tezuka
libresdr login: analog
Password:
Login incorrect
libresdr login: root
Password:

___________                  __
\__    ___/___ __________ __|  | _______
  |    |_/ __ \\___   /  |  \  |/ /\__  \
  |    |\  ___/ /    /|  |  /    <  / __ \_
  |____| \___  >_____ \____/|__|_ \(____  /
             \/      \/          \/     \/
Universal Zynq70x/AD936x firmware builder
Version tezuka-3c71
By F5OEO (2024)
# reboot
# Stopping input-event-daemon: done
Stopping
Stopping NFS
umount: can't unmount /mnt/nfs: Invalid argument
Stopping SD
Stopping maia-httpd: OK
Starting host keys backup: Stopping mosquitto: OK
Stopping dropbear sshd: OK
Stopping crond: no /usr/sbin/crond found; none killed
FAIL
Stopping ntpd: OK
Stopping MSD Daemon: OK
Stopping dhcpd Daemon & httpd Server: FAIL
Stopping ifplugd for eth0: Stopping network: ifdown: interface usb1 not configured
macb e000b000.ethernet eth0: Link is Down
OK
Stopping system message bus: done
Stopping UDC Gadgets
unloading
Stopping klogd: OK
Stopping syslogd: stopped /sbin/syslogd (pid 91)
OK
Seeding 256 bits without crediting
Saving 256 bits of creditable seed for next boot
ffs_data_put(): freeing
umount: devtmpfs busy - remounted read-only
umount: can't unmount /: Invalid argument
The system is going down NOW!
Sent SIGTERM to all processes
Sent SIGKILL to all processes
Requesting system reboot
reboot: Restarting system

U-Boot 2016.07 (Jan 31 2025 - 23:02:45 +0000)

I2C:   ready
[... 361 lines cut ...]
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:974 errors:0 dropped:0 overruns:0 frame:0
          TX packets:120 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:52770 (51.5 KiB)  TX bytes:10925 (10.6 KiB)
          Interrupt:36 Base address:0xb000

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:88 errors:0 dropped:0 overruns:0 frame:0
          TX packets:88 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:8958 (8.7 KiB)  TX bytes:8958 (8.7 KiB)

usb0      Link encap:Ethernet  HWaddr 00:05:F7:10:13:EC
          inet addr:192.168.2.100  Bcast:0.0.0.0  Mask:255.255.255.0
          UP BROADCAST MULTICAST  MTU:1500  Metric:1
          RX packets:0 errors:0 dropped:0 overruns:0 frame:0
          TX packets:0 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:0 (0.0 B)  TX bytes:0 (0.0 B)

#

pa3gsb, 2025-02-02:
found the problem 

# ip route
default via 192.168.2.254 dev eth0
192.168.2.0/24 dev usb0 scope link  src 192.168.2.100
192.168.2.0/24 dev eth0 scope link  src 192.168.2.146

removed one route:
sudo ip route del 192.168.2.0/24 dev usb0

now the ping is working (both sides)

how to avoid to get a double entry?

sq6emm, 2025-02-02:
The problem is that the default ip addressing of Ethernet over USB is 192.168.2.0/24 and it seems your local network has the same IP addressing... So you need to reconfigure in config.txt that the Ethernet over USB is on another network ;)

pa3gsb, 2025-02-02:
I solved the problem; 

Used indeed a seperate internet mask for the ethernet and usb ethernet ip address.

So the versie 95 is working ok.

Thanks for the support.

closed 2025-02-02 by pa3gsb