F5OEO/tezuka_fw #48: Original libresdr sourcecode

F5OEO, 2025-01-31:
It seems that an old firmware is available from a taobao manufacturer.
In order to check all orginal details , help wanted to get this archive (surely quiet big).

johnsonZ72, 2025-01-31:
http://185.154.180.252/adam.img.xz
user1
Cmh7seZ9

xzcat adam.img.xz | dd of=/dev/sdd bs=1M status=progress

F5OEO, 2025-01-31:
Thanks for the link, but look for orginal sources, means complete sources with buildroot/hdl , not the resulting image.

sq6emm, 2025-02-02:
OpenSourceSDRLab shared with me the following links, as official ones:

https://github.com/day0wl/libresdr-fw
https://workupload.com/archive/4QAz3LL2HT

F5OEO, 2025-02-06:
All the above links (appart https://github.com/day0wl/libresdr-fw) are not sourcecode but mainly firmware or kuipper image.
I look paricularly for a reference with HDL source. It is mentionned [here](https://github.com/sdy623/hdl/commit/5b9bcef97dec9202b1c5d43871a1ca2c543c0460) that a  22.5GB archive exists (ADI.zip, available on the support in Taobao shops, distributed by BaiduNetDisk). If anyone has this archive, it could be very helpfull.

tri4ng1e, 2025-02-06:
> I look paricularly for a reference with HDL source

Check the last link from previous message (terabox one). It contains a ~20GB ADI.rar archive, split into 4 parts. Maybe that's the one you are looking for?

Terabox is far from pleasant experience, requiring an account and a PC app to download the archive at around 3MB/s. I made a mirror for those files, [you can try it here](https://url.pytq.ru/adi_terabox_mirror). Link will work until next Monday.

F5OEO, 2025-02-06:
Thanks, @tri4ng1e : get it and is the one I was looking for !

FPSychotic, 2025-02-06:
Hi, any of this images are the ones that works with DJI video system ?

F5OEO, 2025-02-06:
No DJI is only for e200 I think and there is no sourcecode. YOu can get the firmware [here](https://github.com/alphafox02/antsdr_dji_droneid)

sq6emm, 2025-02-06:
@F5OEO do you plan to add DVB-S transmitter (qo-100) feature in tezuka?

F5OEO, 2025-02-13:
Completed with links provided. Thank you !

closed 2025-02-13 by F5OEO