F5OEO/tezuka_fw #49: LibreSDR - how to enable PTT?

sq6emm, 2025-02-01:
Anyone already tried to expose PTT from LibreSDR?
How to do that? from where?
Do I need to enable that in the firmware? (is this already supported?)

pa3gsb, 2025-02-06:
i have connected to the otg port a arduino pro micro.

The micro should get the echo from the libre and must control the PTT.

The terminal shows me 👍 # dmesg | grep -i usb
usbcore: registered new interface driver usbfs
usbcore: registered new interface driver hub
usbcore: registered new device driver usb
usbcore: registered new interface driver rt2500usb
usbcore: registered new interface driver rt73usb
usbcore: registered new interface driver rt2800usb
usbcore: registered new interface driver rtl8187
usbcore: registered new interface driver rtl8192cu
usbcore: registered new interface driver rtl8xxxu
usbcore: registered new interface driver rtl8150
usbcore: registered new interface driver r8152
usbcore: registered new interface driver lan78xx
usbcore: registered new interface driver asix
usbcore: registered new interface driver ax88179_178a
usbcore: registered new interface driver cdc_ether
usbcore: registered new interface driver dm9601
usbcore: registered new interface driver smsc75xx
usbcore: registered new interface driver smsc95xx
usbcore: registered new interface driver rndis_host
usbcore: registered new interface driver r8153_ecm
usbcore: registered new interface driver usb-storage
usbcore: registered new interface driver usbserial_generic
usbserial: USB Serial support registered for generic
usbcore: registered new interface driver ftdi_sio
usbserial: USB Serial support registered for FTDI USB Serial Device
ci_hdrc ci_hdrc.0: new USB bus registered, assigned bus number 1
ci_hdrc ci_hdrc.0: USB 2.0 started, EHCI 1.00
usb usb1: New USB device found, idVendor=1d6b, idProduct=0002, bcdDevice= 6.01
usb usb1: New USB device strings: Mfr=3, Product=2, SerialNumber=1
usb usb1: Product: EHCI Host Controller
usb usb1: Manufacturer: Linux 6.1.0 ehci_hcd
usb usb1: SerialNumber: ci_hdrc.0
hub 1-0:1.0: USB hub found
usbcore: registered new interface driver usbhid
usbhid: USB HID core driver
usbcore: registered new interface driver r8712u
usbcore: registered new interface driver r8188eu
usb 1-1: new full-speed USB device number 2 using ci_hdrc
usbcore: registered new interface driver snd-usb-audio
usb 1-1: New USB device found, idVendor=2341, idProduct=8037, bcdDevice= 1.00
usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 1-1: Product: Arduino Micro
usb 1-1: Manufacturer: Arduino LLC
usb0: HOST MAC 00:e0:22:c6:c5:50
usb0: MAC 00:05:f7:10:13:ec

So the libre sdr is the host; and the arduino is the slave.

# ls /dev/tt*

i do not see a serial device. 

We need to add a driver to make this work. Need to figure how to make this work. Are there ideas?

73 Johan
PA3GSB

F5OEO, 2025-02-06:
Should be fixed with 83012523ccaf90bf7b74543ef036a68763e28e8e

pa3gsb, 2025-02-08:
Tnx i rebuild the image and i connect a arduino with a sketch and iam able to enable or disable the tx.

Having now an additonal serial port for specific controls i want to solve this by modifying the sh script:

DEVICE="/dev/ttyACM0"
...

if [ -e "$DEVICE" ]; then 
	echo "PTT ON" > "$DEVICE" 
fi

.... 
if [ -e "$DEVICE" ]; then 
	echo "PTT OFF" > "$DEVICE" 
fi

Do you agree this approach? and are you willing to accept pull requests or do you modify the file yourself (it is only a small change).

pa3gsb, 2025-02-09:
https://github.com/F5OEO/tezuka_fw/pull/53

pa3gsb, 2025-02-27:
issue solved. https://github.com/F5OEO/tezuka_fw/pull/75

F5OEO, 2025-02-27:
Could be interesting to continue with arduino or pi pico to control libre trough mqtt commands over serial !

DG9BFC, 2025-03-23:
lesha108 has ptt on gpo pin (and dfu botton works and led shows a heartbeat)