F5OEO/tezuka_fw #51: 0.0.95 building failed

Szczecin-Radio, 2025-02-06:
Hello,

Just informational, it seems there is something missing (i.e. maia-httpd.xz) in sources:

wojtek@void:/opt/tmp/tezuka_fw/buildroot-2024.08$ make pluto_maiasdr_defconfig && make
#
# configuration written to /opt/tmp/tezuka_fw/buildroot-2024.08/.config
#
/bin/make -j1  O=/opt/tmp/tezuka_fw/buildroot-2024.08/output HOSTCC="/bin/gcc" HOSTCXX="/bin/g++" syncconfig
make[1]: Entering directory '/opt/tmp/tezuka_fw/buildroot-2024.08'
make[1]: Leaving directory '/opt/tmp/tezuka_fw/buildroot-2024.08'
xz: /opt/tmp/tezuka_fw/buildroot-2024.08/output/build/maia-httpd-main//maia-httpd/target/armv7-unknown-linux-gnueabihf/release/maia-httpd: No such file or directory
>>> maia-httpd main Installing to target
/bin/install -D /opt/tmp/tezuka_fw/buildroot-2024.08/output/build/maia-httpd-main//maia-httpd/target/armv7-unknown-linux-gnueabihf/release/maia-httpd.xz /opt/tmp/tezuka_fw/buildroot-2024.08/output/target/usr/bin/
/bin/install: cannot stat '/opt/tmp/tezuka_fw/buildroot-2024.08/output/build/maia-httpd-main//maia-httpd/target/armv7-unknown-linux-gnueabihf/release/maia-httpd.xz': No such file or directory
make: *** [package/pkg-generic.mk:380: /opt/tmp/tezuka_fw/buildroot-2024.08/output/build/maia-httpd-main/.stamp_target_installed] Error 1

Regards,
/WS

F5OEO, 2025-02-06:
Have you xz installed on your machine ?

Szczecin-Radio, 2025-02-06:
Yes, I have.

wojtek@void:~$ rpm -q xz
xz-5.4.6-3.fc40.x86_64

wojtek@void:~$ cat /etc/redhat-release 
Fedora release 40 (Forty)
root@void:~# 

I have a lot of *.xz packages in tezuka_fw/buildroot-2024.08/* subdirs, but said "maia-httpd.xz" is definitely missing. Did I miss something before building the firmware or forgot to download some extra sources or so? The firmware building has failed at the finish.

Regards,
/WS

F5OEO, 2025-02-11:
Seems that building with wasm has issue. I am currently upgrading to latest buildroot, maybe will be better.

F5OEO, 2025-02-13:
Please try 0.1.0..reopen if needed.

closed 2025-02-13 by F5OEO