F5OEO/tezuka_fw #52: build error 0.95

gariac, 2025-02-09:
I have the entire build output saved but these are the errors.

`
[ERROR]  : Cannot read file - /usr/local/src/tezuka_fw/buildroot-2024.08/output/images/fsbl.elf
generating the boot.frm
cat: /usr/local/src/tezuka_fw/buildroot-2024.08/output/images/boot.img: No such file or directory
b210cd40c5fd904492d00ea1652fc6a2
generating boot.dfu
cp: cannot stat '/usr/local/src/tezuka_fw/buildroot-2024.08/output/images/boot.img': No such file or directory
dfu-suffix (dfu-util) 0.11

`
`
dfu-suffix: Could not open file /usr/local/src/tezuka_fw/buildroot-2024.08/output/images/boot.bin.tmp for reading: No such file or directory
mv: cannot stat '/usr/local/src/tezuka_fw/buildroot-2024.08/output/images/boot.bin.tmp': No such file or directory
generating uboot-env.dfu
dfu-suffix (dfu-util) 0.11

`

`
****** Xilinx Bootgen v2022.2
  **** Build date : Mar  9 2024-23:05:33
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.

[WARNING]: Partition fsbl.elf.0 range is overlapped with partition system_top.bit.0 memory range

[INFO]   : Bootimage generated successfully

`

F5OEO, 2025-02-09:
Can you check your dependencies (see example at https://github.com/F5OEO/tezuka_fw/blob/main/.github/workflows/main.yml)
You surely have some prior issue of this log , in order to investigate where is the missing file

gariac, 2025-02-10:
The output from the build is 62K lines. Too big for one pastebin.
 
I have met the dependencies of that yamal file. I have never run such a file but it was easy enough to extract the line with the dependencies. I have a slight variation using Debian 12 versus Ubuntu. For example I use Xilinux-bootgen and libncurses6 rather than 5. 

Would you like me to chop up the output from the build into multiple pastebins?

gariac, 2025-02-11:
How about we do this one at a time. There is no line in the make output that contains the file name boot.bin.tmp.

`
dfu-suffix: Could not open file /usr/local/src/tezuka_fw/buildroot-2024.08/output/images/boot.bin.tmp for reading: No such file or directory
mv: cannot stat '/usr/local/src/tezuka_fw/buildroot-2024.08/output/images/boot.bin.tmp': No such file or directory
`

F5OEO, 2025-02-11:
Should be fixed with 803c50a14aa91bf005407e2a30581767b28916a0
Please confirm

gariac, 2025-02-12:
I patched those files. I never could figure out where the patch file resided so I just copied the raw file and replace the specific file in all four cases.

Now I assume I just do 
`
make pluto_maiasdr_defconfig && make
`
else the patched code will be overwritten.  So is that correct?

F5OEO, 2025-02-13:
Try with 0.1.0, reopen if needed.

closed 2025-02-13 by F5OEO