F5OEO/tezuka_fw #54: Bricked adalm-pluto after boot.frm and pluto.frm upload.

sq6emm, 2025-02-11:
Hey I have just made a build with pluto_maiasdr.

How can I recover it please?:)

And now my pluto does not provide any device on main USB port, on the debug I get:

```

[47063.380670] usb 5-1: new full-speed USB device number 5 using xhci_hcd
[47063.537695] usb 5-1: New USB device found, idVendor=0403, idProduct=6015, bcdDevice=10.00
[47063.537704] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[47063.537707] usb 5-1: Product: FT230X Basic UART
[47063.537710] usb 5-1: Manufacturer: FTDI
[47063.537712] usb 5-1: SerialNumber: DM03N4P6
[47063.598803] usbcore: registered new interface driver ftdi_sio
[47063.598821] usbserial: USB Serial support registered for FTDI USB Serial Device
[47063.598926] ftdi_sio 5-1:1.0: FTDI USB Serial Device converter detected
[47063.598967] usb 5-1: Detected FT-X
[47063.609002] usb 5-1: FTDI USB Serial Device converter now attached to ttyUSB1
[47155.425685] usb 5-1: USB disconnect, device number 5
[47155.426299] ftdi_sio ttyUSB1: FTDI USB Serial Device converter now disconnected from ttyUSB1

```

This are the last lines of build:

```

>>> maia-wasm main Building
>>> maia-wasm main Installing to target
mkdir -p /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/target/root/pkg
cp -r /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/build/maia-wasm-main//maia-wasm/pkg/* /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/target/root/pkg/
cp -r /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/build/maia-wasm-main//maia-wasm/assets/* /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/target/root/
>>> maiasdr-fw  Extracting
>>> maiasdr-fw  Patching
>>> maiasdr-fw  Configuring
>>> maiasdr-fw  Building
>>> maiasdr-fw  Installing to target
>>>   Finalizing host directory
>>>   Finalizing target directory
mkdir -p /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/host/etc/meson
sed -e "s%@TARGET_CFLAGS@%'-D_LARGEFILE_SOURCE', '-D_LARGEFILE64_SOURCE', '-D_FILE_OFFSET_BITS=64', '-O3', '-g0', '-D_FORTIFY_SOURCE=1'@PKG_TARGET_CFLAGS@%g" -e "s%@TARGET_LDFLAGS@%@PKG_TARGET_LDFLAGS@%g" -e "s%@TARGET_CXXFLAGS@%'-D_LARGEFILE_SOURCE', '-D_LARGEFILE64_SOURCE', '-D_FILE_OFFSET_BITS=64', '-O3', '-g0', '-D_FORTIFY_SOURCE=1'@PKG_TARGET_CXXFLAGS@%g" -e "s%@TARGET_FCFLAGS@%'-O3', '-g0'@PKG_TARGET_FCFLAGS@%g"         -e "s%@TARGET_CC@%/home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/host/bin/arm-none-linux-gnueabihf-gcc%g" -e "s%@TARGET_CXX@%/home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/host/bin/arm-none-linux-gnueabihf-g++%g" -e "s%@TARGET_AR@%/home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/host/bin/arm-none-linux-gnueabihf-gcc-ar%g" -e "s%@TARGET_FC@%/home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/host/bin/arm-none-linux-gnueabihf-gfortran%g" -e "s%@TARGET_STRIP@%/home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/host/bin/arm-none-linux-gnueabihf-strip%g" -e "s%@TARGET_ARCH@%arm%g" -e "s%@TARGET_CPU@%cortex-a9%g" -e "s%@TARGET_ENDIAN@%little%g" -e "s%@TARGET_FCFLAGS@%%g" -e "s%@TARGET_CFLAGS@%%g" -e "s%@TARGET_LDFLAGS@%%g" -e "s%@TARGET_CXXFLAGS@%%g" -e "s%@BR2_CMAKE@%/usr/bin/cmake%g" -e "s%@PKGCONF_HOST_BINARY@%/home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/host/bin/pkgconf%g" -e "s%@HOST_DIR@%/home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/host%g" -e "s%@STAGING_DIR@%/home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/host/arm-buildroot-linux-gnueabihf/sysroot%g" -e "s%@STATIC@%false%g" /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/support/misc/cross-compilation.conf.in > /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/host/etc/meson/cross-compilation.conf.in
sed         -e "s%@TARGET_CC@%/home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/host/bin/arm-none-linux-gnueabihf-gcc%g" -e "s%@TARGET_CXX@%/home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/host/bin/arm-none-linux-gnueabihf-g++%g" -e "s%@TARGET_AR@%/home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/host/bin/arm-none-linux-gnueabihf-gcc-ar%g" -e "s%@TARGET_FC@%/home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/host/bin/arm-none-linux-gnueabihf-gfortran%g" -e "s%@TARGET_STRIP@%/home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/host/bin/arm-none-linux-gnueabihf-strip%g" -e "s%@TARGET_ARCH@%arm%g" -e "s%@TARGET_CPU@%cortex-a9%g" -e "s%@TARGET_ENDIAN@%little%g" -e "s%@TARGET_FCFLAGS@%'-O3', '-g0'%g" -e "s%@TARGET_CFLAGS@%'-D_LARGEFILE_SOURCE', '-D_LARGEFILE64_SOURCE', '-D_FILE_OFFSET_BITS=64', '-O3', '-g0', '-D_FORTIFY_SOURCE=1'%g" -e "s%@TARGET_LDFLAGS@%%g" -e "s%@TARGET_CXXFLAGS@%'-D_LARGEFILE_SOURCE', '-D_LARGEFILE64_SOURCE', '-D_FILE_OFFSET_BITS=64', '-O3', '-g0', '-D_FORTIFY_SOURCE=1'%g" -e "s%@BR2_CMAKE@%/usr/bin/cmake%g" -e "s%@PKGCONF_HOST_BINARY@%/home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/host/bin/pkgconf%g" -e "s%@HOST_DIR@%/home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/host%g" -e "s%@STAGING_DIR@%/home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/host/arm-buildroot-linux-gnueabihf/sysroot%g" -e "s%@STATIC@%false%g" /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/support/misc/cross-compilation.conf.in > /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/host/etc/meson/cross-compilation.conf
/usr/bin/sed -i -e '/# GENERIC_SERIAL$/s~^.*#~ttyPS0::respawn:/sbin/getty -L  ttyPS0 0 vt100 #~' /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/target/etc/inittab
/usr/bin/sed -i -e '/^#.*-o remount,rw \/$/s~^#\+~~' /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/target/etc/inittab
if grep -q CONFIG_ASH=y /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/build/busybox-1.36.1/.config; then grep -qsE '^/bin/ash$' /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/target/etc/shells || echo "/bin/ash" >> /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/target/etc/shells; fi
if grep -q CONFIG_HUSH=y /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/build/busybox-1.36.1/.config; then grep -qsE '^/bin/hush$' /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/target/etc/shells || echo "/bin/hush" >> /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/target/etc/shells; fi
mkdir -p /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/target/etc
echo "libresdr" > /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/target/etc/hostname
/usr/bin/sed -i -e '$a \127.0.1.1\tlibresdr' -e '/^127.0.1.1/d' /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/target/etc/hosts
mkdir -p /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/target/etc
echo "Welcome to Tezuka" > /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/target/etc/issue
/usr/bin/sed -i -e s,^root:[^:]*:,root:"`/home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/host/bin/mkpasswd -m "sha-256" "analog"`":, /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/target/etc/shadow
grep -qsE '^/bin/sh$' /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/target/etc/shells || echo "/bin/sh" >> /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/target/etc/shells
if [ -x /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/target/sbin/swapon -a -x /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/target/sbin/swapoff ]; then /usr/bin/sed -i -e '/^#.*\/sbin\/swap/s/^#\+[[:blank:]]*//' /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/target/etc/inittab; else /usr/bin/sed -i -e '/^[^#].*\/sbin\/swap/s/^/#/' /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/target/etc/inittab; fi
if test -d /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/target/lib/modules/`MAKEFLAGS='--no-print-directory' /usr/bin/make -j17 HOSTCC="/usr/bin/gcc -O2 -isystem /home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/host/include -L/home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/host/lib -Wl,-rpath,/home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/host/lib" ARCH=arm INSTALL_MOD_PATH=/home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/target CROSS_COMPILE="/home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/host/bin/arm-none-linux-gnueabihf-" WERROR=0 REGENERATE_PARSERS=1 DEPMOD=/home/dawszy/git/original/tezuka_fw/buildroot-2024.08/output/host/sbin/depmod INSTALL_MOD_STRIP=1 -C /home/dawszy/git/origi [... cut ...]
Data Size:    20407965 Bytes = 19929.65 KiB = 19.46 MiB
Load Address: 00000000
Entry Point:  00000000
	zip warning: name not matched: boot.dfu
  adding: boot.frm (deflated 98%)
  adding: pluto.frm (deflated 7%)
  adding: pluto.dfu (deflated 7%)
  adding: sdimg/BOOT.bin (deflated 75%)
  adding: sdimg/devicetree.dtb (deflated 73%)
  adding: sdimg/overclock/ (stored 0%)
  adding: sdimg/uEnv.txt (deflated 72%)
  adding: sdimg/uImage (deflated 7%)
  adding: sdimg/uramdisk.image.gz (deflated 1%)

```

F5OEO, 2025-02-11:
If you use libresdr, you need to build make libre_maiasdr_defconfig not pluto.
To unbrick, use a "ready build firmware" and write on sd. SD should boot without issue.

F5OEO, 2025-02-11:
On PlutoRev C, you can't brick it as boot is protected on flash.
You could use dfu procedure to reflash the firmware and bootenv.

sq6emm, 2025-02-11:
I can't... As dfu-util is not detecting my Pluto as dfu device....

F5OEO, 2025-02-11:
- Power up pluto with little push button "pushed"
- release the button
- You should be in dfu

2nd solution : 
- Use usb cable on the de debug port (power port)
- use a terminal on that and you should see u-boot menu

sq6emm, 2025-02-11:
This is exactly what I was doing before opening the Issue...;)
But... it seems that computer reboot solved the issue - dfu-uploaded.

Now testing if I can upload boot.frm again... and dont get bricked.

sq6emm, 2025-02-11:
I tought the issue is with boot.frm.
But it seems its not... I have uploaded pluto.frm over mass-storage and again.. the only way is through dfu-util....

Something wrong with the build?

sq6emm, 2025-02-11:
Steps I am performing now to check if its not related to old builds for libresdr ... (clean git repo):

```
cd git/original/
mv tezuka_fw tezuka_fw-broken
git clone https://github.com/F5OEO/tezuka_fw.git
cd tezuka_fw
cp ../tezuka_fw-broken/buildroot-2024.08.tar.gz .
tar zxf buildroot-2024.08.tar.gz
source sourceme.first
cd buildroot-2024.08/
make pluto_maiasdr_defconfig && make
```

ended up with:

```
****** Xilinx Bootgen v2022.2
  **** Build date : Apr  1 2024-08:38:43
    ** Copyright 1986-2022 Xilinx, Inc. All Rights Reserved.

[WARNING]: Partition fsbl.elf.0 range is overlapped with partition system_top.bit.0 memory range

[INFO]   : Bootimage generated successfully

Image Name:   
Created:      Tue Feb 11 13:23:42 2025
Image Type:   ARM Linux RAMDisk Image (gzip compressed)
Data Size:    19679241 Bytes = 19218.01 KiB = 18.77 MiB
Load Address: 00000000
Entry Point:  00000000
	zip warning: name not matched: boot.dfu
  adding: boot.frm (deflated 98%)
  adding: pluto.frm (deflated 4%)
  adding: pluto.dfu (deflated 4%)
  adding: sdimg/BOOT.bin (deflated 62%)
  adding: sdimg/devicetree.dtb (deflated 73%)
  adding: sdimg/uboot-env.txt (deflated 77%)
  adding: sdimg/uImage (deflated 4%)
  adding: sdimg/uramdisk.image.gz (deflated 1%)
```

now copying to Adalm Pluto... flashing...

and this time all works;)
Seems the problem was with my BUILD environment?

F5OEO, 2025-02-11:
Can this issue be closed ?

closed 2025-02-11 by sq6emm