F5OEO/tezuka_fw #55: LIBRESDR SDRPP not found issue

Notupus, 2025-02-11:
Using the sdrpp that is linked in the readme I cant get it to recognize my libresdr over ethernet on windows 10.
How have people used sdrpp with tezuka? SDR Console works with manual ip but has b/w issues.
The libresdr is connected in a 10.0.0.0/24 lan network

Notupus, 2025-02-11:
Okay, so how do I do that, assuming I am on windows and have a nic, my pc never gets  dhcp ip address or anything like that so what do I need to configure?

F5OEO, 2025-02-11:
No...this is not the source of issue. Do not follow "https://github.com/AlexandreRouma/SDRPlusPlus/issues/1399" as it is outdated and not related to fork.
It is more related to a dll distribution on windows. Will fix it soon.

Notupus, 2025-02-11:
Okay, so using what os /configuration can I get the 45msps currently? Network identification and or connection hasnt worked yet on SDR++ with any method I tried. What is a currently working way to do this.

povnz, 2025-03-09:
I’m back to using 1.0.4 which works great can’t all all late builds including the fork just crash while opening even after patching the iio libs :(

povnz, 2025-03-23:
Ok if you patch IIO as follows:

Download https://github.com/analogdevicesinc/libiio/releases/download/v0.26/Windows.zip

From: Windows.zip\Windows-VS-2022-x64
Copy the following to the lastest build:

libserialport-0.dll
libiio-sharp.dll
libiio.dll (this will overight)
libxml2.dll

And Disable IPv6 on the entire system it works again

Alphafox02 has discovered a bug where if both ipv4 and ipv6 are regiested againast a single pluto serial then it crashes SDRPP:
https://github.com/AlexandreRouma/SDRPlusPlus/issues/1594