F5OEO/tezuka_fw #56: SDRAM O/C with adjusted CL timings

ModderMax, 2025-02-11:
I'm running a Zynq7020 Libresdr, and have issues with running either the FSBL_44_25 or FSBL_44_30 overclocked boot files, I can run the lower clocks fine. I have some logs from the USB Debug port that suggest that there's a memory issue when initializing the clock reference (I will attach these as .txt files. On the hz12opensource's version of the FW I noticed that the timings are modified in system_bd.tcl

`You can run the memory at a much higher clock by increasing the timing parameters. They are defined in hdl/projects/libre/system_bd.tcl (looking for PCW_UIPARAM_DDR_CL and a few others). With a 9-7-9-9 timing I can overclock mine to 750 MHz.`

Within tezuka_fw/board/pluto/bitstream/maia-iio/pluto_iio/system_bd.tcl there doesn't seem to be this entry along with the other DDR configurations, and I also wonder if changing these would apply the DDR settings to the libresdr version of the firmware. 

```
# DDR MT41K256M16 HA-125 (32M, 16bit, 8banks)

ad_ip_parameter sys_ps7 CONFIG.PCW_UIPARAM_DDR_PARTNO {MT41K256M16 RE-125}
ad_ip_parameter sys_ps7 CONFIG.PCW_UIPARAM_DDR_BUS_WIDTH {16 Bit}
ad_ip_parameter sys_ps7 CONFIG.PCW_UIPARAM_DDR_USE_INTERNAL_VREF 0
ad_ip_parameter sys_ps7 CONFIG.PCW_UIPARAM_DDR_TRAIN_WRITE_LEVEL 1
ad_ip_parameter sys_ps7 CONFIG.PCW_UIPARAM_DDR_TRAIN_READ_GATE 1
ad_ip_parameter sys_ps7 CONFIG.PCW_UIPARAM_DDR_TRAIN_DATA_EYE 1
ad_ip_parameter sys_ps7 CONFIG.PCW_UIPARAM_DDR_DQS_TO_CLK_DELAY_0 0.048
ad_ip_parameter sys_ps7 CONFIG.PCW_UIPARAM_DDR_DQS_TO_CLK_DELAY_1 0.050
ad_ip_parameter sys_ps7 CONFIG.PCW_UIPARAM_DDR_BOARD_DELAY0 0.241
ad_ip_parameter sys_ps7 CONFIG.PCW_UIPARAM_DDR_BOARD_DELAY1 0.240
```

I also had some questions about the overclock [scripts?] in tezuka_fw/board/libre/bitstream/overclock. When building I noticed that they used the xilinix bootgen function, I just couldn't tell from the script that seems to iterate through them if the parameters for the clocks are just derived from the file names or do the files contain the parameters for the clocks? If the latter, how could these parameters be modified?

It should be stated that I am mostly interested in getting the most out of this board, and may end up doing many modifications/brick it in the end. I'd love to be able to get the resources out there for high performance SDR for a lower cost. I am very new to the way the firmware is built, though I have some experience with linux and a good bit of experience with Java, C++, and Powershell. Thank you for all your contributions, if there is anything further I can provide don't hesitate to ask-

[Success.txt](https://github.com/user-attachments/files/18754928/Success.txt)

[Failed.txt](https://github.com/user-attachments/files/18755047/Failed.txt)

F5OEO, 2025-02-11:
Overclock is done by this script : https://github.com/F5OEO/maia-sdr/blob/sweep/maia-hdl/projects/libre_iio/overclock.sh
Timings are set here ; https://github.com/F5OEO/maia-sdr/blob/92f78fc05f027466baacf1658e6f22a9d6c80f81/maia-hdl/projects/pluto/system_bd.tcl#L188

ModderMax, 2025-02-11:
Thank you very much, this is exactly what I was looking for! If you would like, leave the thread open and I will post any results I get, otherwise that's all I needed thank you again!

Notupus, 2025-02-12:
I think it might not be a DRAM issue but a CPU one, you are having issues with the higher 44 multiplier. Would be interesting to see more of those overclock options with finer steps, since its not packaged in the libresdr.zip maybe in a libresdr_overclock.zip. With my sdr I can get 40_28 with 44mhz to 43.5 on sdrpp on linux. I am not willing to install vivaldi to make those myself, I gave up after realizing that my ubuntu is 3 versions ahead of any guide and the docker crashed on me..
If you can share more elf files with various timings and cpu multipliers and some testing I would be really interested in that

ModderMax, 2025-02-14:
> I think it might not be a DRAM issue but a CPU one, you are having issues with the higher 44 multiplier. Would be interesting to see more of those overclock options with finer steps, since its not [...]

Yeah, I popped them into a hex editor and made a few adjustments. I have yet to test them, but you can find them here- https://github.com/ModderMax/O-C-Scripts-for-Tezuka_FW

Notupus, 2025-02-15:
Mine is stable with the 42 28 and probably a possible 42 30 too

ModderMax, 2025-02-15:
> Mine is stable with the 42 28 and probably a possible 42 30 too

I added a few more clock scripts, including a fsbl42_30.elf 
A few notes: 41_30 was not stable on my board. Though it would boot without crashing, any higher sample rate use would cause a crash. Surprisingly, I was able to get a decent boost in stability on higher sample rates while using 40_31 compared to 40_30. You may have similar results, though much higher of a memory clock will likely require timing changes if you want to go higher in that respect. 

If you would happen to have a picture of your board's layout I would be interested in comparing any differences there may be.

Notupus, 2025-02-16:
[image]

[image]
I will post overclock results soon.

Notupus, 2025-02-16:
42 30 was my max, we likely need timing change for this.

DG9BFC, 2025-03-30:
what about adding more cooling?? (just thinking)