F5OEO/tezuka_fw #58: Bad or wrong waveform on SSB

sv5byr, 2025-02-11:
Hello, Tezuka firmware 0.95 have problem to SSB TX. The output form is DSB not Single Side Band.
Tested with Libre SDR and SDR Console @ 2.4 GHz

F5OEO, 2025-02-11:
Please, attached screenshot of spectrum.

sv5byr, 2025-02-11:
[image: εικόνα.png]

[image: εικόνα.png]

F5OEO, 2025-02-11:
Try to join pictures on github web directly instead of mail because I can't see pictures attached.

sv5byr, 2025-02-11:
[image]

[image]

sv5byr, 2025-02-13:
> All is working well on libresdr with SdrConsole here. Others have used it
> on QO100 : no issue with DSB like signal. Can you confirm your issue ?

Hello, are they use 0.90 firmware ?? Is all fixed on 1.0 to try it ???

sv5byr, 2025-02-13:
[image]

sv5byr, 2025-02-13:
Fw  1.0.0  uo QSPI and running...!!! SSB is clean as diamond

sv5byr, 2025-02-13:
[image]

Im very happy about the new fw.

(change year to 2025)

F5OEO, 2025-02-13:
Seems fixed. Close for now.

closed 2025-02-13 by F5OEO