F5OEO/tezuka_fw #63: Question: Bandwidth for Release 0.1.0

SimonG4ELI, 2025-02-13:
Hi,

Adding extra support for LibreSDR in SDR Console. Using release 0.1.0, what is the bandwidth range on transmit and receive?

simon@sdr-radio.com

Notupus, 2025-02-13:
In theory with enough bw you can get pretty much full 62msps if you are able to get enough bw. As for using the gigabit port its about 25ish msps for me until the ethernet gets bottlenecked. I suppose it can support full duplex at this speed but sdrpp does not have tx function for me to test. This is due to the difference with the pluto that it uses ethernet as an interface so the interface is full duplex compared to usb 2's overall 480mbps limit. This is with mild overclocking which is safe for the hw. I believe that it would be a better idea to handle buffering well (see sdrpp repo from F5OEO) since even with ethernet sdrconsole currently on my machine cant handle 6msps over ethernet due to buffering issues. If you need any testing I'd be more than happy to help. Keep in mind the top sample rate is done by 12hzopensource at 27.5msps at 12(16) bit.

F5OEO, 2025-02-13:
Using my perf tool (https://github.com/F5OEO/tezuka_fw/blob/main/app/pluto_rx/plutorx.c)
Normal libresdr , RXONLY without underflow :
 
CS16 : Sample Max=20750000
CS8 : Sample Max=41250000

Overclock libresdr (fsbl44-30) : 
CS16 : Sample Max=28250000
CS8 : Sample Max=49500000

I could also add CS12 packed format if needed to save bandwidth without loosing dynamic.

@Notupus _Keep in mind the top sample rate is done by 12hzopensource at 27.5msps at 12(16) bit._
is not true.

Notupus, 2025-02-13:
Sorry for my false judgement, didnt know we had a perf tool for that.

SimonG4ELI, 2025-02-13:
> Using my perf tool (https://github.com/F5OEO/tezuka_fw/blob/main/app/pluto_rx/plutorx.c) Normal libresdr , RXONLY without underflow :
>
> CS16 : Sample Max=20750000 CS8 : Sample Max=41250000
>
> Overclock [...]

Hi,

Excellent, thanks. I'm not a fan of overclocking; I like my electronics to stay cool.

SimonG4ELI, 2025-02-13:
> 
> I could also add CS12 packed format if needed to save bandwidth without loosing dynamic.
> 

Please add this - would increase the USB bandwidth I think. I assume the USB OTG is USB 2?

I must read the hardware spec.

F5OEO, 2025-02-13:
Note that these results are over GbEthernet of course ! 
If transfer over usb for regular pluto (yes usb2), do not use ip over usb but directly usb context to maximize bandwidth.
I will add a CS12 soon.

Just quick remark for performance : (see example in the code)
 - Use number of kernel buffer to avoid underrun https://github.com/F5OEO/tezuka_fw/blob/272be837eb83e21c44d5079d753a57c7e90c2b85/app/pluto_rx/plutorx.c#L292
 - Use register to know if there is any underflow
 https://github.com/F5OEO/tezuka_fw/blob/272be837eb83e21c44d5079d753a57c7e90c2b85/app/pluto_rx/plutorx.c#L318

zx-ecco, 2025-02-13:
can the fw capture directly to the microsd card? if so, what's the max bandwith for that?

F5OEO, 2025-02-13:
Using maia-sdr (included on firmware), you can record on memory and transfer to sd afterall. It is the max bandwidth of ad93xx : 61.44MS. Note it is only chunk of full bandwidth and could not be streamed in real-time over the network.

zx-ecco, 2025-02-13:
does that mean it can only record in short chunks that will fit on memory and are then transferred to the SD or can it continuously capture to SD using memory as a buffer?

F5OEO, 2025-02-13:
Yes : "it can only record in short chunks that will fit on memory".
A script at the end of recording could be added to copy file in memory to sd card.
"continuously capture to SD using memory as a buffer" : NO.

zx-ecco, 2025-02-13:
is this a hardware limitation or just that maiasdr is currently missing the feature? can two libresdrs be  daisy chained or paired with an external clock reference to to expand the bandwith like the b210s?

DG9BFC, 2025-02-27:
what about connecting an ssd via usb otg cable and maybe a usb hub if you wanna connect other usb things like a minitiuoner??

closed 2025-03-30 by F5OEO