F5OEO/tezuka_fw #65: LibreSDR - LED support

sq6emm, 2025-02-14:
It seems that we dont have any LED communication of LibreSDR with outside world.

I was able to identify 4 LEDs on schematic and the real PCB:

[https://github.com/day0wl/libresdr-fw/blob/main/zynqsdr_rev5.pdf](url)

D1 - R11 / DONE_0 (I guess this is our activity/flashing LED)
D2 - PS_POR# - PG_ALL / C7 - PS_POR_B_500 (seems like power LED on 3.3V)
D10 - PL_LED0 - J20 / IO_L17P_T2_AD5P_35 (I guess this is 10mhz or pps ref)
D13 - PL_LED1 - H20 / IO_L17N_T2_AD5N_35 (I guess this is 10mhz or pps ref)
ACTLED0 - ethernet LED
ACTLED1 - ethernet LED

The only reference to LEDs in the code I was able to find in:
board/libre/dts/zynq-libre.dts

```
                led0 {
                        label = "led0:green";
                        gpios = <&gpio0 15 0>;
                        linux,default-trigger = "heartbeat";
                };
```

Unfortunatelly I am unable to map that anyhow to schematics... I guess its done in fpga?

Maybe it would be worth to map D1 diode to have that activity?

F5OEO, 2025-02-18:
Until I misunderstand : 
DONE_0(R11) is wire to D1 : it should be 1 as soon as the fpga is running. It could not be programmed by PS, so it could not be an activity led.
D2 is power led : fixed
So only D10 and D13 could be use to indicate some status.

sq6emm, 2025-02-19:
Seems you are right:)
Only the D10/D13 so 10MHz/PPS diodes can be used to indicate status. But then most probably they should be only used for their designed role... So led0 should be removed from the code?

sq6emm, 2025-02-19:
LibreSDR from OpenSourceSDR on their default firmware after booting from flash is blinking with PPS LED (D13)

DG9BFC, 2025-03-23:
lesha108 fw also does blinky blinky ... and has ptt output .. rx tx frequeny (can be decoded from an arduino?) .. and has a working dfu botton
maybe we can add that also?

crvux, 2025-04-25:
I found that in dts for libre there should be gpio line 68 not 15
https://github.com/F5OEO/tezuka_fw/blob/main/board/libre/dts/zynq-libre.dts#L31

I checked this with this command in the default libre firmware
```bash
cat /sys/class/leds/led0:green/device/of_node/led0/gpios | hexdump
```
0000 0900 0000 4400 0000 0000 => 0x44 => 68

But when I patch dts led still doesn't work. Even if I control via sysfs. Maybe I need to do something with pin IO_L17P_T2_AD5P_35 in hdl project?

crvux, 2025-04-25:
I can submit a PR with new dts for PL_LED0, but I don't have the FPGA knowledge to edit hdl. 
It is not yet clear how to map a pin to a line number for kernel driver in general. So, now it is not clear which line should be 
PL_LED1.

crvux, 2025-04-25:
lesha108 uses lines 106 and 107
https://github.com/search?q=repo%3Alesha108%2Fzynqsdr+LED1&type=code

F5OEO, 2025-04-25:
Interesting, thanks for all your inspection. I think to have enough info to adapt tezuka. Stay tuned.

crvux, 2025-04-25:
It may also be useful to use standard naming https://docs.kernel.org/leds/leds-class.html
`led0:green` -> `blue:led0`