F5OEO/tezuka_fw #66: Sd image not boot after 0.0.9

johnsonZ72, 2025-02-15:
After 0.0.9 libresdr not boot (method copy image to sdcard).             
Check 0.0.95 and 0.1.0 from release

In console:

LIBRESDR> reset
resetting ...

U-Boot 2016.07 (Jan 31 2025 - 23:02:45 +0000)

I2C:   ready
DRAM:  ECC disabled 1 GiB
MMC:   sdhci@e0100000: 0
SF: Detected W25Q256 with page size 256 Bytes, erase size 4 KiB, total 32 MiB
In:    serial@e0000000
Out:   serial@e0000000
Err:   serial@e0000000
Model: LibreSDR Rev.5 (Z7020-AD9363)
Hit any key to stop autoboot:  0
Unknown command 'bootp' - try 'help'
Wrong Image Format for bootm command
ERROR: can't get kernel image!

F5OEO, 2025-02-17:
No issue here : with 0.1.0 you should see

```
U-Boot 2016.07 (Feb 11 2025 - 20:09:24 +0000)

I2C:   ready
DRAM:  ECC disabled 1 GiB
MMC:   sdhci@e0100000: 0
SF: Detected W25Q256 with page size 256 Bytes, erase size 4 KiB, total 32 MiB
In:    serial@e0000000
Out:   serial@e0000000
Err:   serial@e0000000
Model: LibreSDR Rev.5 (Z7020-AD9363)
reading uEnv.txt
6992 bytes read in 17 ms (401.4 KiB/s)
Importing environment from SD ...
Hit any key to stop autoboot:  0 
LIBRESDR> 
```
Are you sure to have extract what is under sdimg to root of your sdcard ?

johnsonZ72, 2025-02-17:
yes , shure  (only 0.0.9.5 and 0.0.1)
also when building from source the same picture

F5OEO, 2025-02-17:
I am surprised because your u-boot version is different , like if BOOT.bin has not been updated.

johnsonZ72, 2025-02-19:
Maybe this help to resolv 
LIBRESDR> printenv
baudrate=115200
bootargs=root=/dev/nfs nfsroot=: ip=::::::off
bootcmd=bootp; setenv bootargs root=/dev/nfs nfsroot=${serverip}:${rootpath} ip=${ipaddr}:${serverip}:${gatewayip}:${netmask}:${hostname}::off; bootm
bootdelay=5
compatible=ad9364
fdtcontroladdr=3ff98f90
modeboot=sdboot
qspiboot=set stdout nulldev;adi_hwref;test -n $PlutoRevA || gpio input 14 && set stdout serial@e0001000 && sf probe && sf protect lock 0 100000 && run dfu_sf;  set stdout serial@e0001000;itest *f8000258 == 480003 && run clear_reset_cause && run dfu_sf; itest *f8000258 == 480007 && run clear_reset_cause && run ramboot_verbose; itest *f8000258 == 480006 && run clear_reset_cause && run qspiboot_verbose; itest *f8000258 == 480002 && run clear_reset_cause && exit; echo Booting silently && set stdout nulldev; run read_sf && run adi_loadvals; envversion;setenv bootargs console=ttyPS0,115200 maxcpus=${maxcpus} rootfstype=ramfs root=/dev/ram0 rw quiet loglevel=4 clk_ignore_unused uio_pdrv_genirq.of_id=uio_pdrv_genirq uboot="${uboot-version}" && bootm ${fit_load_address}#${fit_config} || set stdout serial@e0001000;echo BOOT failed entering DFU mode ... && sf protect lock 0 100000 && run dfu_sf
qspiboot_verbose=adi_hwref;echo Copying Linux from QSPI flash to RAM... && run read_sf && if run adi_loadvals; then echo Loaded AD936x refclk frequency and model into devicetree; fi; envversion;setenv bootargs console=ttyPS0,115200 maxcpus=${maxcpus} rootfstype=ramfs root=/dev/ram0 rw earlyprintk clk_ignore_unused uio_pdrv_genirq.of_id=uio_pdrv_genirq uboot="${uboot-version}" && bootm ${fit_load_address}#${fit_config} || echo BOOT failed entering DFU mode ... && run dfu_sf
ramboot_verbose=adi_hwref;echo Copying Linux from DFU to RAM... && run dfu_ram;if run adi_loadvals; then echo Loaded AD936x refclk frequency and model into devicetree; fi; envversion;setenv bootargs console=ttyPS0,115200 maxcpus=${maxcpus} rootfstype=ramfs root=/dev/ram0 rw earlyprintk clk_ignore_unused uio_pdrv_genirq.of_id=uio_pdrv_genirq uboot="${uboot-version}" && bootm ${fit_load_address}#${fit_config}
stderr=serial@e0000000
stdin=serial@e0000000
stdout=serial@e0000000

Environment size: 2163/131068 bytes

F5OEO, 2025-02-19:
I can't understand from where you get this env. Maybe on QSPI ? Have you tried other firmware ? Are you sure your sd card is OK ? 
You can contact me directly (evaristec@gmail.com) in order to have a teamviewer session for debugging.

F5OEO, 2025-02-19:
Fixed : env seems from coming to an other firmware which has missing variables, specially no variable to boot on sd
Solution : (on u-boot console)
env default -f -a
run uenvboot
env save
reset

closed 2025-02-19 by johnsonZ72