F5OEO/tezuka_fw #69: Building 0.1.0 failed

Szczecin-Radio, 2025-02-16:
Hello,

Just informational. I have tried to compile the latest firmware. I started with  'plutoplus_original_defconfig' and default procedure:

git clone https://github.com/F5OEO/tezuka_fw
cd tezuka_fw
wget https://buildroot.org/downloads/buildroot-2024.11.1.tar.gz

tar -xvf buildroot-2024.11.1.tar.gz

source sourceme.first
cd buildroot-2024.11.1
make plutoplus_original_defconfig && make

Finally, compilation has failed at the end with an error:

GIT_DIR=. PATH="/opt/tmp/tezuka_fw/buildroot-2024.08/output/host/bin:/opt/tmp/tezuka_fw/buildroot-2024.08/output/host/sbin:/home/wojtek/.local/bin:/home/wojtek/bin:/bin:/sbin:/usr/sbin:/usr/bin:/usr/local/bin:/usr/local/sbin:/opt/presto/bin" PKG_CONFIG="/opt/tmp/tezuka_fw/buildroot-2024.08/output/host/bin/pkg-config" PKG_CONFIG_SYSROOT_DIR="/" PKG_CONFIG_ALLOW_SYSTEM_CFLAGS=1 PKG_CONFIG_ALLOW_SYSTEM_LIBS=1 PKG_CONFIG_LIBDIR="/opt/tmp/tezuka_fw/buildroot-2024.08/output/host/lib/pkgconfig:/opt/tmp/tezuka_fw/buildroot-2024.08/output/host/share/pkgconfig" BR_BINARIES_DIR=/opt/tmp/tezuka_fw/buildroot-2024.08/output/images KCFLAGS=-Wno-attribute-alias DTC_FLAGS=-@ /bin/make -j9 HOSTCC="/bin/gcc -O2 -isystem /opt/tmp/tezuka_fw/buildroot-2024.08/output/host/include -L/opt/tmp/tezuka_fw/buildroot-2024.08/output/host/lib -Wl,-rpath,/opt/tmp/tezuka_fw/buildroot-2024.08/output/host/lib" ARCH=arm INSTALL_MOD_PATH=/opt/tmp/tezuka_fw/buildroot-2024.08/output/target CROSS_COMPILE="/opt/tmp/tezuka_fw/buildroot-2024.08/output/host/bin/arm-none-linux-gnueabihf-" WERROR=0 REGENERATE_PARSERS=1 DEPMOD=/opt/tmp/tezuka_fw/buildroot-2024.08/output/host/sbin/depmod INSTALL_MOD_STRIP=1 -C /opt/tmp/tezuka_fw/buildroot-2024.08/output/build/linux-04fbdbb433faae8ffb3a5b24b7435509b11334cb zynq-pluto-sdr.dtb zynq-pluto-sdr-revb.dtb zynq-pluto-sdr-revc.dtb zynq-plutoplus.dtb

Error: arch/arm/boot/dts/zynq-plutoplus.dtsi:12.1-9 syntax error
FATAL ERROR: Unable to parse input tree
make[2]: *** [scripts/Makefile.lib:406: arch/arm/boot/dts/zynq-plutoplus.dtb] Error 1
make[1]: *** [Makefile:1448: zynq-plutoplus.dtb] Error 2
make: *** [package/pkg-generic.mk:289: /opt/tmp/tezuka_fw/buildroot-2024.08/output/build/linux-04fbdbb433faae8ffb3a5b24b7435509b11334cb/.stamp_built] Error 2

I am going to try another configs, but it does not look good.

Regards,
/WS

F5OEO, 2025-02-17:
As maias-sdr is transparent, only config with maia is maintained right now. 
Will remove plutoplus_original, pluto_original until there is a good reason to not having maia-sdr inside.

F5OEO, 2025-02-17:
Closed until request from standard build without maia

closed 2025-02-17 by F5OEO