F5OEO/tezuka_fw #71: USB 2  libre (otg port) connecting  y kabel : apply supply and arduino module.

pa3gsb, 2025-02-17:
Using y-kabel for otg.

Now the arduino is not found; did a short investigation:

# cat /sys/kernel/debug/usb/ci_hdrc.0/role
gadget

set to host modus:
echo host | tee /sys/kernel/debug/usb/ci_hdrc.0/role

# cat /sys/kernel/debug/usb/ci_hdrc.0/role
host

Now the arduino is recognized.

TODO
This must be corrected ; if an device is connected to the second port of the y-cable the usb must be set into host modus.

# cat /sys/kernel/debug/usb/ci_hdrc.0/device
speed             = 0
max_speed         = 3
is_otg            = 1			<============

F5OEO, 2025-02-17:
Y-cable is generally used for OTG, thus I think it is normal that system is set as OTG.
It could be possible to force it to host, but need to specify it on environment variable because else could break other app.
I recommand to use a USB powered hub instead if you want power and USB host ?

pa3gsb, 2025-02-17:
i think you are misunderstand the issue. The moment iam attaching an otg iam expecting that the libre is switched into usb host mode.  That is now not happening i need to write a command to the libre to get it in host modes.

i do not understand your remark about usb powered hub; the usb power does come from the device which is in host mode the other connections are slave.  What do i miss?

F5OEO, 2025-02-18:
I maybe misunderstand the issue. I try to reproduce your setup : I use a raspberry pico instead of arduino.
Pico is programmed to be a usb-serial device which coull be plugged to a host (PC or Pluto)
I use a usb cable usb-C to usb-female 
When I plug it , pluto recognized it and pluto acting as a host : 

[image]

```
sb 1-1: new full-speed USB device number 3 using ci_hdrc
usb 1-1: New USB device found, idVendor=2e8a, idProduct=000a, bcdDevice= 1.10
usb 1-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 1-1: Product: Pico
usb 1-1: Manufacturer: Raspberry Pi
usb 1-1: SerialNumber: E6604430433E672A
cdc_acm 1-1:1.0: ttyACM1: USB ACM device
# cat /sys/kernel/debug/usb/ci_hdrc.0/role
host
# 
```

I could also use a usb-hub like this : 

[image]

```
sb 1-1: Manufacturer: GenesysLogic
hub 1-1:1.0: USB hub found
hub 1-1:1.0: 4 ports detected
usb 1-1.1: new high-speed USB device number 3 using ci_hdrc
usb 1-1.1: New USB device found, idVendor=0b95, idProduct=1790, bcdDevice= 2.00
usb 1-1.1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 1-1.1: Product: AX88179A
usb 1-1.1: Manufacturer: ASIX
usb 1-1.1: SerialNumber: 00EDBB1D
usb 1-1.2: new full-speed USB device number 4 using ci_hdrc
usb 1-1.2: New USB device found, idVendor=2e8a, idProduct=000a, bcdDevice= 1.10
usb 1-1.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
usb 1-1.2: Product: Pico
usb 1-1.2: Manufacturer: Raspberry Pi
usb 1-1.2: SerialNumber: E6604430433E672A
cdc_acm 1-1.2:1.0: ttyACM1: USB ACM device
# cat /sys/kernel/debug/usb/ci_hdrc.0/role
host
```

Y-cable is generally to force an equipment to be a slave (gadget). Thus pluto is acting as a gadget of your arduino. Generally, it is the cable you use for plug a pluto to a phone. I don't think it is the behaviour you want.

In conclusion, try with a "normal" usb cable and not a Y one. Hope that I understand well your issue. At least I tried to clarify it?

pa3gsb, 2025-02-18:
Hi tnx for your detailed reply; really appreciated.

To which port are you applying the power?  

I have in mind to apply the power to the otg port of the libre. 

tnx for your time and showing the output.  i will try the same!

F5OEO, 2025-02-18:
You are welcomed. I power it from debug/power port.

F5OEO, 2025-02-19:
Can you confirm that with a regular cable , it works ?

pa3gsb, 2025-02-20:
The main problem i was facing that i wanted to apply the supply to the OTG port. I thought the debug port was for debugging only.
Now applying 5V to debug or power  port and applying the usb hub or directly my arduino to the otg port and using the ethernet connection for a remote connection it is working perfectly.

Thanks again.

closed 2025-02-20 by pa3gsb