F5OEO/tezuka_fw #73: Device not found

gariac, 2025-02-23:
I have the Zynqsdr working (to some degree) using this github:
https://github.com/hz12opensource/libresdr

This is to prove that at least the device works though I really want the features of Tezuka. 

Here is the dmesg of the Zynqsdr being found on Debian 12.9:

`
[ 3355.100556] usb 5-1: new high-speed USB device number 4 using xhci_hcd
[ 3355.239919] usb 5-1: New USB device found, idVendor=0456, idProduct=b673, bcdDevice= 5.15
[ 3355.239933] usb 5-1: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[ 3355.239938] usb 5-1: Product: PlutoSDR (ADALM-PLUTO)
[ 3355.239943] usb 5-1: Manufacturer: Analog Devices Inc.
[ 3355.333426] rndis_host 5-1:1.0 eth0: register 'rndis_host' at usb-0000:c3:00.3-1, RNDIS device, 00:e0:22:ad:c8:3b
[ 3355.335141] usb-storage 5-1:1.2: USB Mass Storage device detected
[ 3355.335329] scsi host0: usb-storage 5-1:1.2
[ 3355.336519] cdc_acm 5-1:1.3: ttyACM0: USB ACM device
[ 3355.389721] rndis_host 5-1:1.0 enx00e022adc83b: renamed from eth0
[ 3356.357506] scsi 0:0:0:0: Direct-Access     Linux    File-Stor Gadget 0515 PQ: 0 ANSI: 2
[ 3356.358012] sd 0:0:0:0: Attached scsi generic sg0 type 0
[ 3356.359795] sd 0:0:0:0: [sda] 61441 512-byte logical blocks: (31.5 MB/30.0 MiB)
[ 3356.360011] sd 0:0:0:0: [sda] Write Protect is off
[ 3356.360014] sd 0:0:0:0: [sda] Mode Sense: 0f 00 00 00
[ 3356.360269] sd 0:0:0:0: [sda] Write cache: enabled, read cache: enabled, doesn't support DPO or FUA
[ 3356.405406]  sda: sda1
[ 3356.405763] sd 0:0:0:0: [sda] Attached SCSI removable disk

`
Using the 0.1 Tezuka, the device isn't found.

This is the Tezuka image on the SD card:
https://imgur.com/a/RL1hmw1

This is the Libresdr image on the SD card:
https://imgur.com/a/arO6SIO

Now note the Tezuka SD card is being read. This is evident because the blue LED is on near the ethernet port.

Suggestions?

As an aside, I am following your issue on the sdrpp the github.  I cannot get the ethernet working with Libresdr, only the USB.

Jpc-Automation, 2025-02-23:
Hi Gariac,

I have tested the tezuka_fw 0.1.0 for the last week and it is working properly.

- Copy the tezuka_fw SdCard files to the SDCard.
- Insert card into LibreSdr and connect USBC port near to the ethernet jack. that is the USB OTG Port.
- A disk will appear in windows file manager called pluto.
- Open this disk and copy the tezuke_fw file calles "boot.frm" onto this disk.
- Eject the disk. Do not disconnect the usb.

This solved my problems.

Another note:
When testing don't connect ethernet and USB OTG at the same time to the same PC. PlutoSDR driver cannot initialize and crashes.

//Jeff

gariac, 2025-02-24:
The zynqsdr is simply not recognized by linux or Windows with the tezuka sd card. So I cant get  to the second step.

I loaded sdrpp on windows just to test it out using the libresdr sd card. It works on sdrpp using USB but not ethernet. There is a windows program called SDRConsole. That worked poorly on usb. It is supposed to work on ethernet but I couldn't get it going. 

I did a compile of tezuka and have access to boot.frm. It is not in the image directory but in the level above. Adding that file to tezuka led to a state where the blue led by the ethernet port didn't light.

Jpc-Automation, 2025-02-24:
> The zynqsdr is simply not recognized by linux or Windows with the tezuka sd card. So I cant get to the second step.
>
> I loaded sdrpp on windows just to test it out using the libresdr sd card. It [...]

You need to put the libiio drivers(dll files) from analog devices in the sdr++ directory to work with pluto usb and ethernet.

Drivers: https://github.com/analogdevicesinc/libiio/releases/tag/v0.25

Similar subject: https://www.rtl-sdr.com/plutosdr-quickstart-guide/

gariac, 2025-02-24:
No change. The fact that the libresdr SD card works and the tezuka doesn't made this kind of obvious but i figure I would exhaust this possibility.

I am not a windows user.  I really need this to work under linux. Incidentally I was surprised at how poorly the libresdr sd image worked on sdrpp  on windows. What I have done to flog the system is to pick a local FM broadcast station as a benchmark.  As you up the bandwidth, you are essentially upping the sample rate. Under linux, you can go to 8MHz. Under windows, the limit is 6MHz. This is based on my ear detecting the missing packets.

76nic, 2025-03-11:
Dear all,
was someone able to get fw_0.1.0 running under Linux? See issue #76 I opened some days ago.

F5OEO, 2025-03-13:
Maybe try this solution : https://github.com/F5OEO/tezuka_fw/issues/66#issuecomment-2668813170

closed 2025-03-25 by F5OEO