F5OEO/tezuka_fw #76: LibreSDR - libresdr 0.1.0 not working on Ubuntu

76nic, 2025-03-08:
Dear F5OEO and team,
first at all I really like your efforts to maintain some firmware for the LibreSDR - very much needed!!!
I wanted to try your version 0.1.0 on my LibreSDR on Ubuntu but it is not working. Not even the "USB drive" is popping up. I copied all the files as described to the card (I have not burned it). I have another firmware running just fine.
Any suggestions/help?
Many thanks!
Best regards,
DC0NG

76nic, 2025-03-11:
additional note: I just came across issue #73 (https://github.com/F5OEO/tezuka_fw/issues/73); it looks that it is the same problem. The "USB drive" is not recognized/popping up in Linux (Ubuntu).

F5OEO, 2025-03-11:
Can you drop me a mail : evaristec at gmail

F5OEO, 2025-03-13:
Solved with same solution as https://github.com/F5OEO/tezuka_fw/issues/66#issuecomment-2668813170

closed 2025-03-13 by F5OEO