F5OEO/tezuka_fw #77: fw_setenv has no effect to AD9363 when booting from SDcard

PeterBeijer, 2025-03-09:
Hi,

Tezuka works for me on Pluto-plus when booting from SDCard. Login in with ssh I see the environment settings are copied from config.txt. However, the settings for the AD9361 do not have any effect. Checking with satsagen you can see the low band is still limited to 320MHz. 

Any suggestions?

btw, appreciate the work you're doing!

regards,
Peter, PA3AEF

F5OEO, 2025-03-09:
Extension frequencies are not using env variable anymore, and is extended by default. Can you send a screenshot ?

F5OEO, 2025-03-09:
Satsagen seems to not used min and max frequency set by the firmware. You can force the transceiver in satsagen : transceiver type to ad9361. 

[image]

PeterBeijer, 2025-03-09:
I used the auto setting in Satsagen. When I try AD9361 dropdown it fails. 

[image]
[image]
[image]

PeterBeijer, 2025-03-09:
I just checked with the standard firmware: Satsagen picks the AD9361 when set to auto. Using below 320MHz is no problem.

PeterBeijer, 2025-03-09:
Here is a good example: both Satsagen set to auto.

1) standard firmware
2) Tezuka

[image]

F5OEO, 2025-07-19:
should be fixed with 66487c4e7961f92fdea36d08e27136d349cd8237