F5OEO/tezuka_fw #83: Second Rx/Tx channel not working - gnuradio

76nic, 2025-03-18:
Dear F5OEO,
Thanks to your help my LibreSDR is running and I started testing it a bit. I was not able to get the second Rx/Tx channel working. It seams the config attributes differ from the Analog Devices Pluto firmware. I was setting the following attributes (not sure if "rf_input/output" is identical with "mode"):
  attr_name = compatible
  attr_val = ad9363a
  mode = 2r2t
and 
  rf_input = rx2
  rf_output = tx2

I gnuradio using the FMComms2/3/4 Block I'm only able to use channel 1. Both channels simultaneously and channel 2 only is not working (recognized).

Config settings:
  [TEZUKA]
  callsign = NOCALL
  locator = NOLOCATOR
  lnb_power = off
  refclk_source = internal
  maxcpus = 2
  attr_name = compatible
  attr_val = ad9363a
  mode = 2r2t
  audio_mode = off
  rf_input = rx2
  rf_output = tx2
  serial_force = off
  nfs_server = 
  force_model = off
  disable_usb_console = off

Many thanks!
Best regards,
DC0NG

F5OEO, 2025-03-19:
Yes, I need to fix u-boot in order to modify the dts according to fw_env in order to have both channels.

76nic, 2025-03-20:
Many thanks!

F5OEO, 2025-07-19:
Should be fixed with 66487c4e7961f92fdea36d08e27136d349cd8237

76nic, 2025-07-21:
@F5OEO 
Dear F5OEO, many thanks for dedicating your time fixing this bug. I'm really exited to use the dual RX/TX functionality. I'm currently on vacation without access to the SDR. I will thoroughly test it when I'm back (first/second week in August) and let you know.

Only to confirm the config settings - what is used?
mode = 2r2t
or
rf_input = rx2
rf_output = tx2

In general, do you maintain a list of all config commands that can be set?

Many, many thanks for your work!!!
Best regards,
DC0NG

willie-from-texas, 2025-07-23:
Hi. I"m new to the LibreSDR. I've been experimenting with SDRAngel and managed to get Channel 1 working. However, Channel 2 does not function. Not sure how to proceed but saw this thread and thought I'd ask if someone could provide insight in how to proceed. 
Thanks!
Will - KI5HUK

76nic, 2025-08-04:
@F5OEO: Dear F5OEO, I had a quick test regarding LibreSDR 2r2t functionality, unfortunately the problem still remains. I used 0.1.9 and Gnuradio to test the 2r2t functionality but the second Rx/Tx channel is still not working. I will further investigate the problem and do some more tests (also with SDRangel and Python) later this week.
Again many thanks for looking into this important bug.
Best regards,
DC0NG

76nic, 2025-08-04:
@F5OEO: Dear F5OEO, I would like to follow up on my previous comment regarding the LibreSDR 2r2t functionality. I have seen that you have rewritten the u-boot environment. That would mean that I have to update the u-boot environment on my device and can not simply use an SD card with the new firmware to get this bug fix implemented on my device - right? I came across the following Q/A, where they state that to update the u-boot environment I have to copy the unziped firmware ZIP to the device:
https://ez.analog.com/adieducation/university-program/f/q-a/544531/pluto-rev-c-how-to-activate-the-2nd-rx-channel

Would that be the same procedure for the tezuka firmware (coping the unziped tezuka firmware ZIP to the device in order to update the u-boot) ?

Many thanks for your help!
Best regards,
DC0NG

dsittner, 2025-08-05:
I can confirm that when running Tezuka from the SD card, the second Tx/Rx was unavailable within gnuradio, and the output from iio_info shows that the voltage1(input) is missing, as reported in other discussions.  (Interestingly, the F50E0 fork of SDR++ was still able to access Rx2)
HOWEVER, after flashing the most recent Tezuka to the internal SPI flash using the usual USB storage method, overwriting the stock firmware, the FMComms2/3/4 source block is able to access the second Rx (haven't tested Tx on the sink block yet) and the output from iio_info matches stock firmware again.  (Interestingly, the second channel is available regardless of mode = 1r1t or mode = 2r2t)

I had to ditch stock firmware because it wasn't properly handling the attr_val = ad9361 setting, and I was restricted to 325MHz - 3.8GHz.  With the most recent release of Tezuka, everything works now.  This is all on the LibreSDR/ZynqSDR from OpenSourceSDRLabs.  If someone can confirm on the other variants, I think we can consider this fixed.

wolfgang6444, 2025-08-06:
Hmm,

I reinstalled 0.1.9 from scratch (first copied content of sdimg onto an empty SD, then booted from the SD in DFU, the copied boot.frm and pluto.frm onto the storage device then unmounted the storage device, removed SD as well as SD-jumper and did a power cycle.

Here is what gnuradio throws at me  when I try to read from both channels using FmComms/2/3/4 source:

Generating: "/home/buesser/untitled.py"

> Executing: /usr/bin/python3 -u /home/buesser/untitled.py
> 
> Traceback (most recent call last):
>   File "/home/buesser/untitled.py", line 171, in <module>
>     main()
>   File "/home/buesser/untitled.py", line 149, in main
>     tb = top_block_cls()
>          ^^^^^^^^^^^^^^^
>   File "/home/buesser/untitled.py", line 103, in __init__
>     self.iio_fmcomms2_source_0 = iio.fmcomms2_source_fc32('192.168.2.58', [True\
> , True, True, True], 32768)
>                                  ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^\
> ^^^^^^^^^^^^^^^^^^^^^^^^^^^
> RuntimeError: Channel not found
> 
> >>> Done (return code 1)

I am using gnuradio   3.10.9.2 (Python 3.12.3) on a linux Ubuntu.

Now my board now declares itself as : Pluto+ (Z7010-AD9363A)

Do you have any hint of what might be wrong?

Is there a simple way (independent of gnuradio) to check if second channel is available? Maybe my gnuradio is broken/not really up to date.

Many thanks,

Wolfgang

BTW: single channel works fine

dsittner, 2025-08-07:
> Hello dsittner and DC0NG, many thanks for the reply and for trying. I will retry from scratch using 0.1.9. BTW what hardware have you tried it on? I am using plutolplus with a rev 2 board (the one with 3.3V on the UART). My board announces itself as
> 
> Analog Devices PlutoSDR Rev.C (Z7010-AD9363A)
> 
> no idea what an AD9363A is ??
> 
> and I am using the Gigabit interface
> 
> DK1BW

AD9363 is the Analog Devices transceiver module. It's supposed to only operate from 325Mhz to 3.8Ghz, but a firmware setting allows it to pretend to be a AD9361 and operate from 70Mhz to 6Ghz. I have the LibreSDR. I don't have a Pluto+ to play with.

rlabs-oss, 2025-08-31:
Which combination of firmware for the plutoplus and device settings enables coherent dual channel RX ? tried just about everything including removing most of the uboot if statements that remove the 2nd rx channel / device tree dodabs and still only get one rx channel that can be switched via ssh to rx1 or rx2 but never both...

76nic, 2025-08-31:
@rlabs-oss 
Generally you can change single/dual channel with:
mode=1r1t
mode=2r2t
You can do that by modifying the config file on your drive.
Are you using sd-card? On my device (LibreSDR/ZynqSDR) I had to once flash the BOOT.frm and PLUTO.frm flies from tezuka 0.1.9 to the device before I could use the 2r2t functionality. Not sure, but I think that had something to do with the changed u-boot environment in fw 0.1.9.
Hope that will help you!
Best regards,
DC0NG

rlabs-oss, 2025-08-31:
Ive tried to boot from sdcard (plutoplus, jumped on SD-H to 3.3v rev2 board) - still boots from qspi flash. 
Built the whole source to tezuka.zip copying the sdimg files to the root of the card. - didnt boot from the sdcard.
From qspi boot, config.txt has mode=2r2t
# fw_printenv mode
mode=2r2t
# fw_printenv attr_name
attr_name=compatible
# fw_printenv attr_val
attr_val=ad9363a
# fw_printenv compatible
compatible=ad9361
# iio_info | grep "voltage.*input"
			voltage0:  (input)
			voltage2:  (input)
			voltage5: vccoddr (input)
			voltage0: vccint (input)
			voltage4: vccpaux (input)
			voltage7: vrefn (input)
			voltage1: vccaux (input)
			voltage2: vccbram (input)
			voltage3: vccpint (input)
			voltage8:  (input)
			voltage6: vrefp (input)
			voltage0:  (input, index: 0, format: le:S12/16>>0)
			voltage1:  (input, index: 1, format: le:S12/16>>0)

no additional voltage(s) for RX2 unless I'm missing the point somewhere?

Basically ethernet and expanded coverage is working, but I'm not sure if the maia dts is causing a conflict or uboot which seems to be focused more on removing the dual rx mode rather than adding it is also in conflict here?
Or is my understanding of how rx2 is used incorrect? main goal is dual channel coherent rx from the plutoplus where both rx1 and rx1 are streaming via iio_read on another machine. -works great for single channel, but never boots where RX2 is present (other than manually switching via ssh via e.g ./switchrf_input.sh rx2

Would be good to find the root cause of this which I'm attempting to figure out :/

wolfgang6444, 2025-08-31:
Not sure if it makes a difference, but it might be helpful to mention the board revision when posting which configuration worked and which not.
I have not seen a config of  tezuka 0.1.9 yet that works in 2r2t for a rev 2 board.

DK1BW

wolfgang6444, 2025-10-28:
O.K. this appears to be a rather well-known problem:

ipaddr and ipaddr_eth are exchanged in config.txt

For the RJ45 to work set:
ipaddr_eth to the ipaddr you want to assign to plutoplus
gateway_eth to the ipaddr of your gateway (router)
ipaddr to somthing outside your netmask
ipaddr_host to something outside your netmask

DK1BW