F5OEO/tezuka_fw #84: What kind of Board / Version of Pluto+ SDR is that ?

helmarw, 2025-03-23:
Hi, i just got a Pluto+ SDR from Banggood (https://www.banggood.com/Pluto+-70MHz~6GHZ-AD9363-SDR-Radio-Receiver-Transmitter-2xRX-2xTX-Compatible-with-ADI-ADALM-PLUTO-PLUTO+-p-2027918.html?cur_warehouse=CN&ID=6287830)

After testing a few things successfully out of the box with gqrx and satsagen i was wondering if i could update the firmware.
Just to be sure what kind of board ive got i opened the casing... this came i bit of a surprise .. the board looks nothing
like that in the pictures:

[image]

[image]

There is a switch for the firmware bottom right and a second switch next to the ZYNQ which function left of the ZYNQ i assume.
The Boot/DFU switch S1 is on the bottom of the board, at least i assume thats its function.

Does anybody have this board and more information about it ?

Is this repository still active ?

thnx
Helmar

helmarw, 2025-03-25:
additional info after some testing:

A USB drive doesnt pop up no matter which USB Port is connected to the PC.

One USB port is labeled "power", the other one is labeled "data", but both give power to the SDR when plugged in

To get the serial port you have to plug the "power" port into the computer, not the "data" port
on my mac its called /dev/tty.usbserial-DN00JL3N, not like tty.modem or something, but i can connect with
screen /dev/tty.usbserial-DN00JL3N 115200 to it

The "data" labeled port turns out to be used for OTG, e.g. to connect a usb-stick for firmware update.

that brings me to the last point, i just gave it a shot updating the firmware with USB-drive
using update_frm.sh, and its working, but only tested reception on one channel with gqrx for now

Few things are puzzling me still:
1) why does the host identification changes each time i reboot the Pluto+ and how can i stop it from doing that?

```
$ ssh root@192.168.100.54
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@helmar.waiczies@mdc-berlin.dehelma
@    WARNING: REMOTE HOST IDENTIFICATION HAS CHANGED!     @
@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@@helmar.waiczies@mdc-berlin.dehelma
IT IS POSSIBLE THAT SOMEONE IS DOING SOMETHING NASTY!
Someone could be eavesdropping on you right now (man-in-the-middle attack)!
It is also possible that a host key has just been changed.
The fingerprint for the ED25519 key sent by the remote host is
SHA256:Mh8ozNB0KSquo6f9qjKmkOnLruGKDRX0/HKrrWKRTVg.
Please contact your system administrator.
Add correct host key in /Users/helmar/.ssh/known_hosts to get rid of this message.
Offending ED25519 key in /Users/helmar/.ssh/known_hosts:71
Host key for 192.168.100.54 has changed and you have requested strict checking.
Host key verification failed.
```

  if i delete the offending key then it works but only until next power cycle

2) if i edit the /opt/config.txt to add a static ip, it saves ist, but once rebooted the changes are gone 

  im pretty new to this device, i assume the fw is somehow write protected and since it does not give a USB drive, unlike written     
  in most of the pluto+ manuals i read, im probably missing some steps to make the changes permanent.
  any help is appreciated :)

jojoosinga, 2025-06-23:
Have you tried ejecting it after changing the file? That is how to saves the changes

CryptoIceMLH, 2025-10-17:
anyone used this board ? i am trying to get the ext ref clock to work.. anyone know how to do it ? no pin headers or any notes on jumper settings

CryptoIceMLH, 2025-10-18:
I got this thing to run with your latest release 0.2.0 and i have attached a leo bodner gps clock to it.. its terrible.. for one it wont switch to a A9361.. it start as a A9363a... it accepted A9364 and saved but it wont do 61. no idea why. 
anyways.. got network working no probs. the main issue i have is the TX is drifting like mad.. and fast.. you turn on qo100 websdr and you can hear it move. i added the gps signal to it .. i added a jumper to this MOI48 pin and jumped across to the pin next to it that was measuring 1.8v. the fw has been to old external and 40mhz and its showing external.. but FML the TX is drifting like a bat of hell. did i anyone have any better luck ?

CryptoIceMLH, 2025-10-19:
yes sir, i installed on this "new" pluto plus ( clone boards with the above looking new layout - the one with out the RF shields etc) 
i used your 0.2.0 plutoplus.zip

you want me to try the  [fishball.zip](https://github.com/F5OEO/tezuka_fw/releases/download/0.2.0/fishball.zip)   ?

u feel it will work ? as i need to strip my Tracker unit down to get to it so i can USB to it ... 

[image]

CryptoIceMLH, 2025-10-19:
no flased via usb with boot.frm & pluto.frm

F5OEO, 2025-10-19:
All boards with SD should use SD instead of flashing internally (which
could be hazardous)
To unbrick , use the SD method with the plutoplus firmware.

CryptoIceMLH, 2025-10-19:
staring at a black blank putty screen

F5OEO, 2025-10-25:
Hope that you solve your issue...Should work with SD Boot.