F5OEO/tezuka_fw #87: HamGeek E310 SDR

sv5byr, 2025-03-27:
Hello Evariste,
I just get my E310 today. All OK with QSPI firmware (0.38 dirty) and recognized as Pluto clone.
Having fun with tezuka 0.1.1 with my Zynq SDR will be nice to have Tezuka on my E310 also.

Thanks in advance 
Mike
SV5BYR

Szczecin-Radio, 2025-03-27:
HamGeek E310 == AntSDR E310. At this time you can get the latest AntSDR firmware v0.39 (SD card) end enjoy the new SDR.

Regards,
/WS

sv5byr, 2025-03-27:
> HamGeek E310 == AntSDR E310. At this time you can get the latest AntSDR firmware v0.39 (SD card) end enjoy the new SDR.
> 
> Regards, /WS

I have done this already. But Tezuka gives more options as TRX1/TRX2 switching....

gariac, 2025-03-31:
https://imgur.com/a/l1YpYCy

I have the version with the big heatsink. The link has the two sides of the board. It comes from HamGeek. Is this the E310?

Also confusing to me (which granted doesn't take much) are the various versions of Tezuka in the binaries. If you compile from source, there are no options. So I wonder which binary I should use?

[image]
[image]

itaniepitas, 2025-03-31:
> https://imgur.com/a/l1YpYCy
>
> I have the version with the big heatsink. The link has the two sides of the board. It comes from HamGeek. Is this the E310?
>
> Also confusing to me (which granted doesn't [...]

No is not E310. Go with libresdr.zip and you will be fine.

gariac, 2025-03-31:
> No is not E310. Go with libresdr.zip and you will be fine.

I spent some time doing image comparisons last night. I believe this board is an "openwifi". I put a similar post on their subreddit:

https://github.com/open-sdr/openwifi-hw/issues/119

I have some formatted sd cards and will try the latest libresdr, but thus far I haven't had any luck. I'm still a little confuse how these binaries get their "flavor" since there is nothing mentioned in picking a board when you build tezuka yourself. 

The board  does work on the binary from the libresdr github. Just only one receive  port.  Also only on usb though that could be my fault.

gariac, 2025-04-01:
> No is not E310. Go with libresdr.zip and you will be fine.

No luck.

This is from the libresdr gitgub. You can see the notebook finds the device. It works in sdr++ (sdrpp).
`
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 27c6:609c Shenzhen Goodix Technology Co.,Ltd. Goodix Fingerprint USB Device
Bus 001 Device 003: ID 0e8d:e616 MediaTek Inc. Wireless_Device
Bus 001 Device 005: ID 0456:b673 Analog Devices, Inc. LibIIO based AD9363 Software Defined Radio [ADALM-PLUTO]
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 002: ID 0bda:8156 Realtek Semiconductor Corp. USB 10/100/1G/2.5G LAN
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

`

This is how the tezuka_fw using the libresdr binary looks. The device is not found.
`

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 001 Device 002: ID 27c6:609c Shenzhen Goodix Technology Co.,Ltd. Goodix Fingerprint USB Device
Bus 001 Device 003: ID 0e8d:e616 MediaTek Inc. Wireless_Device
Bus 002 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 002 Device 002: ID 0bda:8156 Realtek Semiconductor Corp. USB 10/100/1G/2.5G LAN
Bus 003 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 004 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 005 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 006 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub
Bus 007 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Bus 008 Device 001: ID 1d6b:0003 Linux Foundation 3.0 root hub

`

https://devicehunt.com/view/type/usb/vendor/0456/device/B673

itaniepitas, 2025-04-01:
This device is LibreSDR . First install in your PC the USB drivers from https://wiki.analog.com/university/tools/pluto/drivers/windows.
Second extract the files from https://github.com/F5OEO/tezuka_fw/releases/download/0.1.2/libresdr.zip 
Format one SD card as FAT32. 
Select ALL files in folder sdimg from previous zip and copy all of them in SD card. 
Plug the card in to device. Plug the power USB cable.
Thats it.

gariac, 2025-04-02:
> This device is LibreSDR . First install in your PC the USB drivers from https://wiki.analog.com/university/tools/pluto/drivers/windows. Second extract the files from [...]

I know all that. Tezuka doesn't work on the 7020  variant, at least for me.  I have contacted the author directly.

I can run this on usb. 
https://github.com/hz12opensource/libresdr.

itaniepitas, 2025-04-02:
I wish you good luck my friend

sv5byr, 2025-04-03:
Hello Evariste,
I manage to put TEZUKA into E310 renaming pluto.frm to ant.frm. Voila....everything work EXCEPT the LAN.
Any  idea  ???

Thanks in advance
SV5BYR
Mike

Szczecin-Radio, 2025-04-04:
Have you updated flash or just modified SD files? I guess at this time your E310 run a 677MHz single core CPU (xc7z010clg225-1 def.) 512MB DDR, and can reach 8 Msps instead of 2 cores @766MHz (or 866MHz when configured as speed index -3 ; xc7z020clg400-3 def.) 1GB DDR and16 Msps via ethernet. No external clock should work with said regular Pluto firmware...   

Anyway it is quite  impressive, that E310 can even boot with the regular PLUTO firmware. 

Regards,
/WS

Szczecin-Radio, 2025-04-12:
> I manage to put TEZUKA into E310 renaming pluto.frm to ant.frm. Voila....everything work EXCEPT the LAN. Any idea ???

Well, uh...  E310 has a PHY network controller by Marvell. So, you just add to the kernel config 'CONFIG_MARVELL_PHY=y' rebuild the kernel and you (probably) shall see the lost LAN.

Regards,
/WS

sv5byr, 2025-04-13:
Gm, Im not very familiar with Linux. If someone can help me step by step
will be appreciated. Thanks

F5OEO, 2025-10-26:
Need a board to test. Donate to pay me it or ask manufacturer to send me a sample

closed 2025-10-26 by F5OEO