F5OEO/tezuka_fw #88: RIT on wider BW does not work

DG9BFC, 2025-03-28:
when i use a bw bigger as 6 megs the RIT does not work correct
even only 10 Hz set in rit and sdrtv does not decode anymore, testing it in sdrconsole the sound is very dirty / noisy and all signals become WIDE (see picture)

[image]

 ... so on bigger bw settings something is wrong with the RIT

F5OEO, 2025-03-28:
1- Seems you are not working tezuka firmware 
2- SdrConsole with higher bandwidth is still in development (Simon)
3- Add feedback at sdrconsole forum or G4ELI directly

DG9BFC, 2025-03-28:
it is the same fault with the "overclocked" fw from h12opensource or with YOUR fw (tezuka) ... both do not work with RIT and wider Bw as 6 Megs
sidenote: i can receive well with up to 18 megs in sdrconsole and 16 megs in sdrtv
with BOTH fw i have the same garbeled sound ... and i also dropped Simon a note

closed 2025-03-28 by F5OEO