F5OEO/tezuka_fw #92: New Fw maybe Bug 0.1.2

Hackrfan, 2025-03-31:
I have just tested the new Fw 0.1.2 on QO 100 and noticed that the TX signal looks terrible here.
See pictures , for comparison Fw 0.37 dirty 
Can anyone confirm this?

[image]
[image]

DG9BFC, 2025-03-31:
works totally fine here
dg9bfc sigi

[image]

Hackrfan, 2025-04-01:
Here is another picture with version 0.1.1 all settings on the SDR console unchanged.

[image]

raimund56, 2025-04-01:
/can confirm.  Raimund in3her

[image]

F5OEO, 2025-04-02:
Should be fixed with 148a6e0091c75891c7d5554098440eb73c214e48

Hackrfan, 2025-04-04:
Sorry, is the fix already integrated in 0.1.2? Or how can I test if it works?
Greetings Georg

F5OEO, 2025-04-08:
Fix is integrated with 0.1.3. Please confirm.

Hackrfan, 2025-04-08:
The signal with 0.1.3 is much better but still not clean. sorry but it is.
Compare 0.1.1 with 0.1.3 see screen shot

[image]

F5OEO, 2025-04-08:
Can you check that your TX has same issue using websdr ?

DG9BFC, 2025-04-11:
on a fresh install of sdrconsole the tune tone is set to "pulses" .. switch it to constant tone and signal should be clean
what you see is the "keyclicks" from these pulses

[image]

pulses

[image]

settings window for tune tone

[image]

constant tune tone

DG9BFC, 2025-04-11:
other possible fault .. you may have short interruptions (depending on your soundcard settings, buffer settings etc etc)
there can be a dozen faults that may cause a wide and unclean signal 
and i guess fw is not a reason for it cause in sdrconsole the pluto is used in "pass mode" .. means the fw does "nothing" on the signal

Hackrfan, 2025-04-11:
Hello Sigi,
I don't know if I didn't express myself clearly enough, I did the different FW states with one and the same console setting and just came to this behavior shown in the pictures.
I also never claimed that the FW 0.32 you are using has this error. The behavior only occurred from FW version 01.2 with F5OEO. Everything was fine before that. So this is definitely not due to my settings but to the FW. Other amateurs can also understand this.
vy 73 Georg

Translated with DeepL.com (free version)

F5OEO, 2025-10-25:
Closed as inactivity

closed 2025-10-25 by F5OEO