F5OEO/tezuka_fw #93: How to use CS8 with SoapySDR?

VasylSamoilov, 2025-04-04:
I will be very grateful if somebody can explain how to configure CS8 for more bandwidth with SoapySDR.
Should I set env var in tezuka? Command line parameter? Right now I have the following setup:

SoapySDR from here: https://github.com/F5OEO/SoapyPlutoPAPR
Tezuka 0.1.2 / LibreSDR
Right now SoapySDR with SoapyPlutoPAPR TX test gives me ~8 MSPS, RX test segfaults
"Normal" SoapySDR pluto plugin benchmarks at 19.8365 Msps, 79.3461 MBps for RX, 8.30456 Msps, 33.2182 MBps for TX.
Which, frankly speaking, doesn't seems like correct values.

```
$ SoapySDRUtil --rate=20e6 --direction=TX --args="driver=plutosdr"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[INFO] Opening PlutoSDR #0 ip:libresdr.local...
Tezuka firmware
[INFO] Using format CS16 Tezuka.
[INFO] Has direct TX copy: 1
Stream format: CS16
Num channels: 1
Element size: 4 bytes
Begin TX rate test at 20 Msps
Starting stream loop, press Ctrl+C to exit...                                                                                                                                                                     7.53378 Msps     30.1351 MBps
7.7149 Msps     30.8596 MBps
7.67874 Msps    30.715 MBps
$ SoapySDRUtil --rate=20e6 --direction=RX --args="driver=plutosdr"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[INFO] Opening PlutoSDR #0 ip:libresdr.local...
Tezuka firmware
[INFO] Using format CS16 Tezuka.
Tezuka disable
[INFO] Auto setting Buffer Size: 1048576
[INFO] Set MTU Size: 1048576
Stream format: CS16
Num channels: 1
Element size: 4 bytes
Begin RX rate test at 20 Msps
Starting stream loop, press Ctrl+C to exit...
[INFO] Has direct RX copy: 1
Segmentation fault

```

SoapySDRUtil probe:

```
SoapySDRUtil --probe="driver=plutosdr"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

Probe device driver=plutosdr
[INFO] Opening PlutoSDR #0 ip:libresdr.local...
Tezuka firmware

----------------------------------------------------
-- Device identification
----------------------------------------------------
  driver=PlutoSDR
  hardware=ADALM-PLUTO
  ad9361-phy,model=ad9361
  ad9361-phy,xo_correction=40000000
  backend_version=0.25 (git tag: v0.25)
  fw_version=tezuka-eaeb
  hw_model=LibreSDR Rev.5 (Z7020-AD9361)
  hw_model_variant=1
  hw_serial=HQHKGURO7K542YSI
  ip,ip-addr=192.168.160.179
  library_version=0.24 (git tag: v0.24)
  local,kernel=6.1.0
  uri=ip:libresdr.local

----------------------------------------------------
-- Peripheral summary
----------------------------------------------------
  Channels: 1 Rx, 1 Tx
  Timestamps: NO
  Sensors: xadc_temp0, xadc_voltage0, xadc_voltage1, xadc_voltage2, xadc_voltage3, xadc_voltage4, xadc_voltage5, xadc_voltage6, xadc_voltage7, xadc_voltage8, adm1177_current0, adm1177_voltage0, ad9361-phy_temp0, ad9361-phy_voltage2
     * xadc_temp0: 51.677124 C
     * xadc_voltage0 (vccint): 0.979248 V
     * xadc_voltage1 (vccaux): 1.762939 V
     * xadc_voltage2 (vccbram): 0.976318 V
     * xadc_voltage3 (vccpint): 0.974854 V
     * xadc_voltage4 (vccpaux): 1.757812 V
     * xadc_voltage5 (vccoddr): 1.469971 V
     * xadc_voltage6 (vrefp): 1.244385 V
     * xadc_voltage7 (vrefn): -0.000732 V
     * xadc_voltage8: 0.001709 V
     * adm1177_current0:
     * adm1177_voltage0:
     * ad9361-phy_temp0: 39.474000 C
     * ad9361-phy_voltage2: 0.017399 V

----------------------------------------------------
-- RX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: YES
  Stream formats: CS8, CS12, CS16, CF32
  Native format: CS16 [full-scale=2048]
  Antennas: A_BALANCED
  Full gain range: [0, 73] dB
    PGA gain range: [0, 73] dB
  Full freq range: [70, 6000] MHz
    RF freq range: [70, 6000] MHz
  Sample rates: [0.260417, 61.44] MSps
  Filter bandwidths: 1, 2, 3, 4, 5, 8, 9, 10, 12, 15 MHz

----------------------------------------------------
-- TX Channel 0
----------------------------------------------------
  Full-duplex: YES
  Supports AGC: NO
  Stream formats: CS8, CS12, CS16, CF32
  Native format: CS16 [full-scale=32768]
  Antennas: A
  Full gain range: [0, 89] dB
    PGA gain range: [0, 89] dB
  Full freq range: [70, 6000] MHz
    RF freq range: [70, 6000] MHz
  Sample rates: [0.260417, 61.44] MSps
  Filter bandwidths: 1, 2, 3, 4, 5, 8, 9, 10, 12, 15 MHz
```

Szczecin-Radio, 2025-04-11:
If I understand your question, the client side (i.e. gqrx, GNU Radio, SoapySDRUtil) must be aware / accept said $ENV / parameter first. The SoapySDR allows that.  

$ SoapySDRUtil --rate="20e6" --direction=RX --args="driver=plutosdr,uri=ip:192.168.1.10" --format=CS8
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[INFO] Using format CS8.
[INFO] Auto setting Buffer Size: 524288
[INFO] Set MTU Size: 524288
Stream format: CS8
Num channels: 1
Element size: 2 bytes
Begin RX rate test at 20 Msps
Starting stream loop, press Ctrl+C to exit...
[INFO] Has direct RX copy: 1
19.8563 Msps	39.7126 MBps
19.8949 Msps	39.7898 MBps
19.896 Msps	        39.7921 MBps
19.9153 Msps	39.8306 MBps
19.9247 Msps	39.8494 MBps
19.9349 Msps	39.8697 MBps
19.9422 Msps	39.8845 MBps
19.9495 Msps	39.8989 MBps
19.9699 Msps	39.9398 MBps
19.9735 Msps	39.947 MBps
19.9615 Msps	39.9229 MBps
19.948 Msps	        39.896 MBps
19.9507 Msps	39.9014 MBps
19.9668 Msps	39.9336 MBps
19.9688 Msps	39.9377 MBps

~$ SoapySDRUtil --rate="20e6" --direction=RX --args="driver=plutosdr,uri=ip:192.168.1.10" --format=CS16
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[INFO] Using format CS16.
[INFO] Auto setting Buffer Size: 524288
[INFO] Set MTU Size: 524288
Stream format: CS16
Num channels: 1
Element size: 4 bytes
Begin RX rate test at 20 Msps
Starting stream loop, press Ctrl+C to exit...
[INFO] Has direct RX copy: 1
19.7331 Msps	78.9325 MBps
19.8669 Msps	79.4676 MBps
19.9143 Msps	79.657 MBps
19.9364 Msps	79.7457 MBps
19.9474 Msps	79.7895 MBps
19.958 Msps         79.832 MBps

Regards,
/WS

Szczecin-Radio, 2025-04-13:
One more thing. To get beyond 20Msps bareer (no mater if CS8, CS16, CF32 ) one need to modify ADI libiio on both sides, I mean client and server (SDR). Probably...

`$ SoapySDRUtil --rate="30.72e6" --direction=RX --args="driver=plutosdr,uri=ip:192.168.1.10" --format=CS8
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[INFO] Using format CS8.
[INFO] Auto setting Buffer Size: 2097152
[INFO] Set MTU Size: 2097152
Stream format: CS8
Num channels: 1
Element size: 2 bytes
Begin RX rate test at 30.72 Msps
Starting stream loop, press Ctrl+C to exit...
[INFO] Has direct RX copy: 1
19.9892 Msps	39.9784 MBps
19.9904 Msps	39.9809 MBps
19.9908 Msps	39.9816 MBps
19.9904 Msps	39.9808 MBps
20.009 Msps	40.018 MBps
20.0067 Msps	40.0134 MBps
20.0057 Msps	40.0114 MBps
20.0049 Msps	40.0098 MBps
19.9997 Msps	39.9995 MBps
19.9984 Msps	39.9968 MBps
19.9992 Msps	39.9985 MBps
19.9991 Msps	39.9982 MBps
20.0305 Msps	40.0611 MBps
20.0288 Msps	40.0577 MBps
20.029 Msps	40.058 MBps
20.0293 Msps	40.0586 MBps
20.0294 Msps	40.0588 MBps
20.0289 Msps	40.0577 MBps
20.0291 Msps	40.0582 MBps
20.0291 Msps	40.0582 MBps
20.0285 Msps	40.057 MBps
20.0284 Msps	40.0568 MBps
20.0282 Msps	40.0564 MBps
20.0276 Msps	40.0552 MBps
`
Regards,
/WS

F5OEO, 2025-04-14:
@Szczecin-Radio : you don't seem to use tezuka fork : https://github.com/F5OEO/SoapyPlutoPAPR
Here is an example of output (35MS)

```
SoapySDRUtil --rate="35e6" --direction=RX --args="driver=plutosdr,uri=ip:libresdr.local" 
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[INFO] Opening PlutoSDR #0 ip:libresdr.local...
Tezuka firmware
[INFO] Using format CS16 Tezuka.
Tezuka CS8 input
[INFO] Auto setting Buffer Size: 1093632 with 13 kernel 
[INFO] Set MTU Size: 1093632
Stream format: CS16
Num channels: 1
Element size: 4 bytes
Begin RX rate test at 35 Msps
Starting stream loop, press Ctrl+C to exit...
[INFO] Has direct RX copy: 1
33.7856 Msps	135.142 MBps
34.286 Msps	137.144 MBps
34.5299 Msps	138.119 MBps
```

VasylSamoilov, 2025-04-14:
Great news. It seems to be that after initial readiness it would be required to create documentation how to use tezuka with different software.

Szczecin-Radio, 2025-04-15:
> you don't seem to use tezuka fork : https://github.com/F5OEO/SoapyPlutoPAPR

Correct, I just compiled above *PAPR plugin but metrics were similar to the regular SoapyPlutoSDR, so stayed with it for a while. I thing the limitation is my stock firmware with the CPU @866MHz and DDR @533MHz But... as I ran iiod daemon in real-time mode (round-robin scheduler policy) the metrics went beyond 20MSps. 

```
$ SoapySDRUtil --rate="30720000" --direction=RX --args="driver=plutosdr,hostname=192.168.1.10"
######################################################
##     Soapy SDR -- the SDR abstraction library     ##
######################################################

[INFO] Opening PlutoSDR #0 ip:ant.local...
[INFO] Using format CS16.
[INFO] Auto setting Buffer Size: 4194304
[INFO] Set MTU Size: 4194304
Stream format: CS16
Num channels: 1
Element size: 4 bytes
Begin RX rate test at 30.72 Msps
Starting stream loop, press Ctrl+C to exit...
[INFO] Has direct RX copy: 1
22.5245 Msps	90.098 MBps
22.5312 Msps	90.1249 MBps
22.5305 Msps	90.1221 MBps
22.5344 Msps	90.1377 MBps
22.5348 Msps	90.1392 MBps
22.5376 Msps	90.1502 MBps
22.5381 Msps	90.1523 MBps
22.5416 Msps	90.1663 MBps
22.5422 Msps	90.1688 MBps
22.5422 Msps	90.1689 MBps
22.5368 Msps	90.1474 MBps
22.5402 Msps	90.1606 MBps
22.5381 Msps	90.1525 MBps
22.5409 Msps	90.1634 MBps
22.5445 Msps	90.178 MBps
22.5372 Msps	90.149 MBps
22.5375 Msps	90.1501 MBps
22.5361 Msps	90.1442 MBps
22.536 Msps	90.1439 MBps
22.5372 Msps	90.149 MBps
22.5353 Msps	90.1411 MBps
22.5342 Msps	90.1367 MBps

``` 

[image]

And of course temperature has raised immediately.

```
$ SoapySDRUtil --probe="driver=plutosdr,uri=ip:192.168.1.10" 2>/dev/null | grep -e 'temp0:'
     * xadc_temp0: 50.938879 C
     * ad9361-phy_temp0: 53.509000 C

``` 

Regards,
/WS