F5OEO/tezuka_fw #95: DFU install of pluto.dfu onto Pluto+ crashes after 81%

darkstar007, 2025-04-05:
I've tried 0.1.2 and 0.1.1 and both crash at the same point with the same error:
```
$ dfu-util -a firmware.dfu -D pluto.dfu 
dfu-util 0.11

Copyright 2005-2009 Weston Schmidt, Harald Welte and OpenMoko Inc.
Copyright 2010-2021 Tormod Volden and Stefan Schmidt
This program is Free Software and has ABSOLUTELY NO WARRANTY
Please report bugs to http://sourceforge.net/p/dfu-util/tickets/

Match vendor ID from file: 0456
Match product ID from file: b673
Opening DFU capable USB device...
Device ID 0456:b674
Device DFU version 0110
Claiming USB DFU Interface...
Setting Alternate Interface #1 ...
Determining device status...
DFU state(2) = dfuIDLE, status(0) = No error condition is present
DFU mode device DFU version 0110
Device returned transfer size 4096
Copying data from PC to DFU device
Download        [====================     ]  81%     22941696 bytes failed!
DFU state(10) = dfuERROR, status(14) = Something went wrong, but the device does not know what it was
```

DFU update of the boot.dfu from your 0.1.x zip file works fine.

Reverting back to 0.33-dirty via DFU also works.

This is a Rev1 Pluto+ (going by the 1.8V label on the JTAG connector.

Any suggestions?

Thanks,
Matt

ps. Could you update the release filenames to have the version number in them? Thanks.

F5OEO, 2025-07-19:
Let me know if it is fiwed with latest firmware. I suggest you to use sd card booting instead of flash.

darkstar007, 2025-08-24:
Yeah, managed to update to 0.1.9 OK via sd card:
```
$ iio_attr  -C fw_version --uri usb:
fw_version: tezuka-0.1.9
```

(Sorry for the delay, was having all sorts of USB issues when I upgraded my video card - now fixed!).

Thanks 
Matt