F5OEO/tezuka_fw #96: DHCP lease - weird problem with two differnet DHCP server (plutoplus)

dm6tt, 2025-04-06:
Hi guys,
I have a weird DHCP phenomenon and need your help.

When I use openwrt as a DHCP server, I will finally have no IP-address on eth0 configured.
When I use another network with a commercial router the IP-address is assigned and configured.

I have traced the traffic with wireshark and checked the logs and it seems, that the dhcp-client and avahi come in trouble.

**Case 1: Power off/on without proper reboot via cli**

First, the running configuration wit a  commercial router (Subnet 192.168.3.0/24)

```
2559	2187.554421998	0.0.0.0	255.255.255.255	        DHCP	346	DHCP Discover - Transaction ID 0xf4166b45
2561	2187.568877060	192.168.3.1	192.168.3.111	DHCP	356	DHCP Offer    - Transaction ID 0xf4166b45
2562	2187.634422185	0.0.0.0	255.255.255.255	        DHCP	346	DHCP Request  - Transaction ID 0xf4166b45
2565	2187.709914698	192.168.3.1	192.168.3.111	DHCP	356	DHCP ACK      - Transaction ID 0xf4166b45
2568	2187.975543926	192.168.3.111	192.168.3.1	DHCP	346	DHCP Release  - Transaction ID 0x6e577217
2578	2194.474614623	0.0.0.0	255.255.255.255	        DHCP	346	DHCP Discover - Transaction ID 0xa3720d37
2581	2194.486637228	192.168.3.1	192.168.3.111	DHCP	356	DHCP Offer    - Transaction ID 0xa3720d37
2582	2194.554594137	0.0.0.0	255.255.255.255	        DHCP	346	DHCP Request  - Transaction ID 0xa3720d37
2585	2194.628397661	192.168.3.1	192.168.3.111	DHCP	356	DHCP ACK      - Transaction ID 0xa3720d37
```

One can see, that an IP-address is aligned and then immediately released.
After that a new offer is started (0xa3720d37) and the address is the assigned.

In the /var/log/messages that looks inj that way:
```
Jan  1 00:00:08 (none) kern.info kernel: macb e000b000.ethernet eth0: Link is Up - 1Gbps/Full - flow control tx
Jan  1 00:00:10 (none) user.notice root: udhcpc: broadcasting discover
Jan  1 00:00:10 (none) user.notice root: udhcpc: broadcasting select for 192.168.3.111, server 192.168.3.1
Jan  1 00:00:11 (none) user.notice root: udhcpc: lease of 192.168.3.111 obtained from 192.168.3.1, lease time 1814400
Jan  1 00:00:11 (none) user.notice root: deleting routers
Jan  1 00:00:11 (none) user.notice root: adding dns 192.168.3.1
Jan  1 00:00:11 (none) daemon.info ifplugd(eth0): started: BusyBox v1.36.1 (2025-03-30 14:56:23 UTC)
Jan  1 00:00:11 (none) daemon.info ifplugd(eth0): using SIOCETHTOOL detection mode
Jan  1 00:00:11 (none) daemon.info ifplugd(eth0): link is up
Jan  1 00:00:11 (none) daemon.info ifplugd(eth0): executing '/etc/ifplugd/ifplugd.action eth0 up'

...
Jan  1 00:00:15 (none) kern.info kernel: macb e000b000.ethernet eth0: Link is Up - 1Gbps/Full - flow control tx
Jan  1 00:00:16 (none) daemon.info : starting pid 555, tty '/dev/ttyPS0': '/sbin/getty -L  ttyPS0 0 vt100 '
Jan  1 00:00:16 (none) daemon.info : starting pid 558, tty '/dev/ttyGS0': '/sbin/getty -L ttyGS0 0 vt100 '
Jan  1 00:00:17 (none) daemon.info avahi-daemon[456]: Joining mDNS multicast group on interface eth0.IPv4 with address 192.168.3.111.
Jan  1 00:00:17 (none) daemon.info avahi-daemon[456]: New relevant interface eth0.IPv4 for mDNS.
Jan  1 00:00:17 (none) daemon.info avahi-daemon[456]: Registering new address record for 192.168.3.111 on eth0.IPv4.
Jan  1 00:00:18 (none) daemon.info ifplugd(eth0): exit code: 0

```

eth0 looks:

```
# ifconfig eth0
eth0      Link encap:Ethernet  HWaddr 00:60:88:B8:85:18  
          inet addr:192.168.3.111  Bcast:192.168.3.255  Mask:255.255.255.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:1125 errors:0 dropped:0 overruns:0 frame:0
          TX packets:370 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:186484 (182.1 KiB)  TX bytes:33257 (32.4 KiB)
          Interrupt:34 Base address:0xb000 
```

When doing the same stuff against the openwrt router (Subnet 192.168.11.0/24)

```
2228	1909.837747670	0.0.0.0	255.255.255.255	                DHCP	346	DHCP Discover - Transaction ID 0xf452a95a
2229	1909.840812336	192.168.11.1	192.168.11.219	        DHCP	346	DHCP Offer    - Transaction ID 0xf452a95a
2230	1909.917733935	0.0.0.0	255.255.255.255	                DHCP	346	DHCP Request  - Transaction ID 0xf452a95a
2231	1909.929654497	192.168.11.1	192.168.11.219	        DHCP	346	DHCP ACK      - Transaction ID 0xf452a95a
2234	1910.212588050	192.168.11.219	192.168.11.1	        DHCP	346	DHCP Release  - Transaction ID 0x74026249
2237	1916.697950565	0.0.0.0	255.255.255.255	                DHCP	346	DHCP Discover - Transaction ID 0x8572f25a
2241	1920.385283536	192.168.11.1	192.168.11.219	        DHCP	346	DHCP Offer    - Transaction ID 0x8572f25a

```
The address is assinged, the released but the 2nd assingment get stucked
After the Offer, no Request is sent from the plutolus 
The messages shows:
```
Jan  1 00:00:08 (none) kern.info kernel: macb e000b000.ethernet eth0: Link is Up - 1Gbps/Full - flow control tx
Jan  1 00:00:10 (none) user.notice root: udhcpc: broadcasting discover
Jan  1 00:00:10 (none) user.notice root: udhcpc: broadcasting select for 192.168.11.219, server 192.168.11.1
Jan  1 00:00:10 (none) user.notice root: udhcpc: lease of 192.168.11.219 obtained from 192.168.11.1, lease time 345600
Jan  1 00:00:10 (none) user.notice root: deleting routers
Jan  1 00:00:10 (none) user.notice root: adding dns 192.168.11.1
...
Jifcan  1 00:00:15 (none) daemon.info : starting pid 553, tty '/dev/ttyPS0': '/sbin/getty -L  ttyPS0 0 vt100 '
Jan  1 00:00:15 (none) daemon.info : starting pid 556, tty '/dev/ttyGS0': '/sbin/getty -L ttyGS0 0 vt100 '
Jan  1 00:00:20 (none) daemon.info avahi-autoipd(eth0)[662]: Found user 'avahi' (UID 100) and group 'avahi' (GID 101).
Jan  1 00:00:24 (none) kern.info kernel: macb e000b000.ethernet eth0: Link is Up - 1Gbps/Full - flow control tx
Jan  1 00:00:25 (none) daemon.info avahi-autoipd(eth0)[662]: Callout BIND, address 169.254.9.98 on interface eth0
Jan  1 00:00:25 (none) daemon.info avahi-daemon[456]: Joining mDNS multicast group on interface eth0.IPv4 with address 169.254.9.98.
Jan  1 00:00:25 (none) daemon.info avahi-daemon[456]: New relevant interface eth0.IPv4 for mDNS.
Jan  1 00:00:25 (none) daemon.info avahi-daemon[456]: Registering new address record for 169.254.9.98 on eth0.IPv4.
Jan  1 00:00:29 (none) daemon.info avahi-autoipd(eth0)[662]: Successfully claimed IP address 169.254.9.98
Jan  1 00:00:29 (none) daemon.info ifplugd(eth0): exit code: 1
Jan  1 00:00:29 (none) daemon.err ifplugd(eth0): exiting

```
Here, the ifplugd exists with code 1, so I assume that this is the reason for the missing DHCP handshake packages after Offer.

The interface:
```
# ifconfig 
eth0      Link encap:Ethernet  HWaddr 00:60:88:B8:85:18  
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:30 errors:0 dropped:0 overruns:0 frame:0
          TX packets:52 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000 
          RX bytes:5158 (5.0 KiB)  TX bytes:5486 (5.3 KiB)
          Interrupt:34 Base address:0xb000 

eth0:avahi Link encap:Ethernet  HWaddr 00:60:88:B8:85:18  
          inet addr:169.254.9.98  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          Interrupt:34 Base address:0xb000 
....
```

**Case 2:  reboot via cli**

In that case, everything works fine.
It seems, that the release of the (old) DHCP lease is only done if we do a power off/on and no reboot via cli

DHCP with reboot:

```
154	64.614022889	0.0.0.0	255.255.255.255	                DHCP	346	DHCP Discover - Transaction ID 0x950dc48
155	64.617235132	192.168.11.1	192.168.11.219	        DHCP	346	DHCP Offer    - Transaction ID 0x950dc48
156	64.693980312	0.0.0.0	255.255.255.255	                DHCP	346	DHCP Request  - Transaction ID 0x950dc48
157	64.705543144	192.168.11.1	192.168.11.219	        DHCP	346	DHCP ACK      - Transaction ID 0x950dc48
```

DHCP with power off/on:
```
338	119.130294608	0.0.0.0	255.255.255.255	               DHCP	346	DHCP Discover - Transaction ID 0x55574413
339	119.133511950	192.168.11.1	192.168.11.219	       DHCP	346	DHCP Offer    - Transaction ID 0x55574413
340	119.250280717	0.0.0.0	255.255.255.255	               DHCP	346	DHCP Request  - Transaction ID 0x55574413
341	119.261930642	192.168.11.1	192.168.11.219	       DHCP	346	DHCP ACK      - Transaction ID 0x55574413
344	119.555697331	192.168.11.219	192.168.11.1	       DHCP	346	DHCP Release  - Transaction ID 0x5a79377e

```

So we have two issues:
- Without proper reboot a release is sent after the the ACK
- When the openwrt DHCP server assigns the leases, the client exit with error code 1

Any hints where to search for?
Where and how is the old lease stored? I'm booting from the flash directly?

Thanks
Marcus, DM6TT

F5OEO, 2025-04-07:
Wow, thanks for investigations. Not easy to reproduce here. Can you try with 0.1.3 which disables ipv6 by default ? Don't know of it is related. Yes, avahi and dns seems to have concurent behaviors..need to investigate.

dm6tt, 2025-04-10:
Thanks for reply.
I will test it with 0.1.3.
Since I need the test setup for that, I will need some time.

Other idea: Do we really need avahi  or is it sufficient using the 'normal' dhcp-client for network setup?
I saw in your config that the the avahi  can be switched of, perhaps that would be easier.

embed-3d, 2025-04-17:
I think I have the same issue and I'm able to reproduce it.

```
Jan  1 00:00:04 (none) user.notice root: ip: SIOCGIFFLAGS: No such device
Jan  1 00:00:04 (none) user.notice root: udhcpc: started, v1.36.1
Jan  1 00:00:04 (none) kern.info kernel: macb e000b000.ethernet eth0: PHY [e000b000.ethernet-ffffffff:01] driver [RTL8211E Gigabit Ethernet] (irq=POLL)
Jan  1 00:00:04 (none) kern.info kernel: macb e000b000.ethernet eth0: configuring for phy/rgmii-id link mode
Jan  1 00:00:04 (none) user.notice root: udhcpc: broadcasting discover
Jan  1 00:00:07 (none) user.notice root: udhcpc: broadcasting discover
Jan  1 00:00:08 (none) kern.info kernel: macb e000b000.ethernet eth0: Link is Up - 1Gbps/Full - flow control off
Jan  1 00:00:10 (none) user.notice root: udhcpc: broadcasting discover
Jan  1 00:00:13 (none) daemon.info avahi-autoipd(eth0)[357]: Found user 'avahi' (UID 100) and group 'avahi' (GID 101).
Jan  1 00:00:13 (none) daemon.info avahi-autoipd(eth0)[357]: Successfully dropped root privileges.
Jan  1 00:00:13 (none) daemon.info avahi-autoipd(eth0)[357]: Starting with address 169.254.15.120
Jan  1 00:00:17 (none) daemon.info avahi-autoipd(eth0)[357]: Callout BIND, address 169.254.15.120 on interface eth0
Jan  1 00:00:21 (none) daemon.info avahi-autoipd(eth0)[357]: Successfully claimed IP address 169.254.15.120
Jan  1 00:00:21 (none) user.notice root: udhcpc: no lease, failing
Jan  1 00:00:21 (none) daemon.info ifplugd(eth0): started: BusyBox v1.36.1 (2025-04-07 20:31:45 UTC)
Jan  1 00:00:21 (none) daemon.info ifplugd(eth0): using SIOCETHTOOL detection mode
Jan  1 00:00:21 (none) daemon.info ifplugd(eth0): link is up

```

The issue is that the udhcpc is sending dhcp broadcast before the link is up.

When I manually execute `/sbin/udhcpc` I get a proper ip address.

[Here](https://unix.stackexchange.com/questions/68049/wait-for-interface-to-brought-up-before-udhcp)  is a similar issue.

We need to wait till the network link is up before we execute this command: https://github.com/analogdevicesinc/buildroot/blob/e783aadccebad3413b3d60fcfe98a25eb395d839/board/pluto/S40network#L130

In my personal build scipts for linux i do the following:
```
cat <<EOT > /etc/network/interfaces
auto lo
iface lo inet loopback
auto eth0
allow-hotplug eth0
iface eth0 inet dhcp
EOT
```
i could imagine that we are just missing `allow-hotplug eth0`

embed-3d, 2025-04-17:
> I think I have the same issue and I'm able to reproduce it.
>
> ```
> Jan  1 00:00:04 (none) user.notice root: ip: SIOCGIFFLAGS: No such device
> Jan  1 00:00:04 (none) user.notice root: udhcpc: started, [...]

This didn't help.
I think switching to systemd and networkd might be the best soloution, but will break compatibility to the existing code from analog

twi42, 2025-06-10:
I'm powering a LibreSDR with an IEEE 802.3af 4A splitter, from a Cisco 1000 PoE switch, which also acts as the DHCP server. If I use normal spanning tree timeouts on the port (no spanning-tree portfast), DHCPv4 requests never reach the server. With portfast, I see that the assignment happens twice on boot:

`Jun 10 07:56:14.794 UTC: DHCPD: interface GigabitEthernet1/0/14 coming up
Jun 10 07:56:15.465 UTC: %ILPOWER-7-DETECT: Interface Gi1/0/14: Power Device detected: IEEE PD
Jun 10 07:56:16.108 UTC: %ILPOWER-5-POWER_GRANTED: Interface Gi1/0/14: Power granted
Jun 10 07:56:16.791 UTC: %LINK-3-UPDOWN: Interface GigabitEthernet1/0/14, changed state to down
Jun 10 07:56:20.760 UTC: %LINK-3-UPDOWN: Interface GigabitEthernet1/0/14, changed state to up
Jun 10 07:56:20.760 UTC: DHCPD: interface GigabitEthernet1/0/14 coming up
Jun 10 07:56:21.760 UTC: %LINEPROTO-5-UPDOWN: Line protocol on Interface GigabitEthernet1/0/14, changed state to up
Jun 10 07:56:28.395 UTC: %LINEPROTO-5-UPDOWN: Line protocol on Interface GigabitEthernet1/0/14, changed state to down
Jun 10 07:56:29.396 UTC: %LINK-3-UPDOWN: Interface GigabitEthernet1/0/14, changed state to down
Jun 10 07:56:32.560 UTC: %LINK-3-UPDOWN: Interface GigabitEthernet1/0/14, changed state to up
Jun 10 07:56:32.560 UTC: DHCPD: interface GigabitEthernet1/0/14 coming up
Jun 10 07:56:32.862 UTC: DHCPD: Sending notification of DISCOVER:
Jun 10 07:56:32.862 UTC:   DHCPD: htype 1 chaddr 0060.88a2.004a
Jun 10 07:56:32.862 UTC:   DHCPD: interface = Vlan2
Jun 10 07:56:32.863 UTC:   DHCPD: class id 756468637020312e33362e31
Jun 10 07:56:32.863 UTC:   DHCPD: out_vlan_id 0
Jun 10 07:56:32.863 UTC: DHCPD: Sending notification of DISCOVER:
Jun 10 07:56:32.863 UTC:   DHCPD: htype 1 chaddr 0060.88a2.004a
Jun 10 07:56:32.863 UTC:   DHCPD: interface = Vlan2
Jun 10 07:56:32.863 UTC:   DHCPD: class id 756468637020312e33362e31
Jun 10 07:56:32.863 UTC:   DHCPD: out_vlan_id 0
Jun 10 07:56:32.863 UTC: DHCPD: DHCPOFFER notify setup address x.x.x.x mask 255.255.255.192
Jun 10 07:56:32.942 UTC: DHCPD: Sending notification of ASSIGNMENT:
Jun 10 07:56:32.942 UTC:  DHCPD: address x.x.x.x mask 255.255.255.192
Jun 10 07:56:32.942 UTC:   DHCPD: htype 1 chaddr 0060.88a2.004a
Jun 10 07:56:32.942 UTC:   DHCPD: lease time remaining (secs) = 86400
Jun 10 07:56:32.942 UTC:   DHCPD: interface = Vlan2
Jun 10 07:56:32.942 UTC:   DHCPD: out_vlan_id 0
Jun 10 07:56:33.298 UTC: DHCPD: Sending notification of TERMINATION:
Jun 10 07:56:33.298 UTC:  DHCPD: address x.x.x.x mask 255.255.255.192
Jun 10 07:56:33.298 UTC:  DHCPD: reason flags: RELEASE 255.147 id    013e.bfeb.ce86.5a  Jun 09 2025 07:24 PM
Jun 10 07:56:33.298 UTC:   DHCPD: htype 1 chaddr 0060.88a2.004a
Jun 10 07:56:33.298 UTC:   DHCPD: lease time remaining (secs) = 86399
Jun 10 07:56:33.298 UTC:   DHCPD: interface = Vlan2
Jun 10 07:56:33.298 UTC:   DHCPD: out_vlan_id 0
Jun 10 07:56:33.298 UTC: DHCPD: Destroy binding called for 718EB48
Jun 10 07:56:33.298 UTC: DHCPD: remembering binding x.x.x.x from pool B11-user
Jun 10 07:56:33.298 UTC: DHCPD: returned x.x.x.x to address pool B11-user.
Jun 10 07:56:33.560 UTC: %LINEPROTO-5-UPDOWN: Line protocol on Interface GigabitEthernet1/0/14, changed state to up
Jun 10 07:56:35.290 UTC: %LINEPROTO-5-UPDOWN: Line protocol on Interface GigabitEthernet1/0/14, changed state to down
Jun 10 07:56:36.291 UTC: %LINK-3-UPDOWN: Interface GigabitEthernet1/0/14, changed state to down
Jun 10 07:56:39.366 UTC: %LINK-3-UPDOWN: Interface GigabitEthernet1/0/14, changed state to up
Jun 10 07:56:39.366 UTC: DHCPD: interface GigabitEthernet1/0/14 coming up
Jun 10 07:56:40.382 UTC: %LINEPROTO-5-UPDOWN: Line protocol on Interface GigabitEthernet1/0/14, changed state to up
Jun 10 07:56:40.577 UTC: DHCPD: Sending notification of DISCOVER:
Jun 10 07:56:40.577 UTC:   DHCPD: htype 1 chaddr 0060.88a2.004a
Jun 10 07:56:40.577 UTC:   DHCPD: interface = Vlan2
Jun 10 07:56:40.577 UTC:   DHCPD: class id 756468637020312e33362e31
Jun 10 07:56:40.577 UTC:   DHCPD: out_vlan_id 0
Jun 10 07:56:40.577 UTC: DHCPD: Sending notification of DISCOVER:
Jun 10 07:56:40.577 UTC:   DHCPD: htype 1 chaddr 0060.88a2.004a
Jun 10 07:56:40.577 UTC:   DHCPD: interface = Vlan2
Jun 10 07:56:40.577 UTC:   DHCPD: class id 756468637020312e33362e31
Jun 10 07:56:40.577 UTC:   DHCPD: out_vlan_id 0
Jun 10 07:56:40.577 UTC: DHCPD: Trying to reuse remembered binding x.x.x.x
Jun 10 07:56:40.577 UTC: DHCPD: client requests x.x.x.x.
Jun 10 07:56:40.577 UTC: DHCPD: Reusing remembered binding x.x.x.x
Jun 10 07:56:40.577 UTC: DHCPD: recovering binding x.x.x.x
Jun 10 07:56:40.577 UTC: DHCPD: removing binding x.x.x.x from pool B11-user remember queue
Jun 10 07:56:40.578 UTC: DHCPD: assigned IP address x.x.x.x to client 0100.6088.a200.4a. (1334 0)
Jun 10 07:56:40.578 UTC: DHCPD: DHCPOFFER notify setup address x.x.x.x mask 255.255.255.192
Jun 10 07:56:40.677 UTC: DHCPD: Sending notification of ASSIGNMENT:
Jun 10 07:56:40.677 UTC:  DHCPD: address x.x.x.x mask 255.255.255.192
Jun 10 07:56:40.677 UTC:   DHCPD: htype 1 chaddr 0060.88a2.004a
Jun 10 07:56:40.677 UTC:   DHCPD: lease time remaining (secs) = 86400
Jun 10 07:56:40.677 UTC:   DHCPD: interface = Vlan2
Jun 10 07:56:40.677 UTC:   DHCPD: out_vlan_id 0`

Meanwhile I realized that IPv6 SLAAC works perfectly! And also eg. SDR Console supports it. Regardless of spanning tree timeouts.