F5OEO/tezuka_fw #97: FW Tezuka V0.1.2 with LibreSDR/ZynqSDR & SDR Television TX output distortion

lubuntu-l100, 2025-04-08:
FW Tezuka V0.1.2 with LibreSDR & SDR Television TX output distortion.

Hello Evariste, I did today some tests with the LibrSDR/Zynqsdr and FW Tezika v0.1.2, combined with a DATV transmission with SDR Television, and I noticed that the DATV output created by SDR Television was a completed distortion, with a DATV signal that I was not able to decode, but sideways of the DATV signal was an enormous distortion signal that lifts up the noise level on the WB transponder, See image below.

[image]

I did the same test with the LibreSDR/ZynqSDR and fw v0.1.0 and than I have a relative good signal, easily to decode with only a few times some glitches, but that is a known issue of the SDR Television v1.0. Simon already solved that issue.

So something is going completely wrong during SDR Television transmission with FW 0.1.2. As I said FW 0.1.0 is working fine, see image below.

[image]

I hope this information will help you to identify of what is going wrong during DAT TX with SDR Television.

73 PA2JSA

lubuntu-l100, 2025-04-08:
The DATV Signal is completely at the right at 10.499.250 on both images.

On the first image you could see that the noise floor is much higher than on the second image. Even Open Tuner is thinking that it is a normal transmission signal due to the jumping text, left to the signal.

PA2JSA

F5OEO, 2025-04-08:
Yes, 0.1.2 is buggy. Should be fixed with 0.1.3, please confirm.

lubuntu-l100, 2025-04-08:
Hi Evariste,

I will download and install v0 1.3 and will let you know the results.

Thanks for your quick reply.

73 Jaap

Pa2jsa

⁣BlueMail voor Android downloaden ​

closed 2025-04-08 by F5OEO