F5OEO/tezuka_fw #98: fishball

Lowsp00f, 2025-04-09:
Hi guys, I recently received a fishball. It had a 128mb sd card with the following files.
I copied them here.Maybe they will help someone.
https://github.com/Lowsp00f/fishball

But I'm having problems updating the firmware and putting the device into dfu mode.
What position should the switches on the board be? Has anyone been able to put the device into dfu mode? 

What I tried -
1. I went through all the switch combinations on the board.
2. Copied the frm firmware files (I only tried with fishball and pluto+).Then the eject device flashes green and after installation, the device is rebooted and 1 red LED is lit.You can't boot from a USB flash drive without files.I tried files from the sdimg folder for two devices.
3.Copied the default boot.bin and other fishball files onto the sd-card.
This item brought some movement - 
The firmware has been updated, if you can call it that.It won't boot without an sd card.

F5OEO, 2025-04-09:
I am actively developp for fishball. For now I concentrate on SD boot as it is the best way to update and not break everything. I will test to flash it soon. 
It seems that tezuka is running properly on sd, can you confirm ?

Szczecin-Radio, 2025-04-09:
Negative

I had to erase corrupted flash (thanks to the help of Hamgeek's support) and... device is only able to boot with the original SD firmware. With Tezuka firmware it does not even start to boot. A red (power) led is on as well as a white one (user) but not blinking.. 

Please, bear in mind that my device is the other version of 'fishball'  XC7Z020CLG400-2 and 1GB DDR3L (MT41K256M16TW-107:P) based, so maybe it does matter.

Anyway - no github repositories, no Vivado / Any trace of sources, no *.dfu / *.frm binaries. Just bare metal - with some "unknown" firmware...  
  
Regards,
/WS

Lowsp00f, 2025-04-09:
> I am actively developp for fishball. For now I concentrate on SD boot as it is the best way to update and not break everything. I will test to flash it soon. It seems that tezuka is running properly on sd, can you confirm ?

Yep. Attach report on fist post.

I confirm that I was able to start with some nuances. when I dug into pluto, I found that in the update.sh script there is a hash check. I also found that if you first copy boot.frm to the root and do reject, then the partition is updated. it is also likely that it needs flash drives no larger than 128 MB. In general, the firmware of this board looks confident. I think you need to figure out how to get dfu mode

Lowsp00f, 2025-04-09:
> Negative
> 
> I had to erase corrupted flash (thanks to the help of Hamgeek's support) and... device is only able to boot with the original SD firmware. With Tezuka firmware it does not even start to boot. A red (power) led is on as well as a white one (user) but not blinking..
> 
> Please, bear in mind that my device is the other version of 'fishball' XC7Z020CLG400-2 and 1GB DDR3L (MT41K256M16TW-107:P) based, so maybe it does matter.
> 
> Anyway - no github repositories, no Vivado / Any trace of sources, no *.dfu / *.frm binaries, no documentation at all. Just bare metal - with some of "unknown" firmware...
> 
> Regards, /WS
Man copy files in my repo into sdcard and try boot

Szczecin-Radio, 2025-04-09:
Negative

Still can not boot. I just tried with the firmware v0.1.2 and v0.1.3. with 'fishball' not the 'Neptune ' one.

[image]

Regards,
/WS

Lowsp00f, 2025-04-10:
Jampers?

[image]

Szczecin-Radio, 2025-04-10:
Jumpers settings are the same. The files you put intorepository are the same as delivered with a SDR. 

60670b8ee8bceb393dbadc89937101a386a024ab  fishball-main/BOOT.bin
986c29f2a4fba609d59fd2ac70a1e9aab6592f0e  fishball-main/devicetree.dtb
06fccab63629c0cb2c2ef0a4ca19ece377602e50  fishball-main/uEnv.txt
93a47dc26b829a575f1b22fd864991e17e103d6c  fishball-main/uImage
6ad7a42fbf6e120811b7f0073252c688e2e420f9  fishball-main/uramdisk.image.gz

60670b8ee8bceb393dbadc89937101a386a024ab  SD-BACKUP/BOOT.bin
986c29f2a4fba609d59fd2ac70a1e9aab6592f0e  SD-BACKUP/devicetree.dtb
06fccab63629c0cb2c2ef0a4ca19ece377602e50  SD-BACKUP/uEnv.txt
93a47dc26b829a575f1b22fd864991e17e103d6c  SD-BACKUP/uImage
6ad7a42fbf6e120811b7f0073252c688e2e420f9  SD-BACKUP/uramdisk.image.gz

My 'fishball' does boot from 125MB SD and 512 MB SD (an original firmware) Both types of 'fishball' look identical, both - should - boot from the same SD card / firmware, but... my refuses ( I mean with Tezuka firmware, not the original one)

Regards,
/WS

F5OEO, 2025-04-14:
@Szczecin-Radio  Can you save what serial console saying ? (plug usb nearest of ethernet, serial com at 115200 baud)

Szczecin-Radio, 2025-04-14:
I. Boot sequence ; manufacturer SD (reset button press) ; blue led on ; white led blinking

```
# tio --baudrate 115200 --databits 8 --flow none --stopbits 1 --parity none /dev/ttyUSB1
[14:18:29.790] tio 3.8
[14:18:29.790] Press ctrl-t q to quit
[14:18:29.791] Connected to /dev/ttyUSB1

U-Boot PlutoSDR  (Nov 08 2024 - 23:50:32 -0800)

I2C:   ready
DRAM:  ECC disabled 1 GiB
MMC:   sdhci@e0100000: 0
SF: Detected W25Q128BV with page size 256 Bytes, erase size 4 KiB, total 16 MiB
*** Warning - bad CRC, using default environment

In:    serial@e0001000
Out:   serial@e0001000
Err:   serial@e0001000
Model: FISH Ball SDR Board
reading uEnv.txt
7585 bytes read in 17 ms (435.5 KiB/s)
Importing environment from SD ...
Hit any key to stop autoboot:  0 
Device: sdhci@e0100000
Manufacturer ID: 5
OEM: c
Name: asdfg 
Tran Speed: 50000000
Rd Block Len: 512
SD version 1.10
High Capacity: No
Capacity: 480 MiB
Bus Width: 4-bit
Erase Group Size: 512 Bytes
reading uEnv.txt
7585 bytes read in 17 ms (435.5 KiB/s)
Loaded environment from uEnv.txt
Importing environment from SD ...
Copying Linux from SD to RAM...
reading uImage
4543704 bytes read in 395 ms (11 MiB/s)
reading devicetree.dtb
22516 bytes read in 18 ms (1.2 MiB/s)
reading uramdisk.image.gz
6757608 bytes read in 577 ms (11.2 MiB/s)
## Booting kernel from Legacy Image at 02080000 ...
   Image Name:   Linux-5.15.0
   Image Type:   ARM Linux Kernel Image (uncompressed)
   Data Size:    4543640 Bytes = 4.3 MiB
   Load Address: 00008000
   Entry Point:  00008000
   Verifying Checksum ... OK
## Loading init Ramdisk from Legacy Image at 04000000 ...
   Image Name:   
   Image Type:   ARM Linux RAMDisk Image (gzip compressed)
   Data Size:    6757544 Bytes = 6.4 MiB
   Load Address: 00000000
   Entry Point:  00000000
   Verifying Checksum ... OK
## Flattened Device Tree blob at 02000000
   Booting using the fdt blob at 0x2000000
   Loading Kernel Image ... OK
   Loading Ramdisk to 1f98e000, end 1ffffca8 ... OK
   Loading Device Tree to 1f985000, end 1f98d7f3 ... OK
Unable to update property /axi/ethernet@e000b000:mac-address, err=FDT_ERR_NOTFOUND
Unable to update property /axi/ethernet@e000b000:local-mac-address, err=FDT_ERR_NOTFOUND

Starting kernel ...

Booting Linux on physical CPU 0x0
Linux version 5.15.0 (root@ubuntu) (arm-linux-gnueabihf-gcc (Linaro GCC 7.3-2018.05) 7.3.1 20180425 [linaro-7.3-2018.05 revision d29120a424ecfbc167ef90065c0eeb7f91977701], GNU ld (Linaro_Binutils-2018.05) 2.28.2.20170706) #7 SMP PREEMPT Sun Nov 3 04:49:37 PST 2024
CPU: ARMv7 Processor [413fc090] revision 0 (ARMv7), cr=18c5387d
CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
OF: fdt: Machine model: FISH Ball PlutoSDR Rev.A (Z7020/AD9361)
printk: bootconsole [earlycon0] enabled
Memory policy: Data cache writealloc
cma: Reserved 256 MiB at 0x30000000
Zone ranges:
  Normal   [mem 0x0000000000000000-0x000000002fffffff]
  HighMem  [mem 0x0000000030000000-0x000000003fffffff]
Movable zone start for each node
Early memory node ranges

``` 

[image]

II.  Boot sequence ; Tezuka 0.1.3 SD (reset button press)  ; blue led off ; white led on

```
# tio --baudrate 115200 --databits 8 --flow none --stopbits 1 --parity none /dev/ttyUSB1
[14:45:35.045] tio 3.8
[14:45:35.045] Press ctrl-t q to quit
[14:45:35.046] Connected to /dev/ttyUSB1

BLANK SCREEN

[14:46:07.267] Disconnected

# tio --baudrate 115200 --databits 8 --flow none --stopbits 1 --parity none /dev/ttyUSB0
[14:46:11.957] tio 3.8
[14:46:11.957] Press ctrl-t q to quit
[14:46:11.958] Connected to /dev/ttyUSB0

BLANK SCREEN

[14:46:41.683] Disconnected

``` 

[image]

So, maybe there is something wrong with my hardware / flash (zeroized due to previous errors)

Regards,
/WS

F5OEO, 2025-04-14:
@Szczecin-Radio  This is because the First Stage Boot Loader is not right, because you run the 7020 model. I don't have this 7020 to test, so I didn't develop Tezuka for it. Hope to get some money contribution to get it or from manufacturer to test it.

JoyBed, 2025-04-21:
> @Szczecin-Radio  This is because the First Stage Boot Loader is not right, because you run the 7020 model. I don't have this 7020 to test, so I didn't develop Tezuka for it. Hope to get some money contribution to get it or from manufacturer to test it.

Hi! I have the 7020 Fishball too, in the Vivado project you can just change part from 7010 to 7020. Thats all, I can test it out for you. Also can you do me a big favor and send me the constraints file for the Fishball board? Where you got resources for it? I cant find any that actually match the board. Big thanks in advance!

Szczecin-Radio, 2025-04-21:
I would appreciate it if I could see the Vivado sources of said board too... 

Regards,
/WS

F5OEO, 2025-04-21:
Vivado sources for tezuka located at https://github.com/F5OEO/maia-sdr/tree/sweep/maia-hdl/projects

Fishball original firmwares : https://github.com/Xiaozhang-code-cloud

JoyBed, 2025-04-21:
> Vivado sources for tezuka located at https://github.com/F5OEO/maia-sdr/tree/sweep/maia-hdl/projects
> 
> Fishball original firmwares : https://github.com/Xiaozhang-code-cloud

Interestingly tho, gpio_ctl2 is stated as L17 and in your sources you changed it to R14, may I ask why?

F5OEO, 2025-04-21:
[7010_936x_SDR.pdf](https://github.com/user-attachments/files/19839603/7010_936x_SDR.pdf)

I have not checked all pins, but this fixes the faslock gpio pins

JoyBed, 2025-04-21:
Huh, yeah, that pin is wrong, its R14 and not L17. You did not changed the rest of the pins? Id not then I will try to use your .xdc to port OpenWiFi to Fishball.

Szczecin-Radio, 2025-05-03:
> have not checked all pins, but this fixes the faslock gpio pins

May I have a simple diff / guidepost with changes in the original hdl project?

I managed to download another Vitis / Vivado 2022.2 stuff and successfully compiled the original project finally. Thanks a lot for the link. First thing, the temperature of the board (with that supplied aluminum "micro-coolers")  is very high. It seems the board is going to burn out slowly, even with CPU @666MHz... There must be definitely something wrong. 

```
$ SoapySDRUtil --probe="driver=plutosdr,uri=ip:pluto.local" 2>/dev/null | grep -e 'temp0:'
     * xadc_temp0: 71.609729 C
     * ad9361-phy_temp0: 55.263000 C
```  

Regards,
/WS

drspastic, 2025-05-23:
hi i also have the 7020 fishball. it arrived unrecognised by dmesg. i could log into root via jtag a couple of times but it then gave:
4  3  2  1  0 
Unknown command 'bootp' - try 'help'
Wrong Image Format for bootm command
ERROR: can't get kernel image!
Pluto> 

which seems a common issue. it also never got an ip address even with manual setup. i mainly want it for usb use to try via android tablet instead of linux pc. a mobile unit. 

does anyone have a guide to dfu updating and what files i should try to restore on it? i have the official repo files but not sure what to use.

73, de LZ2ILR
m0xek.pskmail@gmail.com

drspastic, 2025-06-15:
1. Execute sf probe 
2. Execute sf erase 0 0x1000000 
3. 
4. Just execute the above two commands again. The reason for the startup failure is that an incorrect environment variable was set and stored in flash, which needs to be erased

jhart99, 2025-07-03:
While investigating this device, I found the original repository https://github.com/XiaoZhangao/Fish-Wan-plutosdr-fw-7020 that the Hamgeek one is based off of.

tangradioxide, 2025-07-07:
i have the 7020 model. does tezuka support changing usb network modes?
i want a portable setup on android tablet but android only supports ecm not rndis. original Pluto can change mode for windows mac or android.

tangradioxide, 2025-07-13:
Please ignore that, it was a partitioning error. Tezuka boots. But fw_setenv usb_ethernet_mode ecm causes corruption and fail to reboot requiring insert original sd and sf probe and sf erase before it can boot again. Same as original fw. I'm going to try via config file but not holding much hope. 
If anyone has actually got ECM working please advise how. 
My hope is the portable android qo100 with this blackview tablet that has massive battery.

tangradioxide, 2025-07-13:
BTW tezuka doesn't recognize sf erase

rpl-james-overington2, 2025-08-24:
The results I get from 0.1.9 fishball 7020 release files is NOT what I get from a complete project build... I do not see the above issues from a complete build - though I made lots of config changes to get Vivado to build the bitstream. I DO see issues with DDS (not working), A/D calibrations (this is where the AD9363 is NOT the same as AD9361...) and Tx over ~2GHz. IIO is not working correctly. The "Complex 8bit mode to extend streaming bandwidth with host (14Mhz stable bandwidth through usb, 45MHz through GbE network)" implementation is evident in part but not complete or active in the project build.

rpl-james-overington2, 2025-08-29:
The project README cites:

**Complex 8bit mode to extend streaming bandwidth with host (14Mhz stable bandwidth through usb, 45MHz through GbE network)**

Using iperf the absolute max throughput I can achieve on the fishball board with this FW is:

# iperf -c 192.168.3.10 -u -b 950M -P 2 -l 1470 -w 2M -t 30
------------------------------------------------------------
Client connecting to 192.168.3.10, UDP port 5001
Sending 1470 byte datagrams, IPG target: 0.00 us (kalman adjust)
UDP buffer size: 4.00 MByte (WARNING: requested 2.00 MByte)
------------------------------------------------------------
[  2] local 192.168.3.1 port 60121 connected with 192.168.3.10 port 5001
[  1] local 192.168.3.1 port 43594 connected with 192.168.3.10 port 5001
[ ID] Interval       Transfer     Bandwidth
[  2] 0.00-30.00 sec  1.13 GBytes   323 Mbits/sec
[  2] Sent 823245 datagrams
[  1] 0.00-30.00 sec  1.08 GBytes   311 Mbits/sec
[  1] Sent 792334 datagrams
[SUM] 0.00-30.01 sec  2.21 GBytes   633 Mbits/sec
[SUM-2] Sent 1615579 datagrams
[  2] Server Report:
[ ID] Interval       Transfer     Bandwidth        Jitter   Lost/Total Datagrams
[  2] 0.00-30.00 sec  1.13 GBytes   323 Mbits/sec   0.000 ms 0/823244 (0%)
[  1] Server Report:
[ ID] Interval       Transfer     Bandwidth        Jitter   Lost/Total Datagrams
[  1] 0.00-30.00 sec  1.08 GBytes   311 Mbits/sec   0.000 ms 0/792333 (0%)

Note UDP, max MTU, 2M buffer + 2 x threads

Any more is just too much for the A9 to copy data out of the kernel DMA buffer - this is the limiting factor on this board. This throughput is the equivalent of almost 10MSPS using the standard IIO transfer (which itself has CPU overhead and thus further limits throughput...) In practice the highest actual IIO throughput over the gigabit ethernet IF is ~7MSPS.

I am working on a kernel module to effect true zero-copy transfers from Memory to GEM using DMA to hopefully get much closer to the PHY hardware limit (gigabit - along the lines of AF_XDP). Along with some FPGA data format repackaging I am hoping to get much closer to  60MSPS with 1ch 8bit complex samples... (enough acronyms?!?)

F5OEO, 2025-10-25:
Closed as too much subjects under a same issue. Please reopen with unique issue.

closed 2025-10-25 by F5OEO