F5OEO/tezuka_fw #99: ANTSDR E200 Firmware Fails to Set Static IP, Falls Back to DHCP/APIPA

Ali6114, 2025-04-20:
When using the latest firmware release for the ANTSDR E200, the device does not apply the static IP address defined in config.txt or uEnv.txt (e.g., ipaddr=192.168.2.1). Instead, the Ethernet interface (eth0:avahi) is assigned an APIPA address (169.254.2.133) via the Avahi service, indicating a failure to apply the user-defined static IP or obtain a DHCP-assigned IP.

**Environment**

- Device: ANTSDR E200 (Zynq Z-7020, AD9363)
- Firmware: Latest release 0.1.3
- Configuration File: uEnv.txt on USB mass storage or SD card
- Network Interface: eth0 (Cadence GEM, MAC: 00:0A:35:00:01:22)

**Expected Behavior**

The eth0 interface should be assigned the static IP 192.168.2.1 as defined in uEnv.txt.
No APIPA address (169.254.x.x) should be assigned.

**Actual Behavior**

The eth0:avahi interface is assigned an APIPA address (169.254.2.133).
The static IP (192.168.2.1) is not applied.

Output of ifconfig:

eth0      Link encap:Ethernet  HWaddr 00:0A:35:00:01:22
          inet6 addr: fe80::20a:35ff:fe00:122/64 Scope:Link
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          RX packets:296 errors:0 dropped:0 overruns:0 frame:0
          TX packets:48 errors:0 dropped:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:25814 (25.2 KiB)  TX bytes:7146 (6.9 KiB)
          Interrupt:36 Base address:0xb000

eth0:avahi Link encap:Ethernet  HWaddr 00:0A:35:00:01:22
          inet addr:169.254.2.133  Bcast:169.254.255.255  Mask:255.255.0.0
          UP BROADCAST RUNNING MULTICAST  MTU:1500  Metric:1
          Interrupt:36 Base address:0xb000

lo        Link encap:Local Loopback
          inet addr:127.0.0.1  Mask:255.0.0.0
          inet6 addr: ::1/128 Scope:Host
          UP LOOPBACK RUNNING  MTU:65536  Metric:1
          RX packets:16 errors:0 dropped:0 overruns:0 overruns:0 carrier:0
          collisions:0 txqueuelen:1000
          RX bytes:2537 (2.4 KiB)  TX bytes:2537 (2.4 KiB)

**Additional Details**

The boot log indicates a USB gadget failure (UDC core: composite_gadget: couldn't find an available UDC), which may affect the USB Ethernet bridge used for network connectivity.
The Avahi service appears to override the static IP configuration, assigning an APIPA address when no DHCP server is present.
The firmware includes services like dhcpd, httpd, and ifplugd, suggesting network configuration is expected, but the static IP settings are not honored.
The issue persists despite correct syntax in config.txt/uEnv.txt and proper file placement on the USB mass storage or SD card.

**Request**
Please investigate why the static IP configuration is not applied and provide a fix in the next firmware release. Alternatively, provide guidance on ensuring uEnv.txt settings are honored or disabling Avahi to prevent APIPA fallback.

Ali6114, 2025-04-21:
[](url)

F5OEO, 2025-04-22:
IP adress and other parameters are not using uenv.txt but flash environment (fw_setenv, fw_printenv)
Moreover, ipaddr is the ip adress of ip over usb,  not the gigabit one.
To set static ip use 
- config.txt (use usb this time to mount a virtual usb drive)
Set 
ipaddr_eth = 
netmask_eth = 255.255.255.0
gateway_eth =
With desired values, and eject.

Ali6114, 2025-04-22:
Actually, I am using the Window 10 environment. The problem is similar to the issue #96. On boot no IP address is assigned to the eth0 interface. The same  problem mentioned there like eth0 avahi takes over and is assigned a random IP.