F5OEO/tezuka_fw #107: Dual Receiver functionality in SDRangel and GNUradio for LibreSDR.

TeddySpaghetti11, 2025-04-28:
Hi, I was just wondering if anyone has gotten dual receivers to work with the LibreSDR running this firmware. I have set the config.txt to these settings:
mode = 2r2t
audio_mode = off
rf_input = rx1
rf_output = tx1

However in SDRangel, the add MIMO device doesn't seem to actually output anything on the second channel and therefore direction finding doesn't work. And in terms of GNUradio, I have gotten output from using channels [voltage0, voltage1] as the I and Q channels of RX1, but how do I get RX2?
Any help would be much appreciated

76nic, 2025-04-28:
See issue #83 I created some time back. I hope that will be addressed in the upcoming version. Not sure what priories F5OEO has planned to address in future releases.
Best DC0NG

TeddySpaghetti11, 2025-04-28:
> See issue [#83](https://github.com/F5OEO/tezuka_fw/issues/83) I created some time back. I hope that will be addressed in the upcoming version. Not sure what priories F5OEO has planned to address in future releases. Best DC0NG

Oh ok, thanks for that. Out of curiosity, what sampling rate have you been able to get up to with this firmware? For whatever reason with SDR++ when I go beyond 4mhz it tends to lag really bad

TeddySpaghetti11, 2025-04-28:
> See issue [#83](https://github.com/F5OEO/tezuka_fw/issues/83) I created some time back. I hope that will be addressed in the upcoming version. Not sure what priories F5OEO has planned to address in future releases. Best DC0NG

I do actually see there is this: https://github.com/F5OEO/tezuka_fw/commit/15805971a1f2c4948c4fa36783a2c28ba94d7330. I'm not sure if he actually managed to make it work or not though

76nic, 2025-05-04:
@TeddySpaghetti11, I have not tested the sampling rate. How have you tested it? Have you used USB or LAN? Sampling rate should be half with 2r2t - or?
@F5OEO, do you know expected max rates for LibreSDR for "USB 1r1t and 2r2t" as well as "LAN 1r1t and 2r2t". How would a proper testing setup look like?
Many thanks!
Best regards,
DC0NG

F5OEO, 2025-07-19:
Should be fixed with 66487c4e7961f92fdea36d08e27136d349cd8237

dsittner, 2025-08-05:
> Should be fixed with [66487c4](https://github.com/F5OEO/tezuka_fw/commit/66487c4e7961f92fdea36d08e27136d349cd8237)

Can confirm working on LibreSDR from OpenSourceSDRLabs.  See my comment [here](https://github.com/F5OEO/tezuka_fw/issues/83#issuecomment-3156844022)

dsittner, 2025-08-05:
> Oh ok, thanks for that. Out of curiosity, what sampling rate have you been able to get up to with this firmware? For whatever reason with SDR++ when I go beyond 4mhz it tends to lag really bad

I can go up to 5MHz in SDR++ (F50E0 fork) on a Pi 5 connected via usb 3.0 before I get buffer under-runs.

closed 2025-09-21 by TeddySpaghetti11