F5OEO/tezuka_fw #111: ad9361 invalid decimation 0 in filter config

MC42, 2025-05-02:
Hi, amazing work you've done with tezuka!  

Was trying to use it on the go with a LibreSDR (v5, so Plutoplus.dfu/frm, latest image) and I noticed that my hardware seems to be showing some strange glitches visually and audibly in sdrpp.  Using latest nightly for SDRPP (Windows, both patched libiio.dll to use network/ip and the built-in libiio DLL) 

[image]

Additionally, dmesg on the LibreSDR Rev.5 yields the following:

`ad9361 spi0.0: ad9361_validate_enable_fir: Invalid: Decimation 0 in filter config`

Said message only appears when the device is first "opened".  Any advice that could be given would be appreciated as to removing these pops / addressing the strange `dmesg` warning.

Thanks in advance!

MC42, 2025-05-03:
[image]

Further attempts to fix it did not remove the decimation 0 filter config message in `dmesg`, but this was what I saw before I realized that a gain of 32dB seemed to be a threshold, where above it the mass spikes were buried.

Additionally, it seems correlated strongly to use of Bluetooth peripherals, so i might very well just have poor shielding in said peripherals.

Thank you again for the good firmware!

F5OEO, 2025-07-19:
Should be fixed with latest sdrpp fork.

MC42, 2025-07-22:
I appreciate the information, and can confirm that the newest nightly from SDR++ seems to have solved the issue.  Apologies for the misattribution.

closed 2025-07-22 by MC42