F5OEO/tezuka_fw #118: OverlayFS for storing settings

crvux, 2025-05-26:
I don't know about other boards, but Libre has NOR system memory (W25Q256). In the OpenWrt project there is an approach in which the jffs partition of the NOR memory is mounted as an overlay for settings storage: https://openwrt.org/docs/techref/flash.layout

In Tezuka the settings are stored as u-boot environment variables. The typical process of applying the setting is `fw_printenv` in the init script and then writing a line to some config in /etc: https://github.com/search?q=repo%3AF5OEO%2Ftezuka_fw%20fw_printenv&type=code.
It seems that the process of storing settings can be significantly simplified if we switch to the approach used in OpenWrt.

What do you think?

F5OEO, 2025-05-26:
There is already a a JFFS2 where we could store some variables. I had also the same idea in order to use it but didn't implement it not to disturb "pluto analog devices" users. But why not.

crvux, 2025-06-14:
I found that using an environment variable from u-boot you can determine where the boot was made from. With this variable we can select upperdir overlay on SD card or in NOR flash. This way we can store all the persistent settings (and more) exclusively on the media from which the boot was made.

> 4. Bootmode
> 
> Zynq has a facility to read the bootmode from the slcr bootmode register
> once user is setting through jumpers on the board - see page no:1546 on [5]
> 
> All possible bootmode values are defined in Table 6-2:Boot_Mode MIO Pins
> on [5].
> 
> board_late_init() will read the bootmode values using slcr bootmode register
> at runtime and assign the modeboot variable to specific bootmode string which
> is intern used in autoboot.
> 
> SLCR bootmode register Bit[3:0] values
> #define ZYNQ_BM_NOR		0x02
> #define ZYNQ_BM_SD		0x05
> #define ZYNQ_BM_JTAG		0x0
> 
> "modeboot" variable can assign any of "norboot", "sdboot" or "jtagboot"
> bootmode strings at runtime.
https://github.com/analogdevicesinc/u-boot-xlnx/blob/master/doc/README.zynq#L36

I managed to read this variable in the uboot shell, but for some reason it is not available in the userspace (`fw_printenv modeboot`). There seems to be some confusion with how the uboot environment is loaded.
It also seems that this problem is very closely related to https://github.com/F5OEO/tezuka_fw/issues/26
I'll try to figure out the u-boot env loading process.

crvux, 2025-06-14:
JFI https://docs.u-boot.org/en/latest/usage/environment.html