F5OEO/tezuka_fw #137: LibreSDR Problem with *.frm Files since 0.1.6.0

DL1JM, 2025-07-05:
Hello,

everything was fine with v. 0.1.5.2 when I use the frm-files on my LibreSDR, but since 0.1.6.0 and 1.6.1 I get connection-errors when starting SDR-console or SDR-Television (no connection via ethernet possible), but the device is pingable and putty-connection also possible.

[image]

When I use the Libre-SD-version of 0.1.6.0 or 1.6.1. there is no problem.

I tried several times to use the actual frm-files, but no sccess. When I revert to 0.1.5.2 the problem is gone.

Any informations about that?

vy73s Manfred, DL1JM

F5OEO, 2025-07-06:
Thanks in pointing it. I mainly test with sd image as flash (frm) is not recommended. Need to investigate.

DL1JM, 2025-07-07:
Hi Evariste,

V 0.1.7 seems to remove my problem with the frm-Files!

SD-card-Version and frm-Files working fine again with my LibreSDR together with SDR-console and SDR-Television.
No connection problems!

Thanks and vy73s: Manfred, DL1JM

You may close this issue ;-)