F5OEO/tezuka_fw #140: Fishball TX is extremely noisy

Nannk, 2025-07-10:
Transmitting GMSK modulated stream:
Gnu radio flowchart:

[image]

Output to another sdr, fishball tx is connected through a 5m coax to RTL-SDR dongle, fishball is running tezuka fw [0.1.7](https://github.com/F5OEO/tezuka_fw/releases/tag/0.1.7):
[image]

Same setup, except fishball is running firmware provided by hamgeek [on the shop page](https://www.hgeek.com/collections/software-defined-radio/products/hamgeek-70mhz-6ghz-zynq7010-ad9363-sdr-software-defined-radio-development-board-for-pluto-sdr-matlab); direct link:[https://down.hgeek.com/download/92589-93267/7010_AD936X_SDR.zip](https://down.hgeek.com/download/92589-93267/7010_AD936X_SDR.zip) : 

[image]

With calibration mode `manual` or `manual_tx_quad` the problem is also apparent with simple sine/fm modulated signlals:

[image]

Setting calibration mode to `tx_quad` (through `osc, adi iio oscilloscope`) reduces the emissions for sine/fm modulated signals, but changes nothing for gmsk. 

Fishball Zynq7010

Nannk, 2025-10-27:
Thank you!
Seems to be fixed.
on the 0.2.1 1r1t & 2r2t :

[image]

0.1.5.2 looks the same as 0.2.1.

I also re-run the tx test on 0.1.7 just as a sanity check (i dont have rtl-sdr available at the moment) and it looks like this: 

[image]

closed 2025-10-27 by Nannk

F5OEO, 2025-10-27:
Great !